ElderlyPirate/Bus_Pirate #6: Remove code for Bus Pirate v2?

agatti, 2016-08-25:
I removed support for Bus Pirate v1A with 0d99cb7e0acdae197b4835767fb8a6b68a89ac51, which was for a device that is almost six years old now.  I don't know how many Bus Pirate v2 users are out there, and therefore whether to remove support for it or not.

This question should probably be read as "Should we focus only on V3 and V4?", though.  I personally only have one v4 unit and when performing cleanups or changes I cannot validate anything involving v3 devices.  If somebody has such a thing and is willing to help out in testing, then please speak up...

kallisti5, 2016-08-25:
Looks like BPv2 started as a Hack-a-day project circa 2009. (pre DangerousPrototypes)
http://hackaday.com/2009/06/25/how-to-the-bus-pirate-v2-with-usb/

I'd say drop it.. Not sure how many were sold... but i've yet to see one... and getting one for testing would likely be pretty painful.

agatti, 2016-08-26:
Fair enough.  Support for v2 and v2.5 has been removed in 674d801c3bc34823f4cb81748dfad0d044f05206.

closed 2016-08-26 by agatti

meowsqueak, 2016-12-21:
I have a v2go that I've just dug out of storage - although it looks like I'm a few months too late in this respect. It's probably not worth reinstating v2 support _just_ for me, but if a few more users speak up and you did want me to validate anything I could help.

agatti, 2016-12-21:
@meowsqueak v2go is 99% compatible with v3.  If you don't plan to read from the ADC then you should be able to just use the firmware for v3 as it is now.

I'll come up with a custom build for v2go this weekend for you to try, but only as long as you are able to un-brick it if I get something wrong (as in, getting close and personal with PicKit and reflashing the factory image yourself).

meowsqueak, 2016-12-21:
@agatti thanks, I have a RPi and a Ci40, both of which can be configured as PIC programmers, and I can get hold of a PicKit3 if need be, so that should be OK.

No rush though - I'm not able to do much for a week or two anyway. Create a new issue, or contact me at meowsqueak at gmail dot com if you want to move this conversation from here.

USBEprom, 2016-12-21:
@agatti 

agatti wrote:
> I'll come up with a custom build for v2go this weekend for you to try,
 
Sound good, as always you are the best!
Thank you sir.

agatti, 2016-12-24:
@meowsqueak can you please try to flash a v3 firmware from http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=45#p65526 and try to print the pins state with the `v` command?

If you take a look at https://github.com/BusPirate/Bus_Pirate/commit/674d801c3bc34823f4cb81748dfad0d044f05206#diff-e77a498013a2a203e09c0d8eb04279ed you can see that v2go is essentially the same as v3, except for ADC/voltage pin mappings.

Now, if this firmware boots and doesn't blow things up, then I can revert that change and resume v2/v2.5/v2go support without any immediate issues - although the future might be a bit uncertain regarding bugs that I cannot foresee/test for on those boards (I only have a v4 board myself, so v3 users have the same problem).

DavidAntliff, 2016-12-25:
@agatti thank you, I will try this in the first week of 2017 and report back here.

meowsqueak, 2016-12-30:
@agatti: using the v5.10 `pirate-loader` application:

```
$ /pirate-loader_mac --dev=/dev/tty.usbserial-A7004q3B --hex=../Testing/12112016/12112016_s.hex
...
Erasing page 41, a400...OK
Writing page 41 row 328, a400...OK
Writing page 41 row 329, a480...OK
Writing page 41 row 330, a500...OK
Writing page 41 row 331, a580...OK
Writing page 41 row 332, a600...OK
Writing page 41 row 333, a680...OK
Writing page 41 row 334, a700...OK
Writing page 41 row 335, a780...OK
Erasing page 42, a800...ERROR [50]
```

I get the same error with the _1 build also.

**However**, something is working because if I power-cycle the BP and connect via USB serial:

```
HiZ>i
Bus Pirate v3.a
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.4
DEVID:0x0447 REVID:0x3043 (24FJ64GA002 B5)
http://dangerousprototypes.com
```

And:

```
HiZ>v
Pinstates:
1.(BR)  2.(RD)  3.(OR)  4.(YW)  5.(GN)  6.(BL)  7.(PU)  8.(GR)  9.(WT)  0.(Blk)
GND     3.3V    5.0V    ADC     VPU     AUX     CLK     MOSI    CS      MISO
P       P       P       I       I       I       I       I       I       I
GND     0.00V   0.00V   0.00V   0.00V   L       L       L       L       L
```

Incidentally, I do have an ICSP programming method working, using `pickle` via a RPi3, which I've used to reprogram the v4.4 bootloader successfully. I then install the firmware via `pirate-loader` as I haven't been able to "combine" the bootloader and firmware via the ICSP method (the second image overwrites the first, I think).

USBEprom, 2016-12-31:
@meowsqueak 
 
meowsqueak wrote:
> @agatti: using the v5.10 pirate-loader application:
>
> $ /pirate-loader_mac --dev=/dev/tty.usbserial-A7004q3B --hex=../Testing/12112016/12112016_s.hex
...
> Erasing page 41, a400...OK
> Writing page 41 row 328, a400...OK
> Writing page 41 row 329, a480...OK
> Writing page 41 row 330, a500...OK
> Writing page 41 row 331, a580...OK
> Writing page 41 row 332, [...]
 
That is the expected behavior using the firmware v7.0 and it is totally fine.
You must use previous releases in order to do not experimenting the final error.
But easy, never mind, simply you ignores the error which does not involve anything.
That error is due a protection into the bootloader.
 
https://github.com/BusPirate/Bus_Pirate/issues/26
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498#p65296

**However you must use 12112016_1.hex which is optimized, not 12112016_s.hex which is not optimized.
Otherwise in some situations you might have timing issues.
So use 12112016_1.hex, not 12112016_s.hex!
I built 12112016_s.hex only for testing, it is better do not use it but rather 12112016_1 in its place.**

meowsqueak wrote:
> Incidentally, I do have an ICSP programming method working, using pickle via a RPi3, which I've used to reprogram the v4.4 bootloader successfully. I then install the firmware via pirate-loader as I [...]
 
I do not think so.
All was fine from the beginning otherwise you would not have been able to load the newest 12112016_s.hex firmware.
However glad to read you have solved.
Happy new year!

agatti, 2017-01-01:
@meowsqueak That's good to know - I'll revive v2go later on today, then.

meowsqueak, 2017-07-20:
Just wondering if v2go was successfully revived and still operational in latest firmware?

agatti, 2017-07-20:
My bad, I'll do that now.  I'm still not sure if there will be official builds for such a thing though.

agatti, 2017-07-20:
@meowsqueak there you go, please check out the latest version of the source, change the parameter defined here: https://github.com/BusPirate/Bus_Pirate/commit/8ab697a81afc4cdaf110e08af9646d9ffff87b13#diff-0fc478318c3341c2e337e5d7003d2383R24 and rebuild

As always, no guarantees that it will work on your hardware, since I haven't got a v2 board to test things with.