ElderlyPirate/Bus_Pirate #7: Test the build on a real v3 board

agatti, 2016-08-25:
Now that the firmware can build for v3, it would be nice to know whether it actually works or not.  Unfortunately I only have a v4 board here...

CoryHenderson, 2016-08-25:
I can test on v3 hardware. I have a v3.5c, date stamped Sep 2011. Is there a specific test procedure, or are you just making sure it works at all?

CoryHenderson, 2016-08-25:
I can confirm that the latest firmware builds and runs on the v3.5c that I have. Terminal looks like it is working fine. I do have an issue with I2C that I didn't have in 6.1 which I have been using regularly. When reading a register (`[0x34 0][0x35 r r]`), only the first read comes back correct. The second (and all subsequent) read always comes back 0xFF, but the ACKs are working correctly.

kallisti5, 2016-08-25:
Hm.. I wonder if the I2C regressions were caused by 3c820687e8583055821974d4c30356059746d391 ?

CoryHenderson, 2016-08-26:
I can check that tomorrow. I just so happened to be using that when I saw the request for a tester, so it was the first thing I noticed

agatti, 2016-08-26:
@CoryHenderson would it be possible to check whether the board also passes selftest and if the hardware information report makes any sense?

CoryHenderson, 2016-08-26:
@kallisti5 Wasn't 3c82068. Just tried reverting the commit, same results.

CoryHenderson, 2016-08-26:
@agatti Self test looks good:

> Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.79) OK
VPU(4.80) OK
3.3V(3.23) OK
ADC(3.27) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on!
Any key to exit
Found 0 errors.

By hardware information, do you mean just the `i` command in the terminal? Here's what that gives. Looks good. I think the bootloader info is off because I'm flashing directly with an ICD3.

> Bus Pirate v3.5
Community Firmware v7.0 - goo.gl/gCzQnW  Bootloader v255.255
DEVID:0x0447 REVID:0x3043 (24FJ64GA002 B5)
http://dangerousprototypes.com

agatti, 2016-08-26:
@CoryHenderson yep, that's all I needed to check - thanks!

For I2C, I'll try to get the last revision off DP's repo to build for v3 and MPLAB X later on today, I hope you can give it a spin when you have the time.

CoryHenderson, 2016-08-26:
@agatti Sounds good. Today will be the last day I have for a few weeks, but I can hopefully start contributing more starting mid September.

agatti, 2016-08-26:
@CoryHenderson, here's a build from the tree at https://github.com/DangerousPrototypes/Bus_Pirate/commit/10f1ec16ff24ab050ec8d048f398dc4ef6a8514f - disabling enough features (OpenOCD, Basic interpreter, JTAG, PC-AT Keyboard) to fit in the available ROM space.  Thanks for the help :)

[busPirate.X.production.hex.zip](https://github.com/BusPirate/Bus_Pirate/files/439764/busPirate.X.production.hex.zip)

agatti, 2016-08-26:
@kallisti5, do you think it would help to ping Seeed/Sparkfun/Dangerous Prototypes directly and ask for a fully assembled board to support this endeavour?  If it's Seeed, I'll be in Shenzhen later this month so I could even pop by and pick it up in person ;)

CoryHenderson, 2016-08-26:
@agatti v6.3 has the I2C bug as well. I had a version of v6.2 (r1981) laying around from a while back. Tried that, and it works fine. I am attaching text file logs of my I2C read operation from each. As you can see, the version from r1981 has the correct output of 0x8904, which is the device ID of the WM8912 I am reading from. The r2151 build returns 0x89FF
[I2CTest-v6.2-r1981.txt](https://github.com/BusPirate/Bus_Pirate/files/439957/I2CTest-v6.2-r1981.txt)
[I2CTest-v6.3-r2151.txt](https://github.com/BusPirate/Bus_Pirate/files/439958/I2CTest-v6.3-r2151.txt)

agatti, 2016-08-27:
@CoryHenderson great catch!  I'll have a go at it this weekend and see what I can get out of it, although I'll have to first rig something up on an ESP8266 or Arduino to have any I2C bus to read from.

agatti, 2016-08-27:
@CoryHenderson, can you please give this firmware a try when you have the time in a few weeks?  All the DP SVN dumps I found so far stop at r2115 - need to find more recent dumps I guess...

[BPv3-firmware-v6.3-r2088.hex.zip](https://github.com/BusPirate/Bus_Pirate/files/440571/BPv3-firmware-v6.3-r2088.hex.zip)

coelner, 2016-09-02:
So far,  a sniffing between a BMP280 and ESP8266:
```
I2C>i
Bus Pirate v3.5
Firmware v6.3-beta1 r2088  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
CFG1:0xFFDF CFG2:0xFF7F
*----------*
Pinstates:
1.(BR)  2.(RD)  3.(OR)  4.(YW)  5.(GN)  6.(BL)  7.(PU)  8.(GR)  9.(WT)  0.(Blk)
GND     3.3V    5.0V    ADC     VPU     AUX     SCL     SDA     -       -
P       P       P       I       I       I       I       I       I       I
GND     0.00V   0.00V   0.00V   0.00V   H       L       L       H       H
POWER SUPPLIES OFF, Pull-up resistors OFF, Open drain outputs (H=Hi-Z, L=GND)
MSB set: MOST sig bit first, Number of bits read/write: 8
a/A/@ controls AUX pin
I2C (mod spd)=( 0 1 )
*----------*
I2C>W
POWER SUPPLIES ON
Clutch engaged!!!
I2C>(1)
Searching I2C address space. Found devices at:
0xEC(0x76 W) 0xED(0x76 R)
I2C>[0xEC 0xD0[0xED r
I2C START BIT
WRITE: 0xEC ACK
WRITE: 0xD0 ACK
I2C START BIT
WRITE: 0xED ACK
READ: 0x60
I2C>(2)
Sniffer
Any key to exit
[0xEC+0x89+][0xED+0x6E-[][0xEC+]][0xED+]][0xEC+]][0xED+0xCF+][0xEC+]][0xED+]][0xEC+]][0xED+0x00-[][0xEC+]][0xED+0x65+][0xEC+]][0xED+]][0xEC+0x8E+][0xED+]][0xEC+]][0xED+]][0xEC+0x90+][0xED+0x2F+][0xEC+]][0xED+0x17+][0xEC+]][0xED+0xD0-[][0xEC+]][0xED+0x2D+][0xEC+0x94+][0xED+0x19+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x2B+][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+0x9B+][0xED+0x26-[][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+0x15+][0xEC+0x9F+][0xED+0x10-[][0xEC+0x9E+][0xED+0xBD-[][0xEC+]][0xED+0x97+][0xEC+]][0xED+0x03+][0xEC+]][0xED+0xD3+][0xEC+0xE3+][0xED+0x01+][0xEC+0xE4+][0xED+0x29+][0xEC+]][0xED+0x01+][0xEC+0xE6+][0xED+0x01+][0xEC+0xE5+][0xED+0x01+][0xEC+0xE7+][0xED+0x3D+][0xEC+0xF4+0x00+][0xEC+]][0xEC+0xF2+0x01+][0xEC+]][0xEC+0xD0+][0xED+0xC1+][0xEC+]][0xED+0x60-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x4F+][0xEC+]][0xED+0x03+][0xEC+0x80+][0xED+]][0xEC+]][0xED+0xA1+][0xEC+]][0xED+]][0xEC+]][0xED+0x65+][0xEC+]][0xED+0x7D-[][0xEC+0x85+][0xED+0x05+][0xEC+0x86+][0xED+0x71+][0xEC+]][0xED+0x0D+][0xEC+]][0xED+]][0xEC+]][0xED+0xDD+][0xEC+]][0xED+]][0xEC+0x8B+][0xED+0xCF+][0xEC+]][0xED+0x65+][0xEC+0x8D+][0xED+0x01+][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+0x17-[][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+0x93+][0xED+0x0B-[][0xEC+0x94+][0xED+0x0C-[][0xEC+]][0xED+0x2D+][0xEC+0x96+][0xED+0x2B+][0xEC+]][0xED+0x01+][0xEC+0x98+][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+0x4D+][0xEC+]][0xED+0x15+][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+0x21+][0xEC+0xA0+][0xED+0x01+][0xEC+0xA1+][0xED+0x4B-[][0xEC+0xA2+][0xED+0xD1-[][0xEC+0xA3+][0xED+0x00-[][0xEC+]][0xED+0x01+][0xEC+0xA5+][0xED+0x01+][0xEC+0xA6+][0xED+0x01+][0xEC+0xA7+][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x67+][0xEC+]][0xED+0x01+][0xEC+0xAE+][0xED+0x00-[][0xEC+0xAF+][0xED+0xC0-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x54-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+0xB4+][0xED+0x00-[][0xEC+0xB5+][0xED+0x01+][0xEC+]][0xED+0xC1+][0xEC+]][0xED+0x05+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x03+][0xEC+0xBA+][0xED+0xFF-[][0xEC+]][0xED+]][0xEC+]][0xED+0x3F+][0xEC+]][0xED+0x9D+][0xEC+]][0xED+0x11+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x81+][0xEC+]][0xED+0x4F+][0xEC+]][0xED+0xFF-[][0xEC+]][0xED+0x00-[][0xEC+]][0xED+0x01+][0xEC+0xC6+][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x03+][0xEC+0xC9+][0xED+0x01+][0xEC+0xCA+][0xED+0x01+][0xEC+0xCB+][0xED+0x00-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+0xCE+][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+0xD0+][0xED+0x60-[][0xEC+0xD1+][0xED+0x00-[][0xEC+]][0xED+0x01+][0xEC+0xD3+][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x00-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+0xDE+][0xED+0x00-[][0xEC+0xDF+][0xED+0x01+][0xEC+]][0xED+0x01+][0xEC+]][0xED+0xD3+][0xEC+]][0xED+0x03+][0xEC+]][0xED+0x01+][0xEC+0xE4+][0xED+0x14-[][0xEC+0xE5+][0xED+0x00-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0x3D+][0xEC+]][0xED+0x67+][0xEC+]][0xED+0x83+][0xEC+0xEA+][0xED+0xFF-[][0xEC+0xEB+][0xED+0xFF-[][0xEC+0xEC+][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+0xEF+][0xED+]][0xEC+0xF0+][0xED+0xFF-[][0xEC+0xF1+][0xED+0x01+][0xEC+]][0xED+0x03+][0xEC+]][0xED+0x19+][0xEC+]][0xED+0x4F+][0xEC+]][0xED+0x01+][0xEC+0xF6+][0xED+0x00-[][0xEC+0xF7+][0xED+0xB7+][0xEC+]][0xED+]][0xEC+]][0xED+0x01+][0xEC+]][0xED+]][0xEC+0xFB+][0xED+0x62-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0xF9+][0xEC+0xFE+][0xED+]][0xEC+]][0xED+]][0xEC+]][0xED+]][0xEC+0xFB+][0xED+0x67-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+]][0xEC+]][0xED+0x67-[][0xEC+]][0xED+0x01+][0xEC+0xF7+][0xED+0xB7+][0xEC+]][0xED+]][0xEC+0xF9+][0xED+0x01+][0xEC+0xF7+][0xED+0xB7+][0xEC+0xF8+][0xED+0xD6-[][0xEC+]][0xED+0x01+][0xEC+0xF7+][0xED+0xB7+][0xEC+]][0xED+0xD6-[][0xEC+]][0xED+0x01+][0xEC+]][0xED+0xF9+][0xEC+0xFE+][0xED+0x0A-[]
```

The debug output via serial:
```
Starting BME280... result of .begin(): 0x60
Displaying ID, reset and ctrl regs
ID(0xD0): 0x60
Reset register(0xE0): 0x0
ctrl_meas(0xF4): 0x27
ctrl_hum(0xF2): 0x1

Displaying all regs
0x80:8E 50 89 32 7D 02 38 06 EC 6E E7 67 32 00 9F 8F 
0x90:17 D6 D0 0B 0C 16 15 00 F9 FF AC 26 0A D8 BD 10 
0xA0:00 4B D1 00 00 00 00 00 00 00 00 00 33 00 00 C0 
0xB0:00 54 00 00 00 00 60 02 00 01 FF FF 1F 4E 08 00 
0xC0:00 40 27 FF 00 00 00 00 01 00 00 00 00 00 00 00 
0xD0:60 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0xE0:00 69 01 00 14 00 00 1E 33 41 FF FF FF FF FF FF 
[... cut ...]
dig_P5, int16: 21
dig_P6, int16: -7
dig_P7, int16: 9900
dig_P8, int16: -10230
dig_P9, int16: 4285
[... 9 lines cut ...]
Pressure: 98909.00 Pa
Altitude: 210.03m
Altitude: 689.06ft
%RH: 62.00 %

Temperature: 25.27 degrees C
Temperature: 77.52 degrees F
Pressure: 98910.00 Pa
Altitude: 209.94m
Altitude: 688.77ft
%RH: 64.00 %
```
The double read seems ok:
```
I2C>[0xec 0xf2 [0xed r r]
I2C START BIT
WRITE: 0xEC ACK
WRITE: 0xF2 ACK
I2C START BIT
WRITE: 0xED ACK
READ: 0x01
READ:  ACK 0x0C
NACK
I2C STOP BIT
```

agatti, 2016-09-03:
@coelner thanks for the report! Can you please check if using any newer firmware than r2088 trigger the problem on double read? Maybe if you can capture the signals in something I can open with Saleae Logic would also help if double read fails.

coelner, 2016-09-03:
I tried this version: BPv3-firmware-v6.3-r2151_by_tayken-spi_fix.hex
So far, it seems to be the same. Unfortunalty I don't have a logic analyser. At least under windows ols and pulsesiew in combination with the buspirate fail.

mikebdp2, 2016-09-18:
@agatti, USBEprom is providing some test reports for BPv3 at DP forum
(BusPirate firmware 7.0 official thread) You may check it if you are curious...
http://dangerousprototypes.eu/forum/viewtopic.php?f=28&t=8498#p65306

agatti, 2016-09-18:
The scripting engine (BASIC interpreter) is disabled by default on v3, I'll see if I can re-enable it without going out of RAM.  However the VREG LED not lighting up is sort of weird - I'll quickly look in the source code to see if anything went wrong when cleaning the selftest module up.

However, maybe it'd be helpful if people would post bugs here rather than on different forums, as not everyone has an account on dangerousprototypes' forum :)

agatti, 2016-09-18:
The LED issue has been fixed with 10e9ea1877d1040ae52ab56a0e042ea916cfce95 - but for the BASIC interpreter I'm not sure about enabling it by default.  Once enabled, v3 has literally 2% of RAM left, which means that the moment we add anything else we have to switch that one off.

agatti, 2016-09-18:
For the record, the BASIC interpreter can be enabled here: https://github.com/BusPirate/Bus_Pirate/blob/10e9ea1877d1040ae52ab56a0e042ea916cfce95/Firmware/configuration.h#L306

mikebdp2, 2016-09-18:
@agatti, wow, sure USBEprom will be happy to hear about this. Thank you!
By the way I wanted to rebuild BPv3-bootloader but noticed it doesn't contain MPLAB X file,
and can't find the sources too. Were they publicly released at some point of time?

agatti, 2016-09-18:
BPv3's bootloader had a troubled history.  As you can see in /BPv3-bootloader, there are 5 different bootloaders, and there are different steps to take to upgrade from one version to the other.

If you're not using ICD and updating via the serial port, then http://dangerousprototypes.com/blog/2010/01/18/how-to-bus-pirate-bootloader-v4-upgrade/ is what you need to read first - then migrate firmware-v4.1, firmware-v4.3, firmware-v4.5, upgrader-v2tov4, and upgrader-v4tov4 from MPLAB 8 to MPLAB X.

Now, from the link on dangerousprototypes.com it seems like the user in question is using bootloader v4.4 - which is not available in source form.  Chances are it is v4.5 but the bootloader version was not updated in the source code.  Considering the elevated risk of bricking the board and have to resort to ICD, I'd be careful in touching this and just keep the bootloader as it is.  Plus, it's written in 100% assembler, so you can crank up the optimisation level as high as you want but nothing will change...

agatti, 2016-09-18:
So, v4.4 is in firmware-v4.3.  Moreover, here's the diff from v4.3 to v4.5:

```diff
agatti@tardigrade ~/src/Bus_Pirate/BPv3-bootloader $ diff firmware-v4.3/src/ firmware-v4.5/src
diff --git a/firmware-v4.3/src/ds30loader.s b/firmware-v4.5/src/ds30loader.s
index 19ff413..81809c0 100644
--- a/firmware-v4.3/src/ds30loader.s
+++ b/firmware-v4.5/src/ds30loader.s
@@ -147,7 +147,7 @@
 ;		.equ	STARTADDR,	( FLASHSIZE - 2*(PAGESIZE * 2) ) 		/*place bootloader in 2nd last program page*/
 		.equ	STARTADDR,	( FLASHSIZE - (2* (PAGESIZE)) ) 		/*place bootloader in last program page*/
 		.equ	BLCHECKST,  ( STARTADDR - (ROWSIZE) )			/*precalculate the first row write position that would overwrite the bootloader*/
-		.equ	BLVERSION,	0x0404	;bootloader version for Bus Pirate firmware (located at last instruction before flash config words)
+		.equ	BLVERSION,	0x0405	;bootloader version for Bus Pirate firmware (located at last instruction before flash config words)
 ;------------------------------------------------------------------------------
 ; Validate user settings
 ;------------------------------------------------------------------------------
@@ -316,6 +316,8 @@ setup:
 		rcall 	Receive
 		sub 	#HELLO, W0			;check
 		bra 	z, helloOK		; prompt
+		sub 	#'#', W0			;check
+		bra 	z, exit		; prompt

 		SendL   'B'
 		SendL   'L'
```

Whose only difference would be to abort the bootloading process if 0xE4 is sent as the first character instead of 0xC1.  If the user was able to reflash a custom firmware, he's good to go and he's better off leaving that bootloader alone :)

mikebdp2, 2016-09-18:
Interesting note about BASIC interpreter: with "s" option it takes just 69% instead of 98% ;)
But yes its optional feature, probably not many people need it, so could be disabled by default

USBEprom, 2016-09-19:
Hi all.
Hoping not to mess.
I own a Bus Pirate revision 3.6 and normally I write on dangerousprototypes.com.
Here are some thought about the new firmware v7.0 for the Bus Pirate.
First I have to thanks mikebdp2, agatti, kallisti5 and all the guys involved in the project: THANKS!
So said, here you go:

I had a look into this forum but honestly I understood very few things.
I am just a hobbyist, not a software engineer.
The only thing I can do is perhaps help with debug, perhaps...
I am sorry, no more.
I tested both the new releases.
Here you go.

bpv3_fw7.0_opt0_18092016.hex
bpv3_fw7.0_opt1_18092016.hex
bpv3_fw7.0_opt2_18092016.hex
bpv3_fw7.0_optS_18092016.hex

First I have converted them into uppercase in order to use pirate-loader for the upgrade, then I upgrade my Bus Pirate 3.6 with them all.
For each of them I got the error message that I have already written:

Writing page 41 row 334, a700...OK
Writing page 41 row 335, a780...OK
Erasing page 42, a800...ERROR [50]

Error updating firmware :(

Based on explanation of mikebdp2 and of agatti and of others, seems to not be a problem.
So okay.

HiZ>i
Bus Pirate v3.5
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ UART I2C SPI] Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.94) OK
VPU(5.03) OK
3.3V(3.31) OK
ADC(3.34) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on! [OKAY: VREG DOES LIGHT]
Any key to exit
Found 0 errors.
HiZ>#
[... 168 lines cut ...]
(34)>8
Adjust your terminal
Space to continue

It works!

MACRO(1) and MACRO(2) in 2WIRE protocol they do not work.
For what I can understand MACRO(1) fails due a wrong order in the lsb/MSB of the emitted bytes.
Also MACRO(2) fails but I do not know why, perhaps for the same reason of MACRO(1).

For what I know up to 255 characters of syntax may be entered into the Bus Pirate terminal at once.
I have tried but with them all the firmwares only 127 characters seems to be accepted, the extra ones are rejected (in the terminal that rejected characters are shown like not ascii)

I have not an oscilloscope so I can not say if SPI protocol is okay or still has the MOSI problems (viewtopic.php?f=4&t=6667#p58785) fixed by tayken doing correction of typo found by luftek (viewtopic.php?f=4&t=6667&start=15#p59041)

I can not test if the i2c timeout problem has been fixed or not (viewtopic.php?f=4&t=8546)

Last but not least, the software flashrom_v0.9.9_Win does work!

Hoping those are useful.
Thank you friends!

Be seeing you.

U.Sb

agatti, 2016-09-20:
@USBEprom I limited the buffer size on v3 to 127 characters due to memory constraints when trying to fit as many module as possible: https://github.com/BusPirate/Bus_Pirate/blob/10e9ea1877d1040ae52ab56a0e042ea916cfce95/Firmware/configuration.h#L502 - that can be easily set back to 255 but I'm not sure if I can still fit the BASIC interpreter in as well.

agatti, 2016-09-20:
@USBEprom The SPI MOSI typo problem is already fixed in firmware 7.00.

mikebdp2, 2016-09-20:
@agatti , it is possible to fit many features for BPv3 (including BASIC) but only for three opt levels:
opt0 - **_not enough program memory_**
opt1 - 84% occupied,
opt2 - 94% occupied, faster speed,
optS - 77% occupied, slower speed,
opt3 - **_not enough program memory_**

**./Bus_Pirate/Firmware/configuration.h, lines 304-319**
```
#ifdef BUSPIRATEV3
#define BP_ENABLE_1WIRE_SUPPORT // <--- + enabled
#define BP_ENABLE_BASIC_SUPPORT // <--- + enabled
#define BP_ENABLE_DIO_SUPPORT // <--- + enabled
#undef BP_ENABLE_HD44780_SUPPORT
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#undef BP_ENABLE_PIC_SUPPORT
#undef BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#define BP_ENABLE_RAW_2WIRE_SUPPORT // <--- + enabled
#define BP_ENABLE_RAW_3WIRE_SUPPORT // <--- + enabled
#define BP_ENABLE_SMPS_SUPPORT // <--- + enabled
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */
```
BP_ENABLE_HD44780_SUPPORT could also be enabled (takes just about 1% extra)

However, there are problems with BP_ENABLE_PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT. I create new issue about them:
https://github.com/BusPirate/Bus_Pirate/issues/19

tbox1911, 2016-09-20:
Hi all folks,
I have a BP v3.5 Firmware v6.1 r1676 (work well as i can tell)
I'm glad to help in testing this new firmawre  (i2c and SPi memory reading)
btw, what is the latest -stable- firmware version, I mean, without the I2C bug at least ?
v6.2-r1981 ?
Thanks

USBEprom, 2016-09-20:
@agatti
Thanks for the explanation about the 128 characters instead of 256.
In my opinion, course if it is possible, would be better to bring back to 256 characters the manageable string.
This would ensure compatibility with any documented examples.
I am of the idea that if it can do, would be better keep all the features as possible even if approaching the memory limit.
If this is not possible then would be better remove BASIC that seems to be what occupies more space in memory.
But I do not know if there is the need to make room for additional features or not, so if there is not then BASIC could be preserved.
@agatti
Okay I understand that the SPI MOSI typo problem is already fixed in firmware 7.00. thanks!
It is great, well done!
@agatti
I would like compile a new firmware by myself but honestly It is beyond my abilities, sorry.
I guess here you all are software engineers or something like that.
Instead I have no skills in that field.

USBEprom, 2016-09-20:
@mikebdp2 
"it is possible to fit many features for BPv3 (including BASIC) but only for three opt levels:"
"opt0 - not enough program memory"
"opt1 - 84% occupied,"
"opt2 - 94% occupied, faster speed,"
"optS - 77% occupied, slower speed,"
"opt3 - not enough program memory"
"BP_ENABLE_HD44780_SUPPORT could also be enabled (takes just about 1% extra)"
 
mikeb it would be great, course if possible.
As I already wrote if there is not any need to make room for something new (Enhanced Logic Analyzer? http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=6210&sid=8bbf73b) or better, would be better keep the max number of features as possible even if running towards the exhaustion of available memory.
Otherwise, you could remove something like the BASIC interpreter (I have read that it uses 29% of the available RAM's space) or problematic things like PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT that you wrote.
More, as it has from lately, perhaps PIC SUPPORT could be moved among extra features like STK500 and other things like that.
@mikebdp2
Okay I understand the facts about the i2c timeout problem, thanks!

tbox1911, 2016-09-22:
@agatti : thx a lot, I'm testing this version with i2c and SPi memory.
from now it seems to work nicely, I let's you know if I found something ;)

USBEprom, 2016-10-01:
Hi guys.
Following instructions in the wiki I have MPLAB X and MPLAB XC16 correctly installed.
I have tried to compile by myself a firmware like those named "bpv3_fw7.0_features" that are here: dangerousprototypes.com/forum/download/file.php?id=12142
I took the whole repository here:
https://github.com/BusPirate/Bus_Pirate/archive/master.zip
(I used a Windows XP computer)
I want 256 characters for the command line, so even if I am pretty babo, I understand I have to edit configuration.h, lines 496-505 in this way:
 
    /**
    * How big the serial command buffer can be, in bytes.
    *
    * @warning This must be set to a power of two, ie. 256, 128, 64, 32, etc.
    */
    #ifdef BUSPIRATEV3
    #define BP_COMMAND_BUFFER_SIZE 256
    #else
    #define BP_COMMAND_BUFFER_SIZE 256
    #endif /* BUSPIRATEV3 */
 
I have to set too lines 304-319 in this way:
 
#ifdef BUSPIRATEV3
#define BP_ENABLE_1WIRE_SUPPORT
#define BP_ENABLE_BASIC_SUPPORT
#define BP_ENABLE_DIO_SUPPORT
#undef BP_ENABLE_HD44780_SUPPORT
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#undef BP_ENABLE_PIC_SUPPORT
#undef BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#define BP_ENABLE_RAW_2WIRE_SUPPORT
#define BP_ENABLE_RAW_3WIRE_SUPPORT
#define BP_ENABLE_SMPS_SUPPORT
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */
 
By editing I can not compile the source due out of memory error even using option S for compression.
So I remove BASIC_SUPPORT by changing configuration.h, lines 304-319 in this way: 
 
#ifdef BUSPIRATEV3
#define BP_ENABLE_1WIRE_SUPPORT
#undef BP_ENABLE_BASIC_SUPPORT
#define BP_ENABLE_DIO_SUPPORT
#undef BP_ENABLE_HD44780_SUPPORT
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#undef BP_ENABLE_PIC_SUPPORT
#undef BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#define BP_ENABLE_RAW_2WIRE_SUPPORT
#define BP_ENABLE_RAW_3WIRE_SUPPORT
#define BP_ENABLE_SMPS_SUPPORT
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */
 
This time I can compile the source with options 0,1,2 and S.
Firmwares I have got from myself pass self test but some option does not work.
My goal is to put 256 characters for the command line in "bpv3_fw7.0_features.zip" that is here: dangerousprototypes.com/forum/download/file.php?id=12142
How can I do?
Is it possible that by providing 256 characters for the command line instead of 128 then the generated code is too big for Bus Pirate revision 3 so that I have to give up BASIC?
Maybe I am doing something wrong.
I would recompile that same firmware version for obtain 256 characters for the command line.
Here it is argued that the thing should be possible:
 
_it is possible to fit many features for BPv3 (including BASIC) but only for three opt levels:
opt0 - not enough program memory
opt1 - 84% occupied,
opt2 - 94% occupied, faster speed,
optS - 77% occupied, slower speed,
opt3 - not enough program memory

./Bus_Pirate/Firmware/configuration.h, lines 304-319

#ifdef BUSPIRATEV3
#define BP_ENABLE_1WIRE_SUPPORT // <--- + enabled
#define BP_ENABLE_BASIC_SUPPORT // <--- + enabled
#define BP_ENABLE_DIO_SUPPORT // <--- + enabled
#undef BP_ENABLE_HD44780_SUPPORT
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#undef BP_ENABLE_PIC_SUPPORT
#undef BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#define BP_ENABLE_RAW_2WIRE_SUPPORT // <--- + enabled
#define BP_ENABLE_RAW_3WIRE_SUPPORT // <--- + enabled
#define BP_ENABLE_SMPS_SUPPORT // <--- + enabled
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */

BP_ENABLE_HD44780_SUPPORT could also be enabled (takes just about 1% extra)

However, there are problems with BP_ENABLE_PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT._

Thanks.

mikebdp2, 2016-10-05:
@USBEprom , about your last message above, here is a reply - http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=65383#p65383
@USBEprom 
> Otherwise, you could remove something like the BASIC interpreter (I have read that it uses 29% of the available RAM's space) or problematic things like PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT that you wrote.

These two problematic options cannot be enabled currently for BPv3. They will not occupy any space at compiled firmwares until these features are fixed and enabled. Because of that and "256 length problem", I can't currently create any firmware better than what you already have ("features" builds)

agatti, 2016-10-06:
For the record, PIC and PC-AT build problems have been fixed.

USBEprom, 2016-10-09:
@agatti
I can confirm that BP_ENABLE_PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT now are fixed and do not give problems anymore.
Thanks!
 
@all
I just compiled a new firmware by myself.
Since BP_ENABLE_PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT now have been fixed by agatti I have included them.
Whising me 256 characters for the command line I changed line 502 in configuration.h, lines 496-505, in this way:
 
/**
* How big the serial command buffer can be, in bytes.
*
* @warning This must be set to a power of two, ie. 256, 128, 64, 32, etc.
*/
#ifdef BUSPIRATEV3
#define BP_COMMAND_BUFFER_SIZE 256
#else
#define BP_COMMAND_BUFFER_SIZE 256
#endif /* BUSPIRATEV3 */
 
Then in the same file I changed lines 304-319 o in this way:
 
#ifdef BUSPIRATEV3
#define BP_ENABLE_1WIRE_SUPPORT
#undef BP_ENABLE_BASIC_SUPPORT
#define BP_ENABLE_DIO_SUPPORT
#define BP_ENABLE_HD44780_SUPPORT
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#define BP_ENABLE_PIC_SUPPORT
#define BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#define BP_ENABLE_RAW_2WIRE_SUPPORT
#define BP_ENABLE_RAW_3WIRE_SUPPORT
#define BP_ENABLE_SMPS_SUPPORT
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */
 
Practically I have activated all the disponible options discarding BASIC because there is not enough space available in the memory of the PIC into the Bus Pirate v3.
I compiled with those settings and I got a firmware that seems to work.
It has 11 options in its menu:
 
Bus Pirate v3.5
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)
 
It passes the selftest, it has 256 characters for the command line, flashrom is working and all is fine for what I can understand.
Well!
Now, since in the SPI option (5) there are only the standard speed as follow:
 
(1)>5
Set speed:
 1. 30KHz
 2. 125KHz
 3. 250KHz
 4. 1MHz
 
I'd like to add others new because I need more high speeds.
If I am not wrong SPI's speed limits for Bus Pirate v3 are 8MHz for MASTER mode and 10MHz for the SLAVE.
So, if still I am not wrong, new speeds could be added as follows:
 
    0b00000000, /*  31 kHz - Primary prescaler 64:1 / Secondary prescaler 8:1 */
    0b00001100, /*  50 kHz - Primary prescaler 64:1 / Secondary prescaler 5:1 */
    0b00011000, /* 125 kHz - Primary prescaler 64:1 / Secondary prescaler 2:1 */
    0b00011100, /* 250 kHz - Primary prescaler 64:1 / Secondary prescaler 1:1 */
    0b00011101, /*   1 MHz - Primary prescaler 16:1 / Secondary prescaler 1:1 */
    0b00010110, /* 1.3 MHz - Primary prescaler  4:1 / Secondary prescaler 3:1 */
    0b00011010, /*   2 MHz - Primary prescaler  4:1 / Secondary prescaler 2:1 */
    0b00001011, /* 2.6 MHz - Primary prescaler  1:1 / Secondary prescaler 6:1 */
    0b00001111, /* 3.2 MHz - Primary prescaler  1:1 / Secondary prescaler 5:1 */
    0b00011110, /*   4 MHz - Primary prescaler  4:1 / Secondary prescaler 1:1 */
    0b00010111, /* 5.3 MHz - Primary prescaler  1:1 / Secondary prescaler 3:1 */
    0b00011011, /*   8 MHz - Primary prescaler  1:1 / Secondary prescaler 2:1 */
 
I need to change lines 67-71 of spi.c in order to accommodate the new speeds.
that is ok, but what about to change entries in order to activate them?
Seems to me the modify affects too many files, how can I do?
I do not even know if the modify I have in mind is possible with the Bus Pirate v3.
Maybe there isn't enought space to accommodate the new entries.
Thanks.

agatti, 2016-10-09:
@USBEprom Can you please track development of the SPI feature using https://github.com/BusPirate/Bus_Pirate/issues/25 instead?

USBEprom, 2016-10-09:
@agatti
**agatti wrote:
USBEprom Can you please track development of the SPI feature using #25 instead?**
 
Sorry sir, what do you mean?
Honestly I do not know how do it, I have no skills in that kind of things.
I am only capable to do some little testing for what are my knowledge.
However all the informations I have about what I wrote are in the datasheet of PIC24FJ64GA002, page 147 and 150, and here:
 
http://dangerousprototypes.com/forum/viewtopic.php?f=40&t=3864&start=15#p41505
 
I know nothing more, sorry.
 
@agatti
Since I am a Windows user and I am not familiar with Linux I could not use "git checkout spi_extra", so I used your bpv3_extra_spi.zip that you have sent at me.
For me it works fine.
 
~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.95) OK
VPU(5.03) OK
3.3V(3.31) OK
ADC(3.35) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on!
Any key to exit
Found 0 errors.
HiZ>#
RESET
 
Bus Pirate v3.5
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ UART I2C SPI] Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. UART
3. I2C
4. SPI
x. exit(without change)
 
(1)>4
Set speed:
 1.  30KHz
 2. 125KHz
 3. 250KHz
 4.   1MHz
 5.  50KHz
 6. 1.3MHz
 7.   2MHz
 8. 2.6MHz
 9. 3.2MHz
10.   4MHz
11. 5.3MHz
12.   8MHz
 
It pass selftest and all the new speeds I wrote are in there, only 50kHz setting s in the 5th position instead of the more logical 2th.
That is not a problem at all, I think it is only a typo.
Now I will try to get the new repository with your modifications in order to compile by myself a new firmware.
Thank you agatti, I hope what I just wrote can be useful.

agatti, 2016-10-09:
@USBEprom with "track development" I meant posting under the issue at ```https://github.com/BusPirate/Bus_Pirate/issues/25``` or just click on the #25 link in the message.

Regarding the order of entries, if we reorder things right now people who wrote scripts to automate certain operations (like, connect, send enter, send 'm' + enter, send '5' + enter, send '4' + enter, etc.)  will find their script behaving differently.  If both the console UI and the binary I/O protocol get redesigned, sure, things can be sorted up in a better fashion - but that's in the future and we plan to release an "official" 7.1 firmware image that will be a drop-in replacement for what is already available.

USBEprom, 2016-11-03:
Hi guys.
I am trying to understand whether it is possible or not to compile for the Bus Pirate revision 3 a fimware version that allows to simultaneously have up to 256 characters for the command line and BASIC scripting capabilities too.
I am not a software engineer but I understood that enabling 256 characters for the command line uses too much room in RAM so that BASIC interpreter environment can not be put together into the Bus Pirate revision 3.
A workaround I have found to keep the BASIC feature has been to reduce the length of the command line up to 32 characters only.
Now I wonder if is not possible to keep the command line up to 256 characters and reduce the space reserved to the BASIC scripting environment.
I know that BASIC feature normally allow up 64kB, but what about the possibility of reducing the addressable RAM for it to something smaller like 32kB, 24kB, 16kB, 10kB, 8kB, 4kB, 2kB or even 1kB or so?
Based on your opinion would it be possible?
If yes, how?
My goal would be to compile a firmware that allow for up 256 characters into the command line while maintaining the BASIC feature even if limited to a few kB only.
I have already looked into basic.c but I have understood nothing.
 
Thanks.

agatti, 2016-11-03:
@USBEprom you can try changing PGMSIZE in basic.h.  Right now it's set to take 1kb of RAM, if you lower it to 512 bytes then in theory you should be able to fit a 256 bytes commandline buffer.

USBEprom, 2016-11-03:
@agatti 
 
Thank you agatti!
For 256 bytes in command line I had to set 256 in line number 105 of basic.h so that it has changed like this:
 
#define PGMSIZE		256
 
Using 512 I can not reach the hexadecimal file not even using the "s" option while compiling because the file is over the limit of available RAM for the Bus Pirate revision 3.
However even if I have got the hexadecimal the BASIC feature does not work.
By issuing command "s" of the Bus Pirate I get the message "Syntax error at char 1".
I need to dig further, but I do not give up.
Thank you again sir!

USBEprom, 2016-11-04:
@agatti
 
Agatti as always you are the greatest, while I am the most babo in the whole world!
It does work exactly like you have written, it was that I am babo by putting together basic.h and configuration.h from different folders!
Using the correct ones it works!
Please forget my previous post, it is totally wrong, my fault, agatti was right already from the beginning!
I finally, digging for digging, dug my own grave in shame.
I am sorry I wasted your time writing not correct things.
However here you go what I did.
 
For 256 bytes in command line I had to set 512 into line number 105 of basic.h so that it has changed like this:
 
#define PGMSIZE 512
 
In this way I have got a firmware version for the Bus Pirate revision 3 where are allowed up 256 characters for the command line and all the possible features are enabled, even BASIC scripting, althought with reduced RAM room compared to standard release.
For me this is the "gold version" of the firmware for the Bus Pirate fixed for 2WIRE protocol and more.
Thank you agatti, thanks sir!

tbox1911, 2016-11-07:
@USBEprom ,
Glad to ear that good news,
is it possible to have a copy of your firmware , the Golden One !
( don't see any link, to download it)

I want to test this version
Big thanks

USBEprom, 2016-11-07:
Hi tbox1911.
All the firmwares I built I put them on the forum of dangerousprototypes.
However here you go what you are looking for:
 
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=45#p65498
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=45#p65500
 
Please note that the firmware compressed with option "1" in the first link have to be converted uppercase for its use because I forgot to did it when I made the archive.
If you can not do the conversion simply jump on the second link where is the same firmware already converted and ready to use.
All the firmware that I have compiled are the work of agatti and others that I thank for the hard work and skills.
Enjoy!

USBEprom, 2016-11-10:
@agatti

Hi sir.
You are the greatest and incredibly fast!
For me your changes work really fine.
Actually I do not use DIO mode in too complicated ways.
For me it was more of a cosmetic issue.
Now it works well for me, but of course I intend to deepen further.
I do not think, however, that problems will arise.
Here you go a snapshot of my terminal:

m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)

(1)>11
Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'

Ready
DIO>(0)
ERROR: command has no effect here

Syntax error at char 3
DIO><0>
1. <>
2. <>
3. <>
DIO>|0b100110
0x64 = 100 = 0b01100100
DIO>=0x7E
0x7E = 126 = 0b01111110
DIO>
 
Repeat, for me it is solved.
Now it behaves just as I had in mind.
Thank you agatti!!!

USBEprom, 2016-11-12:
Hi guys.
Today I am trying to build a new firmware release starting from the very latest cleaned up repository.
Because I want to compile for the Bus Pirate revision 3 a fimware version that allows to simultaneously have 256 characters for the command line and BASIC scripting capabilities I need to change basic.h putting #define PGMSIZE 512 instead of the standard #define PGMSIZE 1024 that there is.
It should be in line 105 or however somewhere still there in basic.h, but the tag #define PGMSIZE is not present anymore now.
Where is it?
How do I must do change for my purpose?
Thanks.
 
Then, out as simple curiosity.
In the recent past new speeds in the SPI menu have been added.
It was explained how to change spi.c for the new speeds as also the reason for which they have been ordered in a given way, but where the related messages are in the case someone wanted to order them in another way?
I do not know if something is different now with the new cleaned up repository.
Thanks.

agatti, 2016-11-12:
It has been moved to configuration.h (see also https://github.com/BusPirate/Bus_Pirate/blob/master/Firmware/configuration.h#L405).

USBEprom, 2016-11-12:
@agatti
 
Thank you sir, I get it now.
So I have changed lines 304-319 into configuration.h for setting the features I want.
 
#ifdef BUSPIRATEV3
#undef BP_ENABLE_1WIRE_SUPPORT
#undef BP_ENABLE_BASIC_SUPPORT
#undef BP_ENABLE_DIO_SUPPORT
#undef BP_ENABLE_HD44780_SUPPORT
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#undef BP_ENABLE_PIC_SUPPORT
#undef BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#undef BP_ENABLE_RAW_2WIRE_SUPPORT
#undef BP_ENABLE_RAW_3WIRE_SUPPORT
#undef BP_ENABLE_SMPS_SUPPORT
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */
 
I changed the line 405 too in this way
 
#define BP_BASIC_PROGRAM_SPACE 512
 
in order to make room for BASIC scripting.
 
Finally I changed line 523 from
 
#define BP_COMMAND_BUFFER_SIZE 128
 
to
 
#define BP_COMMAND_BUFFER_SIZE 256
 
in order to get up 256 characters for the command line.
While compiling I get these messages
 
"F:\MPLABX\XC16\bin\xc16-gcc.exe"   ../spi.c  -o build/BusPirate_v3/production/_ext/1472/spi.o  -c -mcpu=24FJ64GA002  -MMD -MF "build/BusPirate_v3/production/_ext/1472/spi.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3  -no-legacy-libc    -mlarge-code -mlarge-data -Os -I"../../microchip/include" -I".." -I"../translations" -msmart-io=1 -Werror -Wall -msfr-warn=off   -save-temps
"F:\MPLABX\XC16\bin\xc16-gcc.exe"   ../uart.c  -o build/BusPirate_v3/production/_ext/1472/uart.o  -c -mcpu=24FJ64GA002  -MMD -MF "build/BusPirate_v3/production/_ext/1472/uart.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3  -no-legacy-libc    -mlarge-code -mlarge-data -Os -I"../../microchip/include" -I".." -I"../translations" -msmart-io=1 -Werror -Wall -msfr-warn=off   -save-temps
../openocd.c:11:29: fatal error: bus_pirate_core.h: No such file or directory
compilation terminated.
make[2]: *** [build/BusPirate_v3/production/_ext/1472/openocd.o] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2
"F:\MPLABX\XC16\bin\xc16-gcc.exe"   ../openocd.c  -o build/BusPirate_v3/production/_ext/1472/openocd.o  -c -mcpu=24FJ64GA002  -MMD -MF "build/BusPirate_v3/production/_ext/1472/openocd.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3  -no-legacy-libc    -mlarge-code -mlarge-data -Os -I"../../microchip/include" -I".." -I"../translations" -msmart-io=1 -Werror -Wall -msfr-warn=off   -save-temps
nbproject/Makefile-BusPirate_v3.mk:464: recipe for target 'build/BusPirate_v3/production/_ext/1472/openocd.o' failed
make[2]: Leaving directory 'F:/MPLAB/Bus_Pirate-master_12112016/Bus_Pirate-master/Firmware/busPirate.X'
nbproject/Makefile-BusPirate_v3.mk:84: recipe for target '.build-conf' failed
make[1]: Leaving directory 'F:/MPLAB/Bus_Pirate-master_12112016/Bus_Pirate-master/Firmware/busPirate.X'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed
 
and the operation fails.
What is wrong?
I guess I am doing something bad but I am babo and I do not know what.
 
Also I know I am boring a lot, but I took a look inside configuration.h and I do not understand where are the messages related to speeds coded into spi.c.
Thanks.

agatti, 2016-11-12:
I fixed the build error with https://github.com/BusPirate/Bus_Pirate/commit/4f44d0883afa4d1ad3f8d6de1bcc4c27a2e17422, thanks for noticing.

If you want to reorder the SPI bus speeds, you have to edit the string files in tools/packstrings/bus_pirate_v?_strings.txt, regenerate the messages_v?.h/messages_v?.s running the python script in tools/packstrings, overwrite the message files in the Firmware directory with what you just generated, then you need to reorder the entries in spi_bus_speed inside Firmware/spi.c.  If you want to add more, things get a bit more complicated, and there are limits on how many speeds you can add before breaking the binary serial I/O protocol.

USBEprom, 2016-11-12:
@agatti
 
Thank you very much sir, now it work!
 
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=45#p65526
 
Thanks also for the explanations about what I asked.
These informations are useful to me because I am trying to figure out the matter so I can fend for myself without disturbing anyone.

USBEprom, 2016-11-13:
Hi guys.
Now something a bit hard that I know nothing if it is possible to do or not.
Something that would be an enhancement.
Perhaps this is not the right place to put it, however here is what it is.
Into the dangerousprototypes forum at this link it started a discussion about possible improvements for the Logic Analyzer side of the Bus Pirate (http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=6210&view=unread#p56802).
In the course of that discussion it was provided a working solution with these features:
 
a) sampling rates up to 16 MHz
b) additional samples when using fewer channels (up to 32k for 1 channel)
c) trigger location anywhere in the buffer
d) backward compatible with previous Logic Analyzer modes
 
Known Limitations:
 
-Selecting a trigger position of 100% will set the position at 0%. Workaround is to use 99% instead.
-Due to quirks at 4 and 8 MHz sampling, there is jitter in the samples but the long term rate is exact.
 For example, 4 MHz samples should be every 250 nanoseconds, but it is actually between 187 nanoseconds
 and 375 nanoseconds between samples.
-If compiled without optimization, 1 MHz sample timing is off prior to the trigger.
 
Possible Future Improvements:
 
-Allow selection of channel(s) to be recorded at higher sample sizes
 
[Enhanced Logic Analyzer (latest).zip](https://github.com/BusPirate/Bus_Pirate/files/587723/Enhanced.Logic.Analyzer.latest.zip)
 
All the improvements are into the SUMP.c that is inside the released package of this Enhanced Logic Analyzer release which has too the hexadecimal compiled with option 1 ready to use, the ols.profile-buspirate-enhanced.cfg file to put into the analyzer client and some instructions on how use the thing.
I wonder if could be possible to merge the content of the SUMP.c of the Enhanced Logic Analyzer into the one inside the current repository so that new features are available.
I know nothing but I understand this must to be very hard to reach.
Thanks.

agatti, 2016-11-13:
@USBEprom can you please create a new issue for this? I'm going to close this one as it is no longer for "test firmware on v3" anymore.

USBEprom, 2016-11-13:
@agatti 

Sorry sir.
You are right.
I edited my previous post.
I hope I have added the new issue in the proper way.
Apologize for having put it in the wrong place.
Thanks.

closed 2016-11-13 by agatti

elsni, 2016-12-21:
Just bought a new BP 3.6. It has Firmware v5.10 preinstalled, but unfortunately it has no JTAG support, so I need to flash a different one.

After following a bunch of red herring links wasting almost an hour, I finally managed to find pirate-loader and the 7.0 fw hex files for download.

Unfortunately I can't flash the fw. pirate-loader tells me "Checksum does not match, line 4" on every hex file. Converting an hex file to uppercase makes no difference. What am I doing wrong?
Thanks for any advice!

agatti, 2016-12-21:
@elsni can you please let us know which OS you're using and whether you built the pirate-loader binary yourself from this source code repo or if you're using a pre-built version that came with older firmware packages?

USBEprom, 2016-12-21:
@elsni

Do you mean also the factory ones by writing "on every hex file"?
The factory firmwares should already be in the correct format without any need of having to convert them, then at least those should work without problems.
What do you use to convert the hexadecimal content?
What is the operating system you are using?

elsni, 2016-12-22:
Hi,
Thank you @agatti and @USBEprom  for quick reply!
At fist I used Windows 10 and the precompiled binary from /jevinskie/bus-pirate and the hex files in that repository.
I tried again on my kubuntu machine but I compiled pirate-loader from source. That worked, I managed to flash fw 6.3 successfully from a hex file in that repository. I did not convert any binary file to hex by myself.
Unfortunately that version of 6.3 does have jtag support either. Which one can I use?

USBEprom, 2016-12-22:
Hi elsni.
It is weird that the firmware you have used (https://github.com/jevinskie/bus-pirate) does not support JTAG.
This because here:
 
http://dangerousprototypes.com/docs/Bus_Pirate#JTAG
 
it said that 6.0 and 6.1 firmwares do contain JTAG terminal mode, hence they have the binary JTAG mode.
However take a look at these:
 
http://dangerousprototypes.com/docs/Bus_Pirate_firmware_change_history
 
May be you must compile yourself the source in order to build your own firmware.
If for any reason you can not do that, try these that are ready to use:
 
http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=8637#p65660
 
I do not know if at least one of them allows for JTAG support because I have not seen their configuration setup.
Unfortunately, for what I have read, these repositories do not use configuration.h but something else to switch active features so I can not build from them new firmwares where the JTAG support is active.

[11 more replies]