ElderlyPirate/Bus_Pirate #10: Warning: busPirate.X.production.hex code overlaps current code and will overwrite it ?

mikebdp2, 2016-09-11:
What does this warning mean? I am trying to build a firmware from the latest revision of this repository. It builds successfully with the changes I have shared through my pull requests. However: 

1. it tells me this warning - regardless on what optimization level I use
2. if I ignore this warning and upload this firmware to my BPv4:

> HiZ>~
> Disconnect any devices
> Connect (ADC to +3.3V)
> Space to continue <--- it freezes here!

My build environment: Ubuntu Linux 16.04.1 x64, MPLAB X v3.40 and MPLAB XC v1.26 compiler with PRO mode enabled (60 days free evaluation license, so that I could enable "3" level optimization for speed) When I build a BPv4 bootloader it does not give me this warning, boots flawlessly and feels to be more stable than very old pre-installed bootloader

I suspect this problem is because of incorrect linker settings or caused by configuration of a linker script, some functions code is overlapping, but still could not figure out what stuff exactly is overlapping!!

Memory regions - beginning part of p24FJ256GB106.gld linker script:
```
MEMORY
{
  data  (a!xr)   : ORIGIN = 0x800,         LENGTH = 0x4000
  reset          : ORIGIN = 0x0,           LENGTH = 0x4
  ivt            : ORIGIN = 0x4,           LENGTH = 0xFC
  aivt           : ORIGIN = 0x104,         LENGTH = 0xFC
  program (xr)   : ORIGIN = 0x2000,        LENGTH = 0x289F8
  config4        : ORIGIN = 0x2ABF8,       LENGTH = 0x2
  config3        : ORIGIN = 0x2ABFA,       LENGTH = 0x2
  config2        : ORIGIN = 0x2ABFC,       LENGTH = 0x2
  config1        : ORIGIN = 0x2ABFE,       LENGTH = 0x2
}

__CONFIG3 = 0x2ABFA;
__CONFIG2 = 0x2ABFC;
__CONFIG1 = 0x2ABFE;

__IVT_BASE  = 0x4;
__AIVT_BASE = 0x104;
__DATA_BASE = 0x800;
__CODE_BASE = 0x200;
... ... ...
```

Below is a memory map info that I receive in the end of build process:
(seems normal on the first glance, but maybe it conflicts with a linker script?)
```
xc16-ld 1.26 (A)

Program Memory  [Origin = 0x2000, Length = 0x289f8]
section                    address   length (PC units)   length (bytes) (dec)
-------                    -------   -----------------   --------------------
.text                       0x2000               0x8a2           0xcf3  (3315)
.const                      0x28a2               0x5de           0x8cd  (2253)
.text                       0x2e80             0x13914         0x1d59e  (120222)
.dinit                     0x16794               0x280           0x3c0  (960)
.text                      0x16a14               0x7f6           0xbf1  (3057)
.isr                       0x1720a                 0x4             0x6  (6)
.shared.dinit              0x1720e                 0x2             0x3  (3)
                 Total program memory used (bytes):        0x1fb18  (129816) 52%

Ivt Memory  [Origin = 0x4, Length = 0xfc]
section                    address   length (PC units)   length (bytes) (dec)
-------                    -------   -----------------   --------------------
.ivt                           0x4                0xfc           0x17a  (378)

Aivt Memory  [Origin = 0x104, Length = 0xfc]
section                    address   length (PC units)   length (bytes) (dec)
-------                    -------   -----------------   --------------------
.aivt                        0x104                0xfc           0x17a  (378)

Data Memory  [Origin = 0x800, Length = 0x4000]
section                    address      alignment gaps    total length  (dec)
-------                    -------      --------------    -------------------
.bss                         0x800                   0          0x1004  (4100)
.bss.end                    0x1804                   0          0x1000  (4096)
.bss                        0x2804                   0           0x468  (1128)
.data                       0x2c6c                   0           0x210  (528)
usb_data3                   0x2e7c                   0           0x100  (256)
.bss                        0x2f7c                   0            0x76  (118)
.data                       0x2ff2                   0             0xe  (14)
usb_bdt                     0x3000                   0           0x200  (512)
.bss                        0x3200                   0           0x1a8  (424)
.bss                        0x33a8                   0            0x62  (98)
.data                       0x340a                   0            0x76  (118)
.bss                        0x3480                   0            0x7a  (122)
.bss                        0x34fa                   0             0xa  (10)
usb_data                    0x3504                   0             0xa  (10)
.bss                        0x350e                   0            0x30  (48)
.data                       0x353e                   0             0x4  (4)
.bss                        0x3542                   0             0x6  (6)
                 Total data memory used (bytes):         0x2d48  (11592) 70%

Dynamic Memory Usage
region                     address                      maximum length  (dec)
------                     -------                      ---------------------
heap                             0                                   0  (0)
stack                       0x3548                              0x12b8  (4792)
                 Maximum dynamic memory (bytes):         0x12b8  (4792)
```

agatti, 2016-09-11:
This is interesting.  I didn't remember getting that message on MPLAB-X 3.35 but I do get that on 3.40 on OSX.  The xc16 compiler is same as yours - however I build with 's' when it comes to optimisation settings and it works here.  Something that could be checked would be whether the linker script changed between MPLAB-X versions, which is something I'm going to do shortly after this.

In the meantime, can you please:

* disable all advanced optimisations (honestly, all you need to care for this one is code size, I think :)), try with modes '0', and 's' to see if the problem disappears.
* if not, uncomment BPV4_DEBUG in base.h, connect a UART-to-USB adapter to AUX1 and AUX2 and see what's going on there.

Thank you!

agatti, 2016-09-11:
Ok, linkerscripts differ.  A quick diff between p24FJ256GB106.gld in the source tree and what's in ..../xc16/v1.26/support/PIC24F/gld/p24F256GB106.gld shows a few changes.  I'll have to get out pen and paper to draw all the new sections and offsets to figure out if there are any significant differences, but I also need to trace the history of that file on the original DP repo to see if that got changed over time and if it was written from scratch or not.

agatti, 2016-09-11:
By the way, since I don't have a PicKit2 or anything like that - can you please clarify whether the firmware problem you see happens with the previous "official" bootloader as well?  Because that's what I am using right now.

agatti, 2016-09-11:
For what it's worth, here's how sections look like when compiled with no optimisations:

| Section name    | Start     | End       |
| --------------- | --------: | --------: |
| `.ivt`          | `0x00004` | `0x00100` |
| `.aivt`         | `0x00104` | `0x00200` |
| `.bss`          | `0x00800` | `0x01804` |
| `.bss.end`      | `0x01804` | `0x02804` |
| `.text`         | `0x02000` | `0x028A2` |
| `.bss`          | `0x02804` | `0x02C6C` |
| `.const`        | `0x028A2` | `0x02E82` |
| `.data`         | `0x02C6C` | `0x02E7C` |
| `usb_data3`     | `0x02E7C` | `0x02E7C` |
| `.text`         | `0x02E82` | `0x0F612` |
| `.bss`          | `0x02F7C` | `0x02FDE` |
| `.bss`          | `0x02FDE` | `0x02FFC` |
| `.bss`          | `0x02FFC` | `0x03000` |
| `usb_bdt`       | `0x03000` | `0x03200` |
| `.bss`          | `0x03200` | `0x033A8` |
| `.data`         | `0x033A8` | `0x0347C` |
| `.bss`          | `0x0347C` | `0x034A2` |
| `.bss`          | `0x034A2` | `0x034CC` |
| `.data`         | `0x034CC` | `0x034DA` |
| `.bss`          | `0x034DA` | `0x034E4` |
| `usb_data`      | `0x034E4` | `0x034EE` |
| `.bss`          | `0x034EE` | `0x03508` |
| `.bss`          | `0x03508` | `0x03518` |
| `.data`         | `0x03518` | `0x0351A` |
| `.bss`          | `0x0351A` | `0x03520` |
| `.dinit`        | `0x0F612` | `0x0F8CA` |
| `.text`         | `0x0F8CA` | `0x107D2` |
| `.isr`          | `0x107D2` | `0x107D6` |
| `.shared.dinit` | `0x107D6` | `0x107D8` |

mikebdp2, 2016-09-11:
Hi there! I just tested the same firmware hex with a previous bootloader, and this problem didn't disappear. If I install my bootloader together with bp4.0-fw-7.0.hex firmware, BPv4 passes a self test without any errors/freezes and works great with a flashrom. So this problem is definitely caused by a bad firmware hex, not by bootloader...

About '0' and 's' optimization levels: have not tested them yet on a hardware, because they were giving me the same warning - so I thought they wouldn't work as well. Would test them soon...

Could you please confirm that your new builds at 's' level and MPLAB v3.40 are working , despite you are now getting the same warning message as me?

agatti, 2016-09-12:
I can reproduce your issue only with 's' optimisation level, which is kind of weird as it used to work (I did tests with that optimisation level on in the past), whilst with '0' it works as intended.

Now, it is my understanding that with a PicKit you can set hardware breakpoints and the like - if so, would you be so kind to break in when the problem occurs and get a bit more information so I can make a more educated guess on what's happening?  In the meantime I'll try to use the xc16 linkerscript rather than the one coming with the original DP's source tree and see if that helps somehow.

agatti, 2016-09-12:
Well, with the xc16 linkerscript the message does not show up anymore, however USB support is gone - and therefore the chances for me to debug this further until I purchase a PicKit :/  Sigh.

agatti, 2016-09-12:
The linkerscript in the repo has been modified to move the code out of the bootloader starting point so that the 'go to bootloader' command actually works - config4 is the address of the bootloader jump vector (also named BLJUMPADDRESS in base.h), and program starts at 0x2000 instead of 0x200.

So far the section start/end addresses look OK to me, but there's something left to try when I get my hands on a PicKit: use the linkerscript from xc16 and change the start/length of the program section to ORIGIN = 0x2000 / LENGTH 0x289F8 and set __CODE_LENGTH to 0x289F8, then see if it actually boots, as that does not trigger the overlap warning.

mikebdp2, 2016-09-12:
Sorry for delay, this is very busy week for me, I will return to this problem as soon as possible (in a couple of days hopefully, so curious to try to solve this problem). _About Pickit: if you are going to get it be careful, on aliexpress for example there are both pickit 2 and pickit 3 for about the same price; pickit 3 could appear to be better because of a higher version number but in reality it sucks in a big way_

agatti, 2016-09-13:
I managed to recover my BPv4 without a PicKit, so I was able to give my earlier idea a shot.  Turns out it works just fine with the '0' optimisation level and does not show any warning at all whilst linking.  However, the issue you originally mentioned never went away.  I'll close this and create a new entry that more appropriately describes the issue, if you don't mind.

agatti, 2016-09-13:
Original bug report resolved with 354598936db1ec455b73e2ce9ab866e0dd48dea9

agatti, 2016-09-13:
Moved to #11

closed 2016-09-13 by agatti