ElderlyPirate/Bus_Pirate #16: DIO does not work

agatti, 2016-09-20:
From the report on #7:

>Seems that DIO does not work.
```
Bus Pirate v3.5
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE DIO] Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. DIO
x. exit(without change)
(1)>8
ERROR: command has no effect here
Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'
Ready
Syntax error at char 2
DIO>
```

agatti, 2016-09-20:
Marking it as v4 too as it was not tested on such hardware yet.

agatti, 2016-10-06:
Ok, this is due to protocol_start triggering an error message, since DIO does not need to perform any action at the beginning.  Besides that, the read/write commands work as expected.

Guess DIO's protocol_start callback will be silent then...

USBEprom, 2016-11-07:
What about how to make silent the "protocol_start callback" of DIO?
I have tested how behave the stock firmware v5.10 while DIO is used.
Here you go:
 
#
RESET
 
Bus Pirate v3b
Firmware v5.10 (r559)  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. LCD
9. DIO
x. exit(without change)
 
(1)>9
Ready
DIO>(0)
ERROR: command has no effect here
 
Syntax error at char 3
DIO><0>
1. <>
2. <>
3. <>
4. <>
5. <>
DIO>|0b100110
0x64 = 100 = 0b01100100
DIO>=0x7E
0x7E = 126 = 0b01111110
DIO>
 
How it is possible to see the DIO protocol start without any particular message, it is totally silent.
By inputting not provided instructions, for instance (0) macro list,  there is issued the warning message "ERROR: command has no effect here".
The read/write commands work as expected whitout that any error message is issued.
 
Instead this is how the firmware v7.0 behaves:
 
#
RESET
 
Bus Pirate v3.5
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)
 
(1)>11
ERROR: command has no effect here
Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'
 
Ready
Syntax error at char 2
DIO>(0)
ERROR: command has no effect here
 
Syntax error at char 3
DIO><0>
1. <>
2. <>
3. <>
DIO>|0b100110
0x64 = 100 = 0b01100100
DIO>=0x7E
0x7E = 126 = 0b01111110
DIO>
 
The only remarkable difference is the "protocol_start callback" at the beginning while DIO starts.
I have took a chance to get a look at dio.c an dio.h but I can understand where  is taken the start callback.
Perhaps it would be better to turn off that message of start callback.
Thanks.

agatti, 2016-11-10:
The code involved for such a thing is in bus_pirate_core.c (check the definitions marked with `#ifdef BP_ENABLE_DIO_SUPPORT`).  I'll see what I can do for this, but no real guarantees - the code handling the interactive prompts is in dire need of a cleanup...

agatti, 2016-11-10:
A potential fix has been committed with 77adba645b34b5698293308d77c5562439bcd9b0 - @USBEprom can you please give this one a try and see if it fixes your problem?  It does here, but maybe you're using DIO mode in more complicated ways than I just tested.

agatti, 2016-11-10:
Closing as per https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-259786120

closed 2016-11-10 by agatti