ElderlyPirate/Bus_Pirate #18: I2C ACK/NACK are delayed

agatti, 2016-09-20:
From http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=8546 mentioned in #7:

>hello,
>i have a big problem using i2c whit the buspirate.
>it seems there is a long delay after a start condition and any ACK/NACK.
>IT IS ALMOST 2mS long.
>and it is getting even longer when the baud rate of the buspirate is set to a lower rate.
>normal i2c devises do not seem to care, sofar i could test it.
>but i am trying to talk whit a smbus device, and that one does not like those long timeouts.
[image]

agatti, 2016-11-01:
This might be related to #23

ChristopherSamSoon, 2018-11-29:
This issue is because the Buspirate prints out the I2C events on the UART in a "blocking" way. 

Example of an I2C write sequence:

1) Buspirate puts start condition on I2C line
2) Buspirate prints "I2CSTART" on the UART 
3) Buspirate puts data on I2C line
4) Buspirate prints data in ASCII on the UART 
5) Buspirate ACKS/NACKS

All of these printouts to the UART are "blocking" operations and buspirate will pause I2C communications during these events.

Best

Christopher Sam Soon

tudstudent, 2026-10-24:
Re-checked on the current firmware (v8.0 fork tree with the recent delay-path fixes; Owon scope on the Bus Pirate's own SCL/SDA pins; software-I2C mode, no slave attached - the master's own timing is what this report is about, since the acknowledge slot is clocked either way; repeated queued writes produce back-to-back transactions).

| software I2C speed | SCL half-period | longest gap, anywhere in the capture |
|---|---|---|
| 1 (approx. 6 kHz profile) | 82.6 us (6.05 kHz) | **241 us** (write-to-write turnaround) |
| 2 (approx. 48 kHz profile) | 10.4 us (48.1 kHz) | **154 us** (write-to-write turnaround) |

Within each byte every clock slot - the eight data clocks and the ninth acknowledge slot - is clocked at the profile's own cadence; the only longer gaps sit *between* consecutive write commands, i.e. the terminal printing the previous result and fetching the next one. No delay anywhere near the reported ~2 ms appears at either speed.

The reported "almost 2 ms, and longer at a lower serial baud" is consistent with the result-line print time of the older versions: the ACK/NACK text is printed between transactions at the configured serial rate (a handful of characters is milliseconds at low baud on a v3, hundreds of microseconds at 115200), which is exactly the growing gap the scope shot shows after the acknowledge. Normal I2C slaves ignore idle time between transactions; the SMBus device in the report was objecting to that idle period.

So: not reproducible on the current tree, with numbers - and the original mechanism is terminal output pacing, not the I2C bit timing.