ElderlyPirate/Bus_Pirate #19: BPv3: can't enable BP_ENABLE_PIC_SUPPORT and BP_ENABLE_PC_AT_KEYBOARD_SUPPORT

mikebdp2, 2016-09-20:
When I define BP_ENABLE_PIC_SUPPORT and/or BP_ENABLE_PC_AT_KEYBOARD_SUPPORT, I get many messages like `BPMSG1237 undeclared (first use in this function)`. This happens because messages_v4.h/messages_v4.s have more messages than messages_v3.h/messages_v3.s

I tried replacing v3 message files with v4 to add the missing messages. However there are new errors:

```
build/BusPirate_v3/production/_ext/1472/openocd_asm.o(.text+0x1e): In function `.L0':
: Link Error: Cannot access symbol (_UART1RXRecvd) with file register addressing.
Value must be less than 8192. Suggest large-data model.
build/BusPirate_v3/production/_ext/1472/openocd_asm.o(.text+0x60): In function `.L0':
: Link Error: Cannot access symbol (_UART1TXAvailable) with file register addressing.
Value must be less than 8192. Suggest large-data model.
build/BusPirate_v3/production/_ext/1472/openocd_asm.o(.text+0x68): In function `.L0':
: Link Error: Cannot access symbol (_UART1TXAvailable) with file register addressing.
Value must be less than 8192. Suggest large-data model.
build/BusPirate_v3/production/_ext/1472/openocd_asm.o(.text+0x6c): In function `.L0':
: Link Error: Cannot access symbol (_UART1TXBuf) with file register addressing.
Value must be less than 8192. Suggest large-data model.
make[2]: *** [dist/BusPirate_v3/production/busPirate.X.production.hex] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2

```

agatti, 2016-09-20:
The strings packer should be updated in order to allow sections to be included or excluded from a message file.  Right now that's a bit complicated to pull off and integrate with MPLAB X (as in, I have no clue on how to introduce external stages in the build process).

For what it's worth, those message strings were not removed, but were simply missing.  There are two possible choices at the moment:

* Disable PIC and AT Keyboard support for v3.
* Add the missing strings from messages_v4.txt to messages_v3.txt, rebuild the message files, and try again.

agatti, 2016-10-06:
PC-AT Keyboard support is added with b19edbe4d186e31e0dc06bea6b8f6f2a362def4a

agatti, 2016-10-06:
And PIC support is added with 8e9ad1c72f2a197214da83bce0c9b48aff4daa95

closed 2016-10-06 by agatti