ElderlyPirate/Bus_Pirate #20: Update the messages system to add conditional string inclusion

agatti, 2016-09-20:
As mentioned in #19, the string packer tool right now brings everything in, regardless of the configuration settings being chosen.  Something smarter would be to update the string packer to read a series of flags from the command line and selectively include/exclude strings from being merged in the .s file and referenced in the .h.  A quick and dirty approach for this would be to include the #defines in the .txt and let them pass through in the generated .h/.s files.

agatti, 2016-11-11:
I finally found out how to achieve this.  The problem is to be able to do it on a cross-platform fashion (a.k.a. "how about windows?").  So, changing `/Firmware/BusPirate.X/Makefile` allows for pre/post compilation steps so the string files can be generated upon changes in `configuration.h`, but this requires having Python installed and introduce horrible workarounds for having one single tree that builds on all platforms with no hassles once checked out.

The alternative would be to rewrite the string packer in C and supply a pre-built version for each platform along with the rest of the source but that's my least favourite option, if I have to be honest.

agatti, 2016-11-11:
On second thought, embedding the ifdef/endif markers in the input string file and generating the appropriate #ifdef/#endif and .ifdef/.endif blocks in the output files might be an option.  Let's see...

andersm, 2017-07-20:
If each string was placed in a separate section, the linker garbage collector could strip unused strings automatically.

agatti, 2017-07-20:
True, but then there would be some ROM space being wasted for section alignment, wouldn't it? My understanding is that by using a preprocessor the strings would be packed together and those bytes would still be available for code/other read-only data.

andersm, 2017-07-20:
It could probably be worked around using section attributes, but it looks like XC16-as has an extension that makes it much simpler:

> .section name [, attr1[,...,attrn]]
Assemble the following code into a section named name. If the character * is specified for name, the assembler will generate a unique name for the section based on the input file name in the format filename.scnn, where n represents the number of auto-generated section names.
Sections named * can be used to conserve memory because the assembler will not add alignment padding to these sections. Sections that are not named * may be combined across several files, so the assembler must add padding in order to guarantee the requested alignment.

andersm, 2017-07-20:
It seems that XC16 can't garbage collect __pack_upper_byte character arrays, so attempt 1 was a failure.

andersm, 2017-07-23:
Program memory usage
v3: 55479 -> 53229 (-2250 bytes)
v4: 89496 -> 91185 (+1689 bytes)
That is probably the section alignment at play. The more used strings, the higher the overhead.

One thing that came to mind is that it may be cheaper to use NULL-terminated strings than passing it as a parameter.

PR #60 (2017-07-29): Use linker garbage collection to remove unused strings
Change string packer to place each string in a separate section.
Change message functions to take a pointer instead of an offset.
Enable linker garbage collection in project settings.

Fixes #20
commit 6cea4615 (2017-07-29): Use linker garbage collection to remove unused strings (#60)

Change string packer to place each string in a separate section.
Change message functions to take a pointer instead of an offset.
Enable linker garbage collection in project settings.

Fixes #20
closed 2017-07-29 by agatti