ElderlyPirate/Bus_Pirate #25: Investigate adding new bus speeds to the SPI module

agatti, 2016-10-09:
As mentioned in https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-252490400 - the Bus Pirate can, in theory, have more than just the four SPI speeds it supports at the moment.  This task is supposed to track development for such a thing.

- [ ] Read datasheets for v3 (PIC24FJ64GA004) and v4 (PIC24FJ256GB110) to get the appropriate values for the two prescalers.
- [ ] Add the entries in the SPI module and get them accessible via the serial console.
- [ ] Make the entries accessible via Binary I/O mode.
- [ ] Test on v3.
- [ ] Test on v4.

agatti, 2016-10-09:
@USBEprom - can you please try out the version that is in the spi_extra branch?  To switch to the "spi_extra" branch, check out the code from GitHub to a new directory (just to be sure), and then once you are inside the new source code directory, please run this command: ```git checkout spi_extra```, then please rebuild as normal and see if it works as intended with the extended bus speeds.

agatti, 2016-10-09:
Or, if you get confused with Git and such, download one of the sample firmware builds from here:

[bpv3_extra_spi.zip](https://github.com/BusPirate/Bus_Pirate/files/518415/bpv3_extra_spi.zip)
[bpv4_extra_spi.zip](https://github.com/BusPirate/Bus_Pirate/files/518416/bpv4_extra_spi.zip)

**Caveat Emptor:** those builds have not been tested on real hardware besides making sure that speeds can be chosen correctly.  I still strongly suggest you to rebuild from source and flash that code instead.

agatti, 2016-10-09:
Seems to work on v3 according to https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-252505235

USBEprom, 2016-10-09:
@agatti
agatti wrote:
Caveat Emptor: those builds have not been tested on real hardware besides making sure that speeds can be chosen correctly. I still strongly suggest you to rebuild from source and flash that code instead.
 
Sorry to keep bothering you, seems I did a kinda mess by replying in the wrong place.
The test firmware you wrote is functioning well, I reconfirm here in the right place.
However I want to do what you wrote, only I need the new repository.
I am searching it, just I find it I will compile a new firmware from myself.
Where exactly is it?
Apologize and many thanks

agatti, 2016-10-28:
@USBEprom can you please let me know about the higher SPI speeds concerns expressed in https://github.com/BusPirate/Bus_Pirate/issues/25#issuecomment-252830924 so I can either drop the branch or merge it in?

USBEprom, 2016-10-28:
@agatti
Really sorry my bad.
I could not figure out where the new files were.
I am really babo, thank you agatti for them!
Now that I have your modified spi.c I built some new version up 256 characters command line and all the features but no BASIC and up 32 characters command line and all the features, BASIC too.
For what I can see your improvements work well, any problem from my point of view.
I believe you can merge all them into standard release.
Still apologize me for the late, fault of mine as I just said.
Many thanks

USBEprom, 2016-10-30:
@agatti

A friend of mine has just checked with his logic analyzer the correctness of the new speeds in SPI menu.
Seems that from 1 to 7 all it is OK, but starting from 8 until 12 do not work.
Briefly:
 
Set speed:
 1.  30KHz OK [old]
 2. 125KHz OK [old]
 3. 250KHz OK [old]
 4.   1MHz OK [old]
 5.  50KHz OK [NEW]
 6. 1.3MHz OK [NEW]
 7.   2MHz OK [NEW]
 8. 2.6MHz (DOES NOT WORK) [NEW]
 9. 3.2MHz (DOES NOT WORK) [NEW]
10.   4MHz (DOES NOT WORK) [NEW]
11. 5.3MHz (DOES NOT WORK) [NEW]
12.   8MHz (DOES NOT WORK) [NEW]
 
Perhaps I was wrong to calculate the values for the prescalers, I must check.
The logic analyzer is in the position to acquire up 24MHz so it should not have problem with those speed.
However that friend of mine said me he will check with an oscilloscope.
Sorry.
 
Duty cycle is not always 50% or near, but I guess that is pretty normal, even because using the FLASHROM program 8MHz speed behaves in the same way.

USBEprom, 2016-10-31:
@agatti
 
The usual friend of mine has rechecked with an oscilloscope and actually them all are functioning well.
It has been a fault of his cheap logic analyzer which can manage only 5V and 3,3V levels while moving up in frequency the amplitude of the signal generated by the Bus Pirate is below its acquisition threshold.
So in the end all the entries into the menu are working in the correct way as expected.
 
Set speed:
1. 30KHz OK [old]
2. 125KHz OK [old]
3. 250KHz OK [old]
4. 1MHz OK [old]
5. 50KHz OK [NEW]
6. 1.3MHz OK [NEW]
7. 2MHz OK [NEW]
8. 2.6MHz OK [NEW]
9. 3.2MHz OK [NEW]
10. 4MHz OK [NEW]
11. 5.3MHz OK [NEW]
12. 8MHz OK [NEW]
 
Duty cycle works correctly too.
In some ranges it is far from 50% but that is pretty normal.
It behaves exactly as is to be expected, no problem.
One more time I repeat your improvement is 100% good and functioning.
Thanks!

agatti, 2016-11-01:
@USBEprom Ok, thank you for checking - I'll merge this soon!

agatti, 2016-11-01:
Closed with cf293ece66ff76cf9cb6edceae0f3a5ad28bc579

closed 2016-11-01 by agatti