ElderlyPirate/Bus_Pirate #26: Bootloaders and the pirate-loader don't agree on bootloader address

blacktigersoftware, 2016-10-28:
Okay, so I'm having issues with flashing my BP3.  Of course, the PGC/PGD trick works, but I'm trying to do this automated from with an app.   The pirate-loader does the following in "fixJumps" sets the jump address to 0xa800, which is the beginning address of the DS30 bootloader.  However, the DS30 bootloader  sets a vector at 0xabf8  (the last 8 bytes of program memory).

Which should be used?   

As an aside, I can't compile the DS30 loader to save my life.  The linker does not want to put 6 bytes at 0xabf8....  I'm currently using ASM30 (v3_31), can not tell the linker version.... :<

Finally, the pirate-loader fails on all the 7.00 fw, it appears to be about the last page erase.  

Want all these as separate issues????

agatti, 2016-10-28:
No need to split issues, no worries.  I'll look into that, but I need more information regarding the flashing failure.  That also occurs on v4, and it is not an error on that revision.  I seem to recall that v3 should it be unaffected as well.

So, would it be possible to post the #defines set in configuration.h that apply to your particular firmware (if you are building your own), or where the firmware comes from if it is a prebuilt binary, and the output of pirate-loader with the --verbose flag on, otherwise it'd be mostly speculation on my part rather than just debugging :)

For the flashing vector address, I'll look at that tomorrow morning and hopefully I'll get back with an answer.

Thank you for the report!

agatti, 2016-10-28:
Incidentally, you may want to check out commit d357eee545b85a86206af7d4c2e9fb458da3bd37 - I am able to build a 4.5 bootloader for BPv3 using MPLAB X that way.

agatti, 2016-10-28:
Anyway, since I'm still awake... The pirate-loader binary is right in performing the fixup: in devices.inc, if you look the FLASHSIZE macro for the PIC24FJ64GA002 you'll see it's set to 0xAC00, then in ds30loader.s, STARTADDR is set to (FLASHSIZE - (2 * PAGESIZE)). A PIC24 code page is 512 instructions big, therefore you get 0xAC00 - (2 * 512) = 0xA800.

Hope that helps.

blacktigersoftware, 2016-10-29:
agatti --- thanks, the bootloader 4.5 works great.  I put the r2151 flash on it.  I'm back after 2+ years and want to finish my pirate app.  Starting on the BP3, but got a BP4 laying around here somewhere.  I think it is important to change the app on the fly without having to hardwire PGC/PGD.  The bootloader 4.5 should allow you to exit after a flash and do a reset by sending the '#' command.  Something other versions don't support.

[image]
[image]

blacktigersoftware, 2016-10-29:
just a quick follow up.  For some reason the BP 7.0 flashes attempt to write into the bootloader area.  I modified my flasher to log the error but roll on.  I think it has to do with the reference to 'abf8'.  The linker thinks that since we are referring to that address, we must have put something there.  The linker has an option to fill flash, but not to ignore flash..... go figure. .....
-------------------------
Writing page 41 row 335, a780...OK
Erasing page 42, a800...Protection Error [50]
Protection error
NOT Writing page 42 row 336, a800...OK
NOT Writing page 42 row 337, a880...OK
NOT Writing page 42 row 338, a900...OK
NOT Writing page 42 row 339, a980...OK
NOT Writing page 42 row 340, aa00...OK
NOT Writing page 42 row 341, aa80...OK
NOT Writing page 42 row 342, ab00...OK
NOT Writing page 42 row 343, ab80...OK
Firmware updated successfully

closed 2016-10-29 by blacktigersoftware

USBEprom, 2016-11-13:
I know I am babo as hell.
I have a Bus Pirate v3.6 whic has bootloader 4.4.
Can I use BPv3-Firmware-v4.5.hex (\Bus_Pirate-master\BPv3-bootloader\v4xtov43-bootloader-update\BPv3-Firmware-v4.5.hex) inside the v7.0 firmware repository in order to upgrade my Bus Pirate?
The error in the ending while I am uploading firmware into the device annoys me too much.
I know this is already closed, so apologize me for the hijacking.
Thanks!

blacktigersoftware, 2016-11-14:
The error comes from pirate-loader, and will still come from pirate-loader after upgrade.  The only advantage comes by being able to exit boot-loader by sending a command.  The pirate-loader for the BP4 logs this better.  The real deal is that the hex file has commands which would overwrite the bootloader and so it complains with a 'P' error result (presumably 'P' means protected).

USBEprom, 2016-11-14:
Ok, thanks!
So it is possible to use the new BPv3-Firmware-v4.5.hex with my Bus Pirate v3.6 instead of the stock bootloader 4.4 but however still it does not fix the problem of the error at the end while upgrading with Pirate-Loader.
The only advantage would be to be able to exit from the bootloader state by sending a command instead of having to unplug and replug the Bus Pirate.
I get it now.
Thank you sir