ElderlyPirate/Bus_Pirate #27: Add the rest of protocol documentation in markdown format.

agatti, 2016-11-01:
Work had been started to migrate protocol usage documentation from Dangerous Prototypes' wiki to a set of markdown pages to be put in the repo, there are a few more to be migrated.

agatti, 2016-11-01:
@kallisti5 do you have the time to pick this up and finish the documentation migration?

brimston3, 2016-11-15:
DIO - Needs an expert. The wiki refers users to the binary protocol mode and I'm not sure if it dumps you into binary mode from here or what.
PIC and HD44780 appear to require hardware support, without which, they are nonfunctional.
JTAG - superseded by openocd jtag mode.

I'll give the last 3 a shot (not DIO) today or tomorrow, but they'll likely need a lot of cleanup.

agatti, 2016-11-15:
PIC needs hardware support only in certain cases, according to http://dangerousprototypes.com/docs/Pirate_PIC_Programmer ("It currently supports a limited number of PICs while other features are stabilized. Some PICs require a 6-13volt to enter programming mode, these will require an adapter.").  But to be honest I don't know whether this remark is still valid or not.

JTAG technically is still available also without OpenOCD by using the binary protocol, by the way.  The OpenOCD interface is what's missing on v4 at the moment, but everything else is shared (in fact, v4 can also handle XSVF files and flash CPLDs/FPGAs).

DIO is basically a way to interactively mess with the pins from the serial command line.  When you issue a read command you'll get the pin state bitmask straight from the PORTB register: "000abcde" where: a is the AUX pin, b is the MOSI pin, c is the CLK pin, d is the MISO pin, and e is the CS pin, where 1 means the pin is HIGH and 0 means the pin is LOW.  Issuing a write with bit #9 set to 0 will set the lowest 5 bits of the value you're writing to the abovementioned pins, but if bit #9 is set to 1 it will set the direction of those pins using the same bitmask assignment, where 1 is INPUT and 0 is OUTPUT.

So, don't worry and take your time.  Plus, I'd like to thank you again for your valuable contribution!

brimston3, 2016-11-19:
If the PIC has LVP disabled (MCHP ships recent devices with LVP enabled), it does need HV on MCLR. The HVP adapter will not work with a BPv4+ without a pinout adapter.

agatti, 2016-11-20:
Fair enough, I guess that pretty much the rest of DP's wiki needs to be brought up to date then.

agatti, 2017-04-17:
All existing protocol documentation migrated with 14d6e598a4e3b1d1125446cccf45167321eaa208.

closed 2017-04-17 by agatti