ElderlyPirate/Bus_Pirate #28: Add OpenJTAG support for v4

agatti, 2016-11-01:
OpenJTAG support is currently only enabled for v3.  There were some earlier attempts at porting said code to v4 but nothing stable ever came out of it, as far as I know.

kallisti5, 2016-11-11:
I remember trying to get JTAG going on the BPv4 a few years back... I thought it worked,  but then I got a FlySwatter 2. Must of not worked very well :-|

csholmq, 2016-12-26:
I've gathered only v3 is supported by OpenOCD. But now with discontinued firmware by DP and compilation errors in the community build, it would be super if support could be added for the v4!

agatti, 2016-12-28:
@csholmq indeed, I believe most BP users got theirs mostly for JTAG support. It'd be a bummer if v4 could not handle it...

csholmq, 2016-12-28:
Is OpenOCD supported through binary mode?

agatti, 2016-12-28:
That is the only way to access OpenOCD actually.  When OpenOCD starts, it sends a particular command sequence to the serial port that first puts the BP into bitbang I/O mode (20 x 0x00) and then sends the proper command to enter OpenOCD mode (0x06).  There is no way to access that from the serial command interface (wouldn't be that much useful after all).

agatti, 2016-12-28:
By the way, that's on v3.  On v4 sending that command sequence will do nothing, which is what I plan to fix in the near future :)

valholl, 2017-11-19:
I see that there is no update on this issue since last year: are there news?

I need a JTAG programmer and was considering buying a bus pirate, and I wanted to know if I should buy a v3 or if in the month or so that international shipping can take I'm going to read that support has been added for v4 and regret my purchase :)

agatti, 2017-11-20:
Hi there, unfortunately your best bet is to get a v3 right now.  OpenOCD for v4 is scheduled for v7.2, which is scheduled sometime early next year at least.  Right now we're stuck in figuring out the best way to provide a smooth upgrade path from the old firmwares to this one, and that seems to be tricky to handle in an optimal way...

agatti, 2018-07-02:
Well, this took some time but can some of you folks who asked for OpenOCD support on v4 check whether [this branch](https://github.com/BusPirate/Bus_Pirate/tree/openocd-v4) does work for you?

agatti, 2018-07-04:
Ok, this just got merged - please give mainline a shot instead so maybe more people might be able to try this out.  I'll close this for the time being, please re-open another bug if the code doesn't work for you.

closed 2018-07-04 by agatti

robsoncouto, 2018-10-01:
Hello
First of all, thanks for all your work into this project.

I wanted to try this but could not get it to work. Could you kindly share your configuration script for the Bus Pirate v4?
I believe I installed the firmware correctly. The interactive terminal (?) says:
```
HiZ>i
Bus Pirate v4
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB PIC DIO]
DEVID:0x1019 REVID:0x0004 (24FJ256GB106 UNK)
http://dangerousprototypes.com
HiZ>
```
When trying to run openocd, i get the following:

```
$ openocd -f /usr/local/share/openocd/scripts/interface/buspirate.cfg -f nuv.cfg 

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (buspirate)
The following debug interfaces are available:
1: ftdi
2: usb_blaster
3: presto
4: usbprog
5: openjtag
6: jlink
7: vsllink
8: rlink
9: ulink
10: arm-jtag-ew
11: hla
12: osbdm
13: opendous
14: aice
``` 

I have used openocd before with a ft2232 board (not seriously, just some firmware dumps), but this is the first time I use a Bus Pirate for anything.
I appreciate any help with this :)

JarrettR, 2018-10-01:
Have you compiled OpenOCD with the `--enable-buspirate` flag?

I assume it still works something like this:

http://dangerousprototypes.com/docs/Gonemad%27s_Bus_Pirate/OpenOCD_walk_through

robsoncouto, 2018-10-01:
Thanks for your quick answer and sorry for the trouble!
I was just editing my post, but will reply here.
Yes, just got it to work with the flag. Just was not aware I had to reconfigure openocd. Now it recognizes the cpu.

Still, may I ask another question? It is possible to use the extra auxiliary pins on the v4 to drive the reset signals? If I understand correctly, the v3 can drive SRST or TRST only?

Thanks again

robsoncouto, 2018-10-02:
Hello. Thanks for your reply!
No, just SRST and TRST. The board I am trying to debug has both and was not being able to halt the cpu previously. I will try it out with both signals and the updated config file as soon as possible.

Thanks again

tudstudent, 2018-11-06:
Could someone clarify if JTAG is now working correctly on V4, on this website it states it cannot:
http://dangerousprototypes.com/docs/Bus_Pirate_v4_vs_v3_comparison

On this mentioned website:
*What doesn't work on Bus Pirate v4 right now:
**-     OpenOCD JTAG mode**
-     STK500v2 AVR programmer
-     XSVF player JTAG programmer