ElderlyPirate/Bus_Pirate #29: Add native SWD (ARM Serial Wire Debugging) support

agatti, 2016-11-01:
A script-driven approach has been tried here: http://www.willdonnelly.net/blog/bus-pirate-serial-wire/ .  Something worthwhile would be to reimplement this natively without the need for bit-banging data lines.

agatti, 2016-11-01:
An very simple initial implementation for synchronous SWD communication can be found here: https://gist.github.com/agatti/a92574bf726706f77fa26b9fc4934ee2 although I do not have the appropriate hardware to try it out at the moment...

u1f35c, 2017-02-07:
Is there a corresponding patch to OpenOCD or similar? I have STM32F103 hardware I could test against (I'm currently using OpenOCD with bitbanging on the Bus Pirate for SWD support).

agatti, 2017-02-07:
Unfortunately said implementation was more of a proof of concept to see how difficult would be adding SWD support to the BP.  I believe that adding SWD transport to OpenOCD's BP driver shouldn't be too difficult once SWD support is in place, though.

Right now I'm a bit swamped with work and other things so I cannot really dedicate lots of time to BP firmware development this month, but I'd love to see somebody picking up my initial implementation from https://github.com/BusPirate/Bus_Pirate/issues/29#issuecomment-257568200 and make it work :)

u1f35c, 2017-05-01:
I haven't had any time to look at the SWD implementation on the BusPirate itself, but I have been using the patch at [https://the.earth.li/~noodles/bp-swd.diff](https://the.earth.li/~noodles/bp-swd.diff) with OpenOCD 0.10.0 to enable SWD via bitbanging using the BusPirate. This is a slightly modified version of the diff at [http://openocd.zylin.com/#/c/2444/](http://openocd.zylin.com/#/c/2444/) and has been working fine for me with an STM32F103 device.

ChlorideCull, 2018-02-14:
I'll see if I can get on-board SWD support up and running once I get my Bus Pirate v4, I assume it will take a  while to get here because of the Chinese New Year.

ChlorideCull, 2018-08-02:
I was reminded by this issue by the board finally showing up in the mail. Needless to say, it's in rough shape after spending six months in the postal system, so I don't think I'll be able to do this any time soon.