ElderlyPirate/Bus_Pirate #32: Enhanced Logic Analyzer

USBEprom, 2016-11-13:
Hi guys.
Now something a bit hard that I know nothing if it is possible to do or not.
Something that would be an enhancement.
Perhaps this is not the right place to put it, however here is what it is.
Into the dangerousprototypes forum at this link it started a discussion about possible improvements for the Logic Analyzer side of the Bus Pirate (http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=6210&view=unread#p56802).
In the course of that discussion it was provided a working solution with these features:

a) sampling rates up to 16 MHz
b) additional samples when using fewer channels (up to 32k for 1 channel)
c) trigger location anywhere in the buffer
d) backward compatible with previous Logic Analyzer modes

Known Limitations:

-Selecting a trigger position of 100% will set the position at 0%. Workaround is to use 99% instead.
-Due to quirks at 4 and 8 MHz sampling, there is jitter in the samples but the long term rate is exact.
For example, 4 MHz samples should be every 250 nanoseconds, but it is actually between 187 nanoseconds
and 375 nanoseconds between samples.
-If compiled without optimization, 1 MHz sample timing is off prior to the trigger.

Possible Future Improvements:

-Allow selection of channel(s) to be recorded at higher sample sizes

Enhanced Logic Analyzer (latest).zip

All the improvements are into the SUMP.c that is inside the released package of this Enhanced Logic Analyzer release which has too the hexadecimal compiled with option 1 ready to use, the ols.profile-buspirate-enhanced.cfg file to put into the analyzer client and some instructions on how use the thing.
I wonder if could be possible to merge the content of the SUMP.c of the Enhanced Logic Analyzer into the one inside the current repository so that new features are available.
I know nothing but I understand this must to be very hard to reach.
Thanks.

mike01, 2018-01-27:
Looks nice. As it's SUMP compatible, does it also work with sigrok-cli? Are there any example-calls which are known to work?
Another feature which (normally) should be easy to develop would be Run Length Encoding but looking at the code the timing seems to be too critical (especially at 16MHz) to allow any additional clock cycles to be wasted when sampling.

USBEprom, 2018-01-28:
Since sigrok claims to support the Pirate Bus, sigrok-cli should also support it:

https://sigrok.org/wiki/Dangerous_Prototypes_Buspirate

https://sigrok.org/wiki/Supported_hardware#Logic_analyzers

Something like this

sigrok-cli --driver ols:conn=/dev/ttyUSB0

should start the whole thing when it correctly installed and functioning.
It would be necessary to try it, I do not use neither sigrok nor sigrok-cli though and I could not find any examples to start this last one from Windows which is the operating system that I use normally.
For what I know several of the available clients already support RLE, generally the delay is not managed with the Pirate Bus.
I know that this other is pretty old, but it would seem to work with the Pirate Bus too and provide also RLE support, even if I have never tried it:

http://dangerousprototypes.com/blog/2010/05/31/windows-logic-sniffer-client-update/
https://sumpla.codeplex.com/downloads/get/152536

AreYouLoco, 2021-07-18:
```
areyouloco@debian:~/code$ sudo sigrok-cli --driver ols:conn=/dev/ttyUSB0 --scan
sr: openbench-logic-sniffer: Invalid reply (expected 4 bytes, got 1).
areyouloco@debian:~/code$ sudo sigrok-cli --driver ols:conn=/dev/ttyUSB0 --scan
The following devices were found:
ols - BPv3 with 5 channels: 0 1 2 3 4
areyouloco@debian:~/code$ sudo sigrok-cli --driver ols:conn=/dev/ttyUSB0 --scan
sr: openbench-logic-sniffer: Invalid reply (expected 4 bytes, got 1).
areyouloco@debian:~/code$ sudo sigrok-cli --driver ols:conn=/dev/ttyUSB0 --scan
The following devices were found:
ols - BPv3 with 5 channels: 0 1 2 3 4
```
Seems like it's working every second time. Working, not-working, working, not-working.
The same for:

```
areyouloco@debian:~/code$ sudo sigrok-cli --driver ols:conn=/dev/ttyUSB0 --show
Driver functions:
    Logic analyzer
Scan options:
    conn
    serialcomm
sr: openbench-logic-sniffer: Invalid reply (expected 4 bytes, got 1).
No devices found.
areyouloco@debian:~/code$ sudo sigrok-cli --driver ols:conn=/dev/ttyUSB0 --show
Driver functions:
    Logic analyzer
Scan options:
    conn
    serialcomm
ols - BPv3 with 5 channels: 0 1 2 3 4
Supported configuration options:
    Maximum number of samples: 0
    samplerate (10 Hz - 200 MHz in steps of 1 Hz)
    Supported triggers: 0 1 
    captureratio: 0 (current)
    external_clock: on, off
    pattern: None (current), External, Internal
    swap: on, off
    rle: on, off (current)
```

Also working every second time. I am using master build community firmware from yesterday:
```
HiZ>i
Bus Pirate v3.a
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
```