ElderlyPirate/Bus_Pirate #34: 7.1 release checklist

agatti, 2016-11-14:
- [ ] Prepare builds for v3 and v4.
- [ ] Test builds.
- [ ] Bump version.
- [ ] Create build packages.
- [ ] Write changelog.
- [ ] Remove old packages from repo.
- [ ] Upload builds to Github.
- [ ] Announce the release to dangerousprototypes.com.

avongluck-r1soft, 2016-12-27:
Great work agatti!  Sorry that I haven't been helping much lately.
I have a v4 and think I'll pick up a v3 to help test out 7.1.

kallisti5, 2016-12-28:
Just tested on my BPv4.0... master works great.

**Steps**: Upgraded from 7.0 to 7.1 using PicKit2 attached to BPv4 via pk2cmd
**Compiler**: XC16 - v1.30
**Data**: 70% utilized
**Program**: 36% utilized

pk2cmd output (my bootloader got screwed up with my early build attempts last year)
```
$ sudo pk2cmd -PPIC24FJ256GB106 -M -F package/BPv4-firmware/BPv4-update-v7.1.hex
Using PE
PICkit 2 Program Report
27-12-2016, 21:52:38
Device Type: PIC24FJ256GB106

Program Succeeded.

Operation Succeeded
```

test
```
HiZ>i                                                                                                                                                     
Bus Pirate v4                                                                                                                                             
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO]                                                            
DEVID:0x1019 REVID:0x0004 (24FJ256GB106 UNK)                                                                                                              
http://dangerousprototypes.com                                                                                                                            
HiZ>~                                                                                                                                                     
Disconnect any devices                                                                                                                                    
Connect (ADC to +3.3V)                                                                                                                                    
Space to continue                                                                                                                                         
Ctrl                                                                                                                                                      
AUX OK                                                                                                                                                    
MODE LED OK                                                                                                                                               
PULLUP H OK                                                                                                                                               
PULLUP L OK                                                                                                                                               
VREG OK                                                                                                                                                   
EEPROM                                                                                                                                                    
SCL OK                                                                                                                                                    
SDA OK                                                                                                                                                    
WP OK                                                                                                                                                     
READ&WRITE OK                                                                                                                                             
ADC and supply                                                                                                                                            
Vusb(4.91) OK                                                                                                                                             
5V(4.90) OK                                                                                                                                               
5V VPU(4.69) OK                                                                                                                                           
ADC(3.15) OK                                                                                                                                              
3.3V(3.11) OK                                                                                                                                             
3.3V VPU(3.19) OK                                                                                                                                         
Bus high                                                                                                                                                  
MOSI OK                                                                                                                                                   
CLK OK                                                                                                                                                    
MISO OK                                                                                                                                                   
CS OK                                                                                                                                                     
[... cut ...]
MISO OK                                                                                                                                                   
CS OK                                                                                                                                                     
MODE, VREG, and USB LEDs should be on!                                                                                                                    
Any key to exit                                                                                                                                           
Found 0 errors.
```

agatti, 2017-01-02:
#38 and #39 are currently blockers for 7.1 release.

agatti, 2017-09-25:
Only #38 and #64 left blocking.

kallisti5, 2017-09-25:
Tested current sources on my BPv4. Working as expected minus BL issues in #38

```
HiZ>~
Disconnect any devices
Connect (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
EEPROM
SCL OK
SDA OK
WP OK
READ&WRITE OK
ADC and supply
Vusb(4.84) OK
5V(4.82) OK
5V VPU(4.63) OK
ADC(3.49) OK
3.3V(3.53) OK
3.3V VPU(3.20) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE, VREG, and USB LEDs should be on!
Any key to exit
Found 0 errors.
```

AreYouLoco, 2021-07-22:
Tried latest master:
```
HiZ>i
Bus Pirate v3.a
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE PIC DIO] Bo://dangerousprototypes.com
HiZ>~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(5.00) OK
VPU(5.03) OK
3.3V(3.29) OK
ADC(3.24) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on!
Any key to exit
Found 0 errors.
```

Seems fine but what doesn't work for sure is when I try to flash a BIOS chip with flashrom v1.2 on Linux 5.10.0-7-amd64 and 7.1 community release it reads and detects the chip just fine but when writing it hangs forever without led indicating mode set. Is this: https://github.com/BusPirate/Bus_Pirate/issues/131 related?