ElderlyPirate/Bus_Pirate #37: Move to M-Stack for USB support?

agatti, 2016-12-25:
Maybe migrating the USB stack from the hacked up version that is currently running (and has not been updated in a few years now) to something a tad more modern and still supported could be beneficial to the project.

M-Stack is a GPL/Apache2 licensed USB stack for PIC microcontrollers that allows for a whole lot of options, and more importantly it is being currently maintained.  It is available at http://www.signal11.us/oss/m-stack/ (or https://github.com/signal11/m-stack for the source code).

MattWeatherford, 2016-12-27:
As a casual observer of the Bus Pirate Github repository, and an owner 
of a PB v3,  this USB stack upgrade sounds like a really good idea for 
the long term health of the project - what would be required to get this 
effort underway?

psolyca, 2016-12-27:
As MattWeatherford I am following Bus Pirate Community Firmware as an observer for now and casual user of my PB v3b mainly used for UART and JTAG.
If this new stack is lighter than the original stack it could also be better for futher development.
You could count on me to test the firmware.

agatti, 2016-12-28:
Right now there are two main issues regarding switching to M-Stack.

First, pretty much all code in the repo is public domain or CC-0 (with the exception of some XSVF support code for v4 that nobody ever used, I think), while M-Stack is Apache2/GPL.  I am not an expert when it comes to mixing and matching licences, so I am not sure whether mixing Apache2 and CC-0/PD might be possible from a legal standpoint (or even advisable).

Second, M-Stack's code size.  I am yet to try out the code to see whether it can be stripped down to a size comparable to the current USB stack or whereabouts.  I am willing to lose a couple hundred bytes of available space in exchange for a more stable USB stack, but again, depends on what the tradeoff is.

If anyone is willing to do some investigation on how big can M-Stack be or whether mixing licences is fine, please reach out - I would love to update that part of the code if I have the chance :)

agatti, 2016-12-28:
By the way, I forgot to mention that v3 is using a FTDI USB to serial interface rather than a software USB stack - so v3 is as stable as it can be from that regard.  Sorry for the misunderstanding :|

andersm, 2017-09-22:
The Apache Software Foundation have an [FAQ](https://www.apache.org/legal/resolved.html#can-works-placed-in-the-public-domain-be-included-in-apache-products) on combining Apache- and PD-licensed code.

I have no idea about its size or quality, but for what it's worth, Microchip's [MLA USB stack](https://github.com/MicrochipTech/mla_usb) is now also available under the Apache 2 license.

agatti, 2017-09-25:
That's awesome news!  I guess that'll be on the table for 7.2 then.  I'll create a new ticket and hopefully track things from there.

closed 2017-09-25 by agatti