ElderlyPirate/Bus_Pirate #38: Inability to enter bootloader?

kallisti5, 2016-12-28:
Might be an issue with the bootloader after upgrading to v7+ (it could be just me though, I was really hacking on mine early on in the fork)

```
hm just in-case the bootloader issue is something more, here is what happens when I try $ to jump to it:
```
HiZ>$
Are you sure? y
BOOTLOADER

(1)>   <disconnect>   <reconnect>
HiZ>
```

agatti, 2016-12-28:
Yep, can confirm...  Would you be so kind todo a bit of regression testing and see when it started doing that?

avongluck-r1soft, 2016-12-28:
Can do.  I'll test with a few versions.

@agatti Interestingly I noticed this because holding the two pins to start the bootloader never worked after updating to v7.0.   Since the bootloader is acting up, it might be rough for people to upgrade to v7.1 without a pickit or similar.

agatti, 2016-12-28:
Seriously?  That's how I'm doing things here...

kallisti5, 2016-12-28:
Can do. I'll test with a few versions.

@agatti Interestingly I noticed this because holding the two pins to start the bootloader never worked after updating to v7.0. Since the bootloader is acting up, it might be rough for people to upgrade to v7.1 without a pickit or similar.

kallisti5, 2016-12-28:
(sorry, had to switch github accounts :-) )

agatti, 2016-12-28:
NP.  As I said earlier, that's how I've been updating the firmware on my v4, and I guess I flashed my board hundreds of times so far.  Maybe I'm using an old bootloader... I'll check this soon.

USBEprom, 2017-01-03:
I know that I am babo as hell and I could be wrong.
What bootloader do you mean, the newest v4.5 or previous?
I have a Bus Pirate v3 with bootloader v4.4 and I have not any problem while upgrading via software.
I must check via hardware by the jumper.
I will do it and then I will post the result in the weekend because I am pretty busy right now.

agatti, 2017-01-05:
This is not a bootloader-specific issue.  I suspect it happened when section overlaps have been removed, but I cannot test that hypothesis at the moment (issue #10).

USBEprom, 2017-01-07:
I can confirm that with Bus Pirate revision 3 using bootloader v4.4 there are no problems upgrading by terminal either by jumper.
I have checked it.
I had read that it was a problem of Pirate-Loader, not the bootloader.
I knew it is so.

agatti, 2017-01-08:
@USBEprom please check the tags on the right hand side of the screen - this bug is v4 only, it does not apply on v3 boards.

agatti, 2017-01-08:
@kallisti5 did you find out on which build it started doing it?  I got a pickit from a friend to try to debug this issue, and from what I can see the hardcoded address for the bootloader entry point is still actually inside the v4 main firmware (right now it jumps right in the I2C code...)

kallisti5, 2017-02-02:
I got a BusPirate v3b (SparkFun edition) and can confirm out of the box the bootloader works...

```
HiZ>i
Bus Pirate v3b
Firmware v5.10 (r559)  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
BL4+BL4+BL4+
```

kallisti5, 2017-02-02:
Upgraded to v6.1 using the built-in bootloader:

```
kallisti5@nyx BPv3-firmware :) $ ./pirate-loader_lnx --dev=/dev/ttyUSB0  --hex="BPv3-frimware-v6.1.hex"
+++++++++++++++++++++++++++++++++++++++++++
  Pirate-Loader for BP with Bootloader v4+  
  Loader version: 1.0.2  OS: Linux
+++++++++++++++++++++++++++++++++++++++++++

Parsing HEX file [BPv3-frimware-v6.1.hex]
Found 21502 words (64506 bytes)
Fixing bootloader/userprogram jumps
Opening serial device /dev/ttyUSB0...OK
Configuring serial port settings...OK
Sending Hello to the Bootloader...OK

Device ID: PIC24FJ64GA002 [d4]
Bootloader version: 1,02
Erasing page 0, 0000...OK
Writing page 0 row 0, 0000...OK
Writing page 0 row 1, 0080...OK
Writing page 0 row 2, 0100...OK
Writing page 0 row 3, 0180...OK
Writing page 0 row 4, 0200...OK
Writing page 0 row 5, 0280...OK
Writing page 0 row 6, 0300...OK
Writing page 0 row 7, 0380...OK
Erasing page 1, 0400...OK
Writing page 1 row 8, 0400...OK
Writing page 1 row 9, 0480...OK
Writing page 1 row 10, 0500...OK
Writing page 1 row 11, 0580...OK
Writing page 1 row 12, 0600...OK
Writing page 1 row 13, 0680...OK
Writing page 1 row 14, 0700...OK
Writing page 1 row 15, 0780...OK
Erasing page 2, 0800...OK
Writing page 2 row 16, 0800...OK
Writing page 2 row 17, 0880...OK
Writing page 2 row 18, 0900...OK
Writing page 2 row 19, 0980...OK
Writing page 2 row 20, 0a00...OK
Writing page 2 row 21, 0a80...OK
Writing page 2 row 22, 0b00...OK
Writing page 2 row 23, 0b80...OK
Erasing page 3, 0c00...OK
Writing page 3 row 24, 0c00...OK
Writing page 3 row 25, 0c80...OK
Writing page 3 row 26, 0d00...OK
Writing page 3 row 27, 0d80...OK
Writing page 3 row 28, 0e00...OK
Writing page 3 row 29, 0e80...OK
Writing page 3 row 30, 0f00...OK
Writing page 3 row 31, 0f80...OK
Erasing page 4, 1000...OK
Writing page 4 row 32, 1000...OK
Writing page 4 row 33, 1080...OK
Writing page 4 row 34, 1100...OK
Writing page 4 row 35, 1180...OK
Writing page 4 row 36, 1200...OK
Writing page 4 row 37, 1280...OK
Writing page 4 row 38, 1300...OK
Writing page 4 row 39, 1380...OK
Erasing page 5, 1400...OK
Writing page 5 row 40, 1400...OK
Writing page 5 row 41, 1480...OK
Writing page 5 row 42, 1500...OK
Writing page 5 row 43, 1580...OK
Writing page 5 row 44, 1600...OK
Writing page 5 row 45, 1680...OK
[... 318 lines cut ...]
Writing page 41 row 328, a400...OK
Writing page 41 row 329, a480...OK
Writing page 41 row 330, a500...OK
Writing page 41 row 331, a580...OK
Writing page 41 row 332, a600...OK
Writing page 41 row 333, a680...OK
Writing page 41 row 334, a700...OK
Writing page 41 row 335, a780...OK

Firmware updated successfully :)!
Use screen /dev/ttyUSB0 115200 to verify
```

Working as expected:
```
HiZ>i
Bus Pirate v3.5
Firmware v6.1 r1676  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
BL4+BL4+BL4+BL4+
```

kallisti5, 2017-02-02:
v3 Upgrade to v6.2 "beta"

```
kallisti5@nyx BPv3-firmware :) $ ./pirate-loader_lnx --dev=/dev/ttyUSB0 --hex=BPv3-firmware-v6.2-r1981.hex
+++++++++++++++++++++++++++++++++++++++++++
  Pirate-Loader for BP with Bootloader v4+  
  Loader version: 1.0.2  OS: Linux
+++++++++++++++++++++++++++++++++++++++++++

Parsing HEX file [BPv3-firmware-v6.2-r1981.hex]
Found 21502 words (64506 bytes)
Fixing bootloader/userprogram jumps
Opening serial device /dev/ttyUSB0...OK
Configuring serial port settings...OK
Sending Hello to the Bootloader...OK

Device ID: PIC24FJ64GA002 [d4]
Bootloader version: 1,02
Erasing page 0, 0000...OK
Writing page 0 row 0, 0000...OK
Writing page 0 row 1, 0080...OK
Writing page 0 row 2, 0100...OK
Writing page 0 row 3, 0180...OK
Writing page 0 row 4, 0200...OK
Writing page 0 row 5, 0280...OK
Writing page 0 row 6, 0300...OK
Writing page 0 row 7, 0380...OK
Erasing page 1, 0400...OK
Writing page 1 row 8, 0400...OK
Writing page 1 row 9, 0480...OK
Writing page 1 row 10, 0500...OK
Writing page 1 row 11, 0580...OK
Writing page 1 row 12, 0600...OK
Writing page 1 row 13, 0680...OK
Writing page 1 row 14, 0700...OK
Writing page 1 row 15, 0780...OK
Erasing page 2, 0800...OK
Writing page 2 row 16, 0800...OK
Writing page 2 row 17, 0880...OK
Writing page 2 row 18, 0900...OK
Writing page 2 row 19, 0980...OK
Writing page 2 row 20, 0a00...OK
Writing page 2 row 21, 0a80...OK
Writing page 2 row 22, 0b00...OK
Writing page 2 row 23, 0b80...OK
Erasing page 3, 0c00...OK
Writing page 3 row 24, 0c00...OK
Writing page 3 row 25, 0c80...OK
Writing page 3 row 26, 0d00...OK
Writing page 3 row 27, 0d80...OK
Writing page 3 row 28, 0e00...OK
Writing page 3 row 29, 0e80...OK
Writing page 3 row 30, 0f00...OK
Writing page 3 row 31, 0f80...OK
Erasing page 4, 1000...OK
Writing page 4 row 32, 1000...OK
Writing page 4 row 33, 1080...OK
Writing page 4 row 34, 1100...OK
Writing page 4 row 35, 1180...OK
Writing page 4 row 36, 1200...OK
Writing page 4 row 37, 1280...OK
Writing page 4 row 38, 1300...OK
Writing page 4 row 39, 1380...OK
Erasing page 5, 1400...OK
Writing page 5 row 40, 1400...OK
Writing page 5 row 41, 1480...OK
Writing page 5 row 42, 1500...OK
Writing page 5 row 43, 1580...OK
Writing page 5 row 44, 1600...OK
Writing page 5 row 45, 1680...OK
[... 321 lines cut ...]
Writing page 41 row 331, a580...OK
Writing page 41 row 332, a600...OK
Writing page 41 row 333, a680...OK
Writing page 41 row 334, a700...OK
Writing page 41 row 335, a780...OK

Firmware updated successfully :)!
Use screen /dev/ttyUSB0 115200 to verify
```

**Not working:**
```                                                                                                                                                                                                                                                                                           
HiZ>i                                                                                                   
Bus Pirate v3.5
Firmware v6.2-beta1 r1981  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? 
HiZ>$ 
Are you sure? y
BOOTLOADER

<dead>
```

kallisti5, 2017-02-02:
So... it looks like the bootloader entry breakage occurred sometime before the "community firmware" was forked. ( Regression between r1676-r1981   v6.1  and v6.2)

It also looks like v3 and v4 are impacted.

kallisti5, 2017-02-02:
Confirmed the --hello function also fails after v6.2...

```
$ ./pirate-loader_lnx --dev=/dev/ttyUSB0 --hello
+++++++++++++++++++++++++++++++++++++++++++
  Pirate-Loader for BP with Bootloader v4+  
  Loader version: 1.0.2  OS: Linux
+++++++++++++++++++++++++++++++++++++++++++

Opening serial device /dev/ttyUSB0...OK
Configuring serial port settings...OK
Sending Hello to the Bootloader...ERROR
No reply from the bootloader, or invalid reply received: 0
Please make sure that PGND and PGC are connected, replug the devide and try again
```

kallisti5, 2017-02-02:
The v6.1 firmware package is dated:
``` Jan 11  2012 BPv3-frimware-v6.1.hex```

The v6.2 firmware package is dated:
```Aug 14  2012 BPv3-firmware-v6.2-r1981.hex```

So I assume the regression was introduced somewhere in-between.

kallisti5, 2017-02-02:
I've tagged the sources for the likely v6.2 and v6.1 releases based on the timestamps above.

305 commits could of caused the regression.
```
$ git rr v6.1..v6.2 | wc -l
305
```

Only 1 commit mentions "boot loader"... Too bad it's cosmetic (68a8914)

kallisti5, 2017-02-02:
Weird..

https://github.com/BusPirate/Bus_Pirate/search?utf8=%E2%9C%93&q=BLJUMPADDRESS

That address seems a bit all over the road

kallisti5, 2017-05-07:
Ok.  I had some spare time to do a little research into this. Still not working as of latest builds. I noticed the bootloader version is v255.255 as well on my BPv3 with the latest build. (i command)

https://github.com/BusPirate/Bus_Pirate/blob/master/Bootloaders/BPv3-bootloader/firmware-v4.5/ds30loader.s#L148 seems to be key in locating where the bootloader went. Here are the intended locations:

```
.equ	STARTADDR,	( FLASHSIZE - (2* (PAGESIZE)) ) 		/*place bootloader in last program page*/
.equ	BLVERSION,	0x0405	;bootloader version for Bus Pirate firmware (located at last instruction before flash config words)
```

So. The bootloader should be in "the last program page"  lets look at our linker script:
https://github.com/BusPirate/Bus_Pirate/blob/master/Firmware/p24FJ64GA002.gld#L54

```
  program (xr)   : ORIGIN = 0x200,         LENGTH = 0xA5FC
  CONFIG2        : ORIGIN = 0xABFC,        LENGTH = 0x2
  CONFIG1        : ORIGIN = 0xABFE,        LENGTH = 0x2
```

So. With that logic, the BL_VERSION should be 0xABFA (0xABFC - 0x2) (which it is currently set to)
The bootloader is currently ABF8 which is 0xABFA - 0x2

```
PROGRAM        0x200 - 0xA7FC
BOOTLOADER  0xA7FD - 0xABF8  (entry set currently to 0xABF8)
BLVERSION      0xABFA - 0xABFB
CONFIG2          0xABFC - 0xABFD
CONFIG1          0xABFE - 0xABFF
```

So the current settings seem to make "sense" with the latest linker scripts.  I think this definitely means something is wrong with the state of the bus pirate when attempting to jump into the BL.

This doesn't seem to be an issue of the new firmware "overwriting" the bootloader though. You can flash 6.1 and have the BL begin working again. (although... I assume the 6.1 fw didn't include the bootloader?)

agatti, 2017-05-08:
Interesting.  Maybe something that can be done is to reserve some bytes at the end of the code space that just contains either a hardcoded branch instruction to the bootloader start, or a soft-reset trigger.  Since the address of this trampoline is known in advance on either v3 and v4, maybe this should be easier to handle.  Moreover, pirate-loader can also unconditionally write said trampoline on the flash upon FW upload to make sure things are properly set up.

Luckily I'll have some more free time to play with this next week so hopefully something will come up and we can finally push 7.1 out of the door - unfortunately I've been delaying things a bit too much than I'd have liked to, but family and work and the like have priority over this.

Anyway, @kallisti5 thanks for having taken another look on the v3, that was really helpful!

agatti, 2017-06-25:
A first attempt at fixing this has been committed as e2bd0a0.  I've had a quick go on my v4 and it seems to work, but there are no guarantees of this to be either stable or working for you :)

If it doesn't work on v3 then some changes in the v4.5 assembly bootloader need to be performed.  I tried to be as careful as possible but I have no ways to test this.

Unfortunately this is not as "clean" as it used to be, as when entering the bootloader a device reset gets performed rather than just dropping into the bootloader - so if you use something like minicom you'll get a prompt saying "connection lost" or something like that.  A minor annoyance, but after all, how many times normal users need to trigger the bootloader without connecting PGC and PGD together?

agatti, 2017-06-25:
Oh, I forgot to mention that trying this out means reflashing *both* the bootloader and the firmware.  The fastest way to achieve this is to flash the bootloader via MPLAB-X (IDE or IPE doesn't matter), then use the new bootloader to flash the new firmware.

USBEprom, 2017-06-25:
@agatti 

Thank you Sir for the news.
I'd like to check if it works with Bus Pirate revision 3 but I do not understand some things and more I have not any programmer like an ICD2 or PicKit.
I understood that I have to compile the BOOTLOADER v4.5 from myself using MPLAB-X IDE or IPE, does not matter which, but beyond how to do it, then I would need a real programmer like PicKit or ICD2.
Also I am aware that things might go wrong so that anyway it is need something like a real programmer in order to rescue the BusPirate.
In the past there was a special upgrader utility that made it possible to do everything simply with "ds30 Loader" or "pirate-loader" without any need to use a real programmer like PICkit2 or ICD2.
I am talking about the upgrader utility written in order to upgrade BOOTLOADER from revision v4.x to v4.3, namely "BPv3-bootloader-upgrade-v4xtov4.3.hex".
Just put the hex file into the BusPirate by using "pirate-loader" or "ds30 Loader", start the utility from the terminal and then again with "pirate-loader" or "ds30 Loader" download a valid firmware, actually does not matter exactly what because it always possible to upgrade/change it to any else.
Starting from MPLAB-X IDE or IPE it would be possible to build an hexadecimal file of the BOOTLOADER v4.5  suitable for the purpose?

agatti, 2017-06-25:
@USBEprom, on v3 things are a bit more complicated since you must already have the v4.5 bootloader already installed first.  Then, I am not sure if things can be recovered without a PICkit if the bootloader does not work - if you don't happen to have one handy it's recommended to leave things as they are at the moment.  The firmware should work just fine even with the old bootloader though.

Anyhow, you can now build the bootloader v4.5 with MPLAB-X now, as I fixed the building process - you'll end up with a hex file you can flash at will.

USBEprom, 2017-06-29:
OK. 
I understood that it is better to leave things as they are because the firmware should work just fine even with the old bootloader, my only purpose was try to contribute with testing.
I will investigate on how to build the bootloader v4.5 with MPLAB-X, that honestly now I do not know, but seems to me however it is need a real programmer like PICkit2 or ICD2 in order to do that and finally put the HEX file into the PIC of the Bus Pirate.
From my side I bought a PICkit2 in order to be ready for everything but I do not think that many others will follow me.
I think that a solution like what is explained here could be the better option because does not require any special programmer or things like that:
 
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=619&hilit=P24QP.exe
 
Surely exists more than a chance to bricking the Bus Pirate though, so that having a real programmer at your fingertips would be the best scenario.

agatti, 2017-07-18:
@kallisti5 do you happen to have some time to check this on the boards you've got there?  If it works as intended then we can wrap this release up and set it free...

agatti, 2017-09-15:
@kallisti5 You've got both a v3 and a PicKit, so I guess you can take a look at this on why on v3 the code does not behave as expected (as per report from #58) :)

satscan, 2017-09-20:
I spent some time looking at the code and might have identified the root cause....

The inability to enter bootloader mode is caused by proc_menu.c (static bool __attribute__((address(0x27FA),persistent)) skip_pgc_pgd_check;).

It is referenced in ds30loader.s (.section *, bss, address(0x27FA) / skip_pgc_pgd_check: .space 2) of bootloader 4.5. Previous BPv3 bootloaders (<4.5) don't know anything about skip_pgc_pgd_check so it won't work without putting a jumper between the two pins.

With BPv4 you won't hit this problem because skip_pgc_pgd_check was present in it's first bootloader revision.

Why did it work with previous firmwares? Because someone changed

asm volatile ("mov #BLJUMPADDRESS, w1 \n" //bootloader location" goto w1 \n");

to

skip_pgc_pgd_check = true;
asm volatile ("RESET");

The new style flags skip_pgc_pgd_check. Then it does a reset and expects the bootloader to check the contents of skip_pgc_pgd_check. It should decide to stay in the bootloader instead of branching to the application. This will only work with an updated bootloader (>=4.5).

The old style did a direct jump so it will work with any bootloader revision.

agatti, 2017-09-21:
@satscan ok, so you can confirm that the current firmware code *with* the current bootloader for v3 does in fact allow to enter into bootloader mode from the main menu with the '$' command?  If so then I can close the ticket and hopefully start the 7.1 release process.

satscan, 2017-09-21:
@agatti Yes, I can confirm the issue is solved by upgrading to latest bootloader 4.5 for BPv3 hardware. My BPv3a had this problem with bootloader 4.3. Updating to bootloader 4.5 fixed it.
I have compiled BPv3-bootloader-upgrade-4.xto4.5.hex and sent it to USBEprom so you should get another feedback. He has a BPv3.x with factory installed bootloader 4.4.
The difficulty seems to be the fact that the only available update file is BP3-bootloader-upgrade-4.xto4.3.hex. Bootloader 4.3 and 4.4 are the same and both will have this problem. It's only factory installed (4.4) and field update (4.3).

agatti, 2017-09-21:
@satscan thanks for the report.  I'll do some cleanup in the bootloaders directory right away and hopefully have an official update package to check in the repo as well.

USBEprom, 2017-09-22:
**@satscan**

Thank you Sir for the **BPv3-bootloader-upgrade-4.xto4.5.hex** file you sent me!
I used it to update the bootloader of my Bus Pirate v3 with **pirate-loader.exe** and it works like a charm!
My BusPirate v3 had the bootloader v4.4 and with your **BPv3-bootloader-upgrade-4.xto4.5.hex** now it has the bootloader v4.5, really great!
I followed these instructions:

http://dangerousprototypes.com/blog/2010/06/23/bus-pirate-bootloader-v4-3v4-4-upgrade/

**pirate-loader.exe** works, while weirdly enough **ds30_Loader_GUI.exe** does not, it fails with message "nothing to do".
Surely it is my fault because I am not accustomed to use **ds30_Loader_GUI.exe** that I do not even know so well.
Indeed for this kind of matter generally I always use **pirate-loader.exe** and I have never had a problem.
Now my Bus Pirate has bootloader v4.5 and I could test a my personal firmware version dated July 20, 2017 that had the problem of not getting into the bootloader mode using the command "**$**" while now it work as all can see:

HiZ>i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] **Bootloader v4.5**
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ> ~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.96) OK
VPU(5.03) OK
3.3V(3.31) OK
ADC(3.35) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on!
Any key to exit
Found 0 errors.
**HiZ>$
Are you sure? y
BOOTLOADER**

Thanks a lot sir!

**@agatti**

Thank you Sir for keeping update the repository and all the changes you are doing!
I confirm you that with bootloader v4.5 (also thanks **satscan** and **kallisti5**!) the problem is gone as for **#38** which is the same. 
Sadly still I can not build any new firmware because I did not finish repairing my computer but as soon as I will succeed with it I will go back to testing the repository.
Thanks a lot sir!

USBEprom, 2017-09-24:
@agatti, @kallisti5, @satscan, @all

OK, thank to satscan I have succeeded.
As a reference:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67056#p67056

USBEprom, 2017-09-28:
About the message "**nothing to do**" issued by **ds30_Loader_GUI.exe** of which I wrote, actually it is due a wrong configuration made by myself in **ds30_Loader_GUI.exe**.
It was my fault, **ds30_Loader_GUI.exe** does not give any problem, it works exactly as expected as I just checked.
Please, sorry for the mistake.

NTICompass, 2017-10-31:
If I am currently on "Firmware v6.2-beta1 r1981  Bootloader v4.4" (on my Bus Pirate v3.5), is there any way to "fix" it so I can access the bootloader again?

agatti, 2017-10-31:
@NTICompass I suppose you can try to update the firmware via the package mentioned in https://github.com/BusPirate/Bus_Pirate/issues/38#issuecomment-331713632 if all you have is the BPv3 unit and no other way to program PIC24 chips.

However, is such a function really useful?  I mean, you can always short PGC & PGD and then reconnect the device to achieve the same effect, and the bootloader should only be messed with when uploading new firmware images after all.

NTICompass, 2017-10-31:
@agatti Shorting PGC/PGD did the trick :)

agatti, 2018-09-05:
Moving to 7.2...