ElderlyPirate/Bus_Pirate #41: uart.c bit_sample uninitialized in uart_get_baud_rate function

Fl0-0, 2017-01-07:
uart.c line ~524:
uint32_t uart_get_baud_rate(const bool quiet) {
  size_t samples;
  uint32_t current_sample;
  uint32_t bit_sample;

  BP_MISO = LOW;

bit_sample is not initialized.
Warning at compilation:
../uart.c: In function 'uart_run_macro':
../uart.c:524:12: warning: 'bit_sample' may be used uninitialized in this function

Used at line 658:
if ((samples > 0) && ((bit_sample == 0) || (bit_sample > current_sample))) {
      bit_sample = current_sample;
    }

USBEprom, 2017-01-07:
Maybe it is #39

agatti, 2017-01-08:
Interesting.  This didn't cause an error here both on v3 or v4, with MPLAB X 3.50 and XC16 1.30.  Anyway, it's been fixed with 0ec8c381531c83b3e39ec3c525ac8f9ae6ed94d9.

closed 2017-01-08 by agatti

USBEprom, 2017-01-08:
Here it is MPLAB X 3.40 and XC16 1.26 and there was the issue when building for Bus Pirate revision 3.
Now I am ready to upgrade at MPLAB X 3.50 and XC16 1.30.
Many thanks sir!