ElderlyPirate/Bus_Pirate #43: Trouble updating bootloader and firmware for openocd

ntzrmtthihu777, 2017-01-27:
As stated. I'm intending to grab a flyswatter2 from tincan tools as a more 'proper'
jtag interface for u-boot development, but wish to at least test I've got the right pinout.

I wished to update my firmware to support openocd/jtag, and cloned the repo, and
attempted to flash via
`./BPv3-firmware/pyrate-loader_lnx --dev=/dev/ttyUSB0 --hex=bootloader/BPv3-Bootloader-v4.4.hex`

It all goes well until it hits here: `Erasing page 42, a800...ERROR [50]`

As it currently stands my bus pirate is unusable. Tips for recovery/getting this done?

USBEprom, 2017-01-27:
@ntzrmtthihu777

Please take a look here:
https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-273879409
 
What do you mean "As it currently stands my bus pirate is unusable"?
Actually the message of error is not a real issue.

ntzrmtthihu777, 2017-01-28:
I mean, when I `screen /dev/ttyUSB0 115200` it, it does not present the 'normal' bus pirate interface.
I managed to recover by reflashing stock firmware though, which is good.

ntzrmtthihu777, 2017-01-29:
Got it figured out. Sorry to pester :)

closed 2017-01-29 by ntzrmtthihu777

kkey00, 2019-06-25:
If you don't mind me asking how did you fix this problem? I'm having the same issue

USBEprom, 2019-06-25:
@kkey00

Actually that error message does not mean that something is wrong with the firmware.
More than a matter of bootloader it is a issue with pirate-loader or related tools.
The error does nothing, just ignore it, it does not matter if it is shown in the output while upgrading because actually all is fine, no problen at all.
It has already been discussed at other times, please read here:

https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-273879409