ElderlyPirate/Bus_Pirate #45: SPI protocol does not work anymore.

USBEprom, 2017-02-04:
I have just built a firmware for the BP3 by compiling from the current repository (February 4, 2017) with MBLAB 3.50 putting flag only on "**Do not override 'inline'**" leaving the rest of its settings empty or disabled.
For Bus Pirate revision 3 optimizations 0, 2 and 3 are run out of space, the only valid ones are 1 and s, so i built by using them.
This report is about firmware optimized option 1.
My configuration.h is this:
[configuration.h.TXT](https://github.com/BusPirate/Bus_Pirate/files/752259/configuration.h.TXT)
(Pay attention at the fact that "github" does not allow me to attach configuration.h, not even by compressing it as zip, so I had to change its extension as txt.)
 
As check I began to test the SPI protocol trying to retrieve the JEDEC of different SPI chips discovering that the SPI protocol no longer works.
On all speed always I get wrong JEDEC on the terminal while actually it is totally correct on the bus as I verified by using a logic analyzer.
So timing and data exchange on the bus are always correct while the reads printed on the terminal are always wrong.
I have tried many chips and speeds but the result is always the same, SPI protocol does not work anymore in this latest firmware where the patch #23 is applied.
As countercheck I have repeated the same tests with a firmware built by compiling the repository issued immediately before the application of the patch #23 and everything works perfectly as supposed.
For what I know the only differences between the firmware that does work and the one that does not are:
 
The functioning firmware has not the patch #23
The functioning firmware has not the workarounds in order to unlock hardware I2C protocol

Here is the log of the terminal:
[TERMINAL LOG.txt](https://github.com/BusPirate/Bus_Pirate/files/752264/TERMINAL.LOG.txt)
(**Note tha it is not a terminal problem because I have tried different ones and the result is always the same, it does not matter the terminal program used, JEDEC is always wrong**)

While here is a shot taken with the logic analyzer

[image]

agatti, 2017-04-10:
Fixed with bd7c8f27b40dda8b6cef72d320f1fb34127fb933.

closed 2017-04-10 by agatti

USBEprom, 2017-04-12:
I am sorry to bother you again here that the issue has been already closed.
Trying to compile a new test firmware starting from the new repository I got these errors from MPLAB:
 
[Errors 12042017.txt](https://github.com/BusPirate/Bus_Pirate/files/917928/Errors.12042017.txt)
 
If I am not doing wrong there might be a problem.
Thanks!

agatti, 2017-04-13:
Your modifications to configuration.h are invalid, please double check those.

```
../configuration.h:547:2: error: #endif without #if
```

USBEprom, 2017-04-13:
Thank you sir, you are the best!
As always you are right.
I am babo.
I had mistakenly changed the line 316 into my configuration.h.
After correcting it now everything works.
I will test this new firmware and I will post the results.
Many thanks again sir!