ElderlyPirate/Bus_Pirate #46: 2WIRE protocol does not work anymore.

USBEprom, 2017-02-04:
I have just built a firmware for the BP3 by compiling from the current repository (February 4, 2017) with MBLAB 3.50 putting flag only on "**Do not override 'inline'**" leaving the rest of its settings empty or disabled.
For Bus Pirate revision 3 optimizations 0, 2 and 3 are run out of space, the only valid ones are 1 and s, so i built by using them.
This report is about firmware optimized option 1.
My configuration.h is this:
[configuration.h.TXT](https://github.com/BusPirate/Bus_Pirate/files/752291/configuration.h.TXT)
(Pay attention at the fact that "github" does not allow me to attach configuration.h, not even by compressing it as zip, so I had to change its extension as txt.)
 
As check I began to test the 2WIRE protocol trying to retrieve the ATR of a SLE4442 chip card discovering that the 2WIRE protocol no longer works.
Always I get wrong ATR both by performing macros and by inserting directly the commands.

As countercheck I have repeated the same tests with a firmware created by compiling the repository issued immediately before the application of the patch #23 and everything works perfectly as supposed.
For what I know the only differences between the firmware that does work and the one that does not are:
 
The functioning firmware has not the patch #23
The functioning firmware has not the workarounds in order to unlock hardware I2C protocol
 
Here is the log of the terminal:
[TERMINAL LOG.txt](https://github.com/BusPirate/Bus_Pirate/files/752292/TERMINAL.LOG.txt)
(**Note tha it is not a terminal problem because I have tried different ones and the result is always the same, it does not matter the terminal program used, ATR is always wrong**)
 
The real and right ATR  actually it is the usual 0xA2 0x13 0x10 0x91 of that kind of cards, while this latest firmware always reads 0xAA 0x11 0x11 0x99 which is totally wrong.

agatti, 2017-04-10:
Fixed with bd7c8f27b40dda8b6cef72d320f1fb34127fb933.

closed 2017-04-10 by agatti

USBEprom, 2017-04-13:
I confirm that it is gone, now works properly.
Thank you sir!
You are the best!