ElderlyPirate/Bus_Pirate #48: Bus Pirate V3

Marsfan, 2017-03-19:
What is the status of the firmware for the Bus Pirate v3b (the sparkfun model)? Does the newest version load into the chip, and am I missing any features? Running i lists version 5.10 and these modes:

1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. LCD
9. DIO

agatti, 2017-03-20:
Right now you have to rebuild things from source.  You may want to take a look at http://dangerousprototypes.com/forum/viewtopic.php?p=65526#p65526 - I'm not sure whether those firmwares were tested on a v3b to be honest.

crapp, 2017-03-30:
@agatti Looks like there is some problem with software i2c on the buspirate v3. This seems to have been introduced with firmware 7.1. Can you confirm this?

Marsfan, 2017-03-30:
@crapp I saw something on the posts, but @agatti's link seems to be the last successful build made for v3 so far. My current worry is about overwriting the bootloader. (see my issue #49). Anyone willing to clear that up for me.

USBEprom, 2017-04-19:
@agatti

Firmware here:

http://dangerousprototypes.com/forum/viewtopic.php?p=65526#p65526

and better, latest version, here:

http://dangerousprototypes.com/forum/viewtopic.php?p=65526#p66033

as all the firmwares I put on the dangerousprototypes site I confirm that they are tested and functioning on Bus Pirate revision 3.
I own a Bus Pirate v3.

@Marsfan

Take a look at this:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=60#p66038

@crapp
 
I do not know if it is only revision 3 or also revision 4.
However with the latest firmwares linked above I2C is functioning on Bus Pirate revision 3 even if with the known limitations described here: 

https://github.com/BusPirate/Bus_Pirate/issues/23

Marsfan, 2017-05-03:
Ok, USBEPROM's builds work. Closing this down

closed 2017-07-17 by Marsfan