ElderlyPirate/Bus_Pirate #49: Updating via ds30 says it will overwrite bootloader on BPv3

Marsfan, 2017-03-29:
I have been trying to update the boot loaderwith [this file](http://dangerousprototypes.com/forum/viewtopic.php?p=65526#p65526), but the ds30 updoader says I will overwrite the bootloader on my bus pirate. 

What's more, when I try to proceed, it just says that it has nothing to do. How can I fix these errors

JarrettR, 2017-03-30:
Have you tried the [Pirate loader](http://dangerousprototypes.com/docs/Pirate-Loader_console_upgrade_application_(GNU/Linux,_Mac,_Windows)) instead?

Marsfan, 2017-03-30:
@JarrettR I have not, but I am more worried about breaking the bootloader instead of how to bypass it.

JarrettR, 2017-03-30:
I'm not sure what your question is

JarrettR, 2017-03-30:
That might be the source of your confusion, The link that you provided contains files for the firmware, not the bootloader.

The Pirate loader will communicate with your existing bootloader v4 (no upgrade necessary), and install the firmware that you linked. I've never used ds30, so I don't know how it works.

Marsfan, 2017-03-30:
@JarrettR it could be a DS 30 bug, but I know it is firmware. DS30 is warning me it might overwrite the bootloader when it puts the firmware on it. Can you flag this for someone else to look at?

agatti, 2017-04-02:
@Marsfan pirate-loader won't touch the bootloader, and it is the only supported (as in, we can provide help with) firmware update method at this point in time.

Marsfan, 2017-04-04:
@agatti So to verify, despite the fact that DS30 shows an error on that, the command line system will not overwrite the boot loader?

I saw some stuff on the main release thread, but it seems to have dried up. Anything special I need to do?

Marsfan, 2017-04-05:
@agatti to clarify, I am talking about the` Erasing page 42, a800...ERROR [50] ` message that one gets on pirate loader (happened with me), how would one fix this?

Marsfan, 2017-04-05:
@agatti ok so it worked despite that warning. someone want to tell me what the heck that warning meant?

agatti, 2017-04-05:
@Marsfan If I recall correctly, that's where the bootloader is actually stored on v3.  Please check out the discussion for issue #26.

agatti, 2017-04-05:
After a bit more thought, maybe pirate loader should be changed to be a bit more descriptive when tries (and fails) to overwrite the bootloader.

closed 2017-04-06 by agatti