ElderlyPirate/Bus_Pirate #51: Wait until interrupt command

dspmind, 2017-04-16:
I was looking for feature in BPv3 hardware that enable one to configure AUX pin as an H->L or L->H interrupt i/p pin and use that modify the binary bit bang protocol to generate an protocol EVENT on the interrupt arrival. This will help in scripting several h/w devices. An exact same feature was already asked in 
https://github.com/0x90/the-bus-pirate/issues/5

Is this feature very difficult to implement, extending the binary bit bang protocol of BP.

Thanks in Advance

agatti, 2017-04-16:
It doesn't look that difficult to implement, we'll see if something like this can fit the usage in 7.2.

agatti, 2017-04-17:
On second thought, the changes for this seem to be a bit more extensive than expected.  Issue #52 should also cover this.

tudstudent, 2026-10-23:
Implemented in #218.

The change adds a wait-for-edge command to the binary protocol, in the free
`0111` command group - so the same command works in raw bitbang, SPI, I2C,
UART, 1-Wire and raw-wire modes:

    01110000 <edge> <timeout-high> <timeout-low>

`<edge>` is `0b01` for rising, `0b10` for falling and `0b11` for either.
`<timeout>` is 16-bit big-endian milliseconds; `0` checks the pin once and
returns at once. The reply is two bytes: the result code (`0x01` edge,
`0x00` timeout) followed by the AUX pin level at that moment - so a timeout
of `0` doubles as a plain pin read.

On the asynchronous part: the packaging of a protocol event byte while other
commands are in flight is what the protocol revision discussion in #52
covers (in the current protocol an unsolicited byte could be mistaken for
read data). Until that framing exists this implements the synchronous form -
the command returns exactly when the event happens - with the edge selection
(rising/falling/either) that the discussion settled on. Host scripts can
already stop busy-polling: they issue the wait and block, e.g. right before
reading a slow sensor that raises an interrupt line.

While the wait runs, the pin's change-notification hardware latches the
edge, so an edge cannot be missed between polls; the wait honours the
alternate-AUX selection and leaves the pin as an input like every other AUX
read. Verified on hardware: timeouts, timing, levels, the NACK paths, and
the command being accepted in all six binary modes; the real-edge check was
done with a jumper (GND -> 3.3 V -> GND) on a v4.