ElderlyPirate/Bus_Pirate #53: Put compiled builds under the Releases tag

Marsfan, 2017-06-25:
I brought this up over on the forum, but not much came from it (no ones fault). Perhaps we could start taking the completed builds that @USBEprom puts up on the forum, and start placing them under the releases function within github. This would allow for people to quickly see all of the newest releases, instead of having to dig through the forum posts. 

The main issue with this is that @USBEprom is not sure how this is done on github (totally understandable), so doing this would require some collaboration between him and someone with more github experience.

agatti, 2017-06-26:
Right now the only thing blocking release is the verification of the fix for #38.  Once that's over with then we'll have official tested builds for v3 and v4 put under the release tag, no worries :)

Marsfan, 2017-06-26:
@agatti Oh ok. I though that maybe use of that had faded, and main releases were @USBEprom's Perhaps we could create a beta tag to put his under as well?

closed 2017-06-26 by Marsfan