ElderlyPirate/Bus_Pirate #54: HARDWARE mode with I2C protocol, how it works? (if it does).

USBEprom, 2017-07-18:
Hi guys.
Is there somebody here who know the differences between SOFTWARE and HARDWARE mode in I2C protocol?
I thought the only one was about how synchronization is generated for the bus, but I am not sure it is just that.
In fact in the recent past thanks to agatti it was possible to free the HARDWARE mode for the I2C protocol also with Bus Pirate v3 (https://github.com/BusPirate/Bus_Pirate/issues/39) but although in my device I have a silicon revision B8 (DEVID:0x0447 REVID:0x3046 = 24FJ64GA002 B8) I noticed a weird behavior.
The weird thing is that while doing thing on a I2C serial EEPROM via HARDWARE mode I can hit the chip only the first time, performing new access the answers are always wrong (0xFF).
So, for example, while performing read of data from a given block of memory I get them right only the first time because by repeating the command I get wrong data as 0xFF.
Take a look at this:

Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)

(1)>4
I2C mode:
 1. Software
 2. Hardware

(1)>2
Set speed:
 1. 100KHz
 2. 400KHz
 3. 1MHz
(1)>1
Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'
Ready
I2C>WP
POWER SUPPLIES ON
Clutch engaged!!!
Pull-up resistors ON
I2C>[0xA0 0x00][0xA1 r:256]
I2C START BIT
WRITE: 0xA0 ACK 
WRITE: 0x00 ACK 
I2C STOP BIT
I2C START BIT
WRITE: 0xA1 ACK 
READ: 0x00  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x1F 
NACK
I2C STOP BIT
I2C>[0xA0 0x00][0xA1 r:256]
I2C START BIT
WRITE: 0xA0 ACK 
WRITE: 0x00 ACK 
I2C STOP BIT
I2C START BIT
WRITE: 0xA1 ACK 
READ: 0x00  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF  ACK 0xFF 
NACK
I2C STOP BIT
I2C>

In order to fix I have to reset the Bus Pirate with command "#" and setting again all the I2C parameters.
Though 'Macro (1)' always works also by repeating it.

Instead by using SOFTWARE mode I have not any problem performing all the commands I want without getting wrong answers.
I know that due the different PIC used the Bus Pirate v4 has always had the HARDWARE mode unlocked as opposed to v3, which also depends on the silicon revision (http://ww1.microchip.com/downloads/en/D ... 00470j.pdf).
Is there someone who owns the v4 and can confirm that it is the intended behavior?
Otherwise it is very likely to be a bug.
Thanks!

agatti, 2017-07-18:
@USBEprom can you please try past versions of the firmware to see whether the problem has always been there or if it's a new issue?

agatti, 2017-07-18:
Incidentally, I tried to read a Microchip 24LC02B with my Bus Pirate v4 and I get the same result both in software and hardware modes, as long as the speed is less than 400kHz (as per datasheet).

USBEprom, 2017-07-19:
@agatti
 
Thank you very much for your help sir!
Since for what I know  the only firmware for the Bus Pirate v3 which has the HARDWARE mode unlocked in the I2C protocol is what that I built by myself I can not test other releases.
The firmwares I talking about are these:
  
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=45#p66033  
(built starting from the repository dated April 13, 2017)
 
and
 
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=60#p66716  
(built starting from the repository dated July 16, 2017)
 
Both two them have the issue I wrote but only while using HARDWARE mode.
Instead when in SOFTWARE mode there are no problems, maybe Bus Pirate v4 behaves different.
In my case the chip is a 24LC16.
In my case the chip is a 24LC16 and it does not matter the speed.
From the datasheet the chip is ranked 400kHz but actually it can also bear 1MHz speed, although for a single time because in HARDWARE mode the bug I wrote sadly it exists.
It should be also considered that the facts that have emerged here are still valid:
 
https://github.com/BusPirate/Bus_Pirate/issues/39
https://github.com/BusPirate/Bus_Pirate/issues/23
 
So the timings for the HARDWARE mode are correct for all the speed into the menu, while for the SOFTWARE mode only the ~400kHz it is supposed to be so since its real value is about half of that (~200kHz)
 
Sorry I know nothing else.
The only thing I can say is that due the HARDWARE mode into the I2C protocol has always been unlocked for the Bus Pirate v4, would be useful to test previous releases, of course factory version too.

agatti, 2017-07-19:
Ok, I'll see what can be done.  It'd help if I could get a logic analyser dump of the problem happening, but not just a screenshot but the signal dump so I could open the file in Logic and take a closer look at the timings and whatnot.

Unfortunately I still don't own a v3 board and the companies that still sell those boards don't seem to want to help its continued development, so it might take some time to sort this out I'm afraid.

We'll look into this for FW version 7.2, as software mode still works and we've been delaying the release for too much time already.

USBEprom, 2017-07-20:
@agatti

OK sir, you are great!
Thanks!!!
Maybe I am wrong and actually it does not matter, but seems to me the problem only exists while doing sequential access.
This because for instance by repeating [0xA0 9][0xA1 r][0xA2 9][0xA3 r] there is not any problem, while by doing the same with [0xA0 0][0xA1 r:37] it appears.
Anyway here you go some of the captures I did:

[LOGIC.zip](https://github.com/BusPirate/Bus_Pirate/files/1163558/LOGIC.zip)

I hope this is exactly what you asked for.
If I misunderstood and it is not what you need, please feel free to let me know so that I can try to provide you the right thing.
Thank you sir!

agatti, 2017-07-20:
@USBEprom yes, the data is in the right format - thank you.  I'll take a closer look at it this weekend.

Marsfan, 2017-07-21:
@agatti have you taken a look at sparkfun? Their board is a v3 with slight tweaks to the layout, and they seem to maintain support on it. I would guess that a majority of v3 boards sold today are because of them.

agatti, 2017-07-21:
@Marsfan "Note: This product is a collaboration with Ian Lesnet. A portion of each sales goes back to them for product support and continued development." guess we should just close down shop then :)

Marsfan, 2017-07-21:
@agatti I don't think I follow. Ian is the original creator of the Bus Pirate, he works for Dangerous Prototypes, so it is logical that the product is licensed to him, I would guess that other companies pay a small portion of their BP sales to dangerous prototypes, if only to support them. I was just suggesting a site that still makes/supports the bus pirate.

agatti, 2017-07-22:
@Marsfan fair enough, I was just sneering at the fact that money was still being paid for "product support and continued development", which in an ideal world would make this very project useless as DP would still fix firmware bugs and the like :)  That said, maybe we're all better off to stick with technical issues only over here ;)

Marsfan, 2017-07-24:
@agatti I would not be surprised if that is something old that they are not aware about. Probably not worth bugging them over, I just wanted to select a good source for getting a  v3. The only BP I have is that model, and it works like a charm with all of @USBEprom's builds.

USBEprom, 2017-07-25:
@Marsfan 

I agree.
Since the PIC used in the Bus Pirate v3 has some problem requiring workarounds in order to gain I2C in HARDWARE mode, in my opinion it is really possible that it is something old that nobody has noticed until now, also because for what I know never it is existed any firmware for the v3 with natively unlocked I2C HARDWARE protocol.
Someone should have compiled the firmware by himself finding the issue.
Anyway due the HARDWARE mode into the I2C protocol has always been unlocked for the Bus Pirate v4, would be useful to know if at least one of the released firmwares it has not the problem.

agatti, 2017-08-05:
@USBEprom can you please try to use the HW I2C on v3 now?  I did some cleanups on those bits of code just now.

USBEprom, 2017-09-02:
Sorry sir, I had and I have problems with the computer and only now I have seen your message.
My bad!
Because of this still now I can not build any new firmware with your changes because the computer where MPLAB is does not work and I have to repair it.
Unfortunately I think it will be a long thing.
I am really very sorry for all this.

agatti, 2017-09-15:
@USBEprom no worries, if hardware mode on v3 doesn't work we'll get reports about that - I guess.  However, if you happen to sort the issues out please give that a try and let us know!

USBEprom, 2017-09-24:
@agatti

I am sorry to annoy.
This afternoon I just got to restore my computer and now MPLAB works again.
Then I tried to compile a new custom firmware from the current repository as it is today September 24, 2017, but unfortunately MPLAB gives me errors.
I have errors even by using original configuration.h provided with the content of the repository and I have verified that by compiling a my backup of the repository dated July 20, 2017, it works, so I guess is not my MPLAB enviroment.
As attachment I put the errors log I get while compiling option 1 and s with MPLABv3.65-XC16v1.31 that I am using.
Thanks in advance.

[Option 1.txt](https://github.com/BusPirate/Bus_Pirate/files/1327551/Option.1.txt)
[Option s.txt](https://github.com/BusPirate/Bus_Pirate/files/1327552/Option.s.txt)

agatti, 2017-09-25:
Thanks for letting me know, I fixed the issue you've been experiencing.  Please give it another shot when you've got the time.

USBEprom, 2017-09-25:
@agatti

Thank you very much Sir, you are the best!
I am sorry to keep annoy though.
With the current September 25, 2017 repository I was able to build new customized firmwares using option 1 and option s but when I change line 382 and 391 inside the configuration.h file in order to unlock hardware mode for the I2C protocol, then MPLAB still gives me errors as shown in attachments:

[O1.txt](https://github.com/BusPirate/Bus_Pirate/files/1330519/O1.txt)
[Os.txt](https://github.com/BusPirate/Bus_Pirate/files/1330520/Os.txt)
[configuration.h.txt](https://github.com/BusPirate/Bus_Pirate/files/1330529/configuration.h.txt)
(Please pay attention to the fact that  in order to put here the configuration.h file I had to rename it as TXT)

Already in the past by myself I have built customized firmwares using the same setting in configuration.h as now and I have never had any problems in this regard.
Sadly I guess there is still some problem somewhere.
Thanks in advance.

USBEprom, 2018-01-10:
@agatti

Ok Sir, right now I am testing new firmware built starting from the latest repository dated January 07, 2018 and it seems that the problem into I2C protocol while in HARDWARE mode has been solved!
Please, take a look at what follows:

i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)

(1)>4
I2C mode:
 1. Software
 2. Hardware

(1)>2
Set speed:
 1. 100KHz
 2. 400KHz
 3. 1MHz
(1)>1
Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'
Ready
I2C>WP
POWER SUPPLIES ON
Clutch engaged!!!
Pull-up resistors ON
I2C>[0xA0 0x00][0xA1 r:256]
I2C START BIT
WRITE: 0xA0 ACK 
WRITE: 0x00 ACK 
I2C STOP BIT
I2C START BIT
WRITE: 0xA1 ACK 
READ: 0x00  ACK 0x01  ACK 0x02  ACK 0x03  ACK 0x04  ACK 0x05  ACK 0x06  ACK 0x07  ACK 0x08  ACK 0x09  ACK 0x0A  ACK 0x0B  ACK 0x0C  ACK 0x0D  ACK 0x0E  ACK 0x0F  ACK 0x10  ACK 0x11  ACK 0x12  ACK 0x13  ACK 0x14  ACK 0x15  ACK 0x16  ACK 0x17  ACK 0x18  ACK 0x19  ACK 0x1A  ACK 0x1B  ACK 0x1C  ACK 0x1D  ACK 0x1E  ACK 0x1F  ACK 0x20  ACK 0x21  ACK 0x22  ACK 0x23  ACK 0x24  ACK 0x25  ACK 0x26  ACK 0x27  ACK 0x28  ACK 0x29  ACK 0x2A  ACK 0x2B  ACK 0x2C  ACK 0x2D  ACK 0x2E  ACK 0x2F  ACK 0x30  ACK 0x31  ACK 0x32  ACK 0x33  ACK 0x34  ACK 0x35  ACK 0x36  ACK 0x37  ACK 0x38  ACK 0x39  ACK 0x3A  ACK 0x3B  ACK 0x3C  ACK 0x3D  ACK 0x3E  ACK 0x3F  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x1F 
NACK
I2C STOP BIT
I2C>[0xA0 0x00][0xA1 r:256]
I2C START BIT
WRITE: 0xA0 ACK 
WRITE: 0x00 ACK 
I2C STOP BIT
I2C START BIT
WRITE: 0xA1 ACK 
READ: 0x00  ACK 0x01  ACK 0x02  ACK 0x03  ACK 0x04  ACK 0x05  ACK 0x06  ACK 0x07  ACK 0x08  ACK 0x09  ACK 0x0A  ACK 0x0B  ACK 0x0C  ACK 0x0D  ACK 0x0E  ACK 0x0F  ACK 0x10  ACK 0x11  ACK 0x12  ACK 0x13  ACK 0x14  ACK 0x15  ACK 0x16  ACK 0x17  ACK 0x18  ACK 0x19  ACK 0x1A  ACK 0x1B  ACK 0x1C  ACK 0x1D  ACK 0x1E  ACK 0x1F  ACK 0x20  ACK 0x21  ACK 0x22  ACK 0x23  ACK 0x24  ACK 0x25  ACK 0x26  ACK 0x27  ACK 0x28  ACK 0x29  ACK 0x2A  ACK 0x2B  ACK 0x2C  ACK 0x2D  ACK 0x2E  ACK 0x2F  ACK 0x30  ACK 0x31  ACK 0x32  ACK 0x33  ACK 0x34  ACK 0x35  ACK 0x36  ACK 0x37  ACK 0x38  ACK 0x39  ACK 0x3A  ACK 0x3B  ACK 0x3C  ACK 0x3D  ACK 0x3E  ACK 0x3F  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  ACK 0x11  A [... cut ...] 8  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x88  ACK 0x8F 
NACK
I2C STOP BIT
I2C>

I do not know how and why but it is really gone, thank you very much Sir!

agatti, 2018-01-10:
The only major change I can see on the I2C subsystem is building with Optimisation Level 1 by default, nothing much else.  Oh well, if it works for you, that's still good to know!

closed 2018-01-10 by agatti