ElderlyPirate/Bus_Pirate #59: I2CEEPROMWIN.c problem.

USBEprom, 2017-07-21:
First to explain the facts I have to make a premise.
I know nothing if this is the right place neither the contents are pertinent, since actually what I will describe is not directly related with the firmware for the Bus Pirate or its developing.
My purpose is not even to solve the matter, the only goal is to report the thing as reference for all.
So said, in order to investigate the weird behavior in the I2C protocol while using HARDWARE mode (https://github.com/BusPirate/Bus_Pirate/issues/54 and http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=8760), I tried to use I2CEEPROMWIN whose its source code is inside the Bus_Pirate-master.zip archive at the path \Bus_Pirate-master\scripts, where exactly there is the C source named I2CEEPROMWIN.c (https://github.com/BusPirate/Bus_Pirate/blob/master/scripts/I2CEEPROMWIN.c and https://github.com/mkschreder/the-bus-pirate/blob/master/scripts/I2CEEPROMWIN.c) from which I built the executable for Windows.
Seems to me I2CEEPROMWIN.exe does not work as expected since always it fails while reading the starting address simply by omitting him and shifting up all the bytes towards the high putting in the end a dummy 0x00 value.
(Follow-up to http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=8763)
Here is an example.
This is the real content of the chip:
 
0x01 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11
0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x11 0x1F
0x02 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22
0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x22 0x2F
0x03 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33
0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x33 0x3F
0x04 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44
[... 69 lines cut ...]
0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x44 0x4F 0x00
 
Also seems to me I2CEEPROMWIN.exe does not work while writing because although no alarm is displayed and indeed in the end it is stated that everything is successfully done, actually the content of the chip has not been changed in any way.
From the source (I2CEEPROMWIN.c) the syntax is this:
 
pc_serial [com?] [chip addr 0-8] [R/W] [file] (read len) (mem addr)
 
That it is not clear as it states that the address must be 2 bytes, while actually using
 
I2CEEPROMWIN.exe com3 0 R TEST 2048 0000
I2CEEPROMWIN.exe com3 0 W TEST 2048 0000
 
or
 
I2CEEPROMWIN.exe com3 0 R TEST 2048 00
I2CEEPROMWIN.exe com3 0 W TEST 2048 00
 
it does not make any difference for the result.
More, [chip addr 0-8] seems to make no sense because available blocks should be from 0 to 7, not from 0 to 8.
In fact 24xx64 has a memory capacity of 8 contiguous blocks (0-7) and data bus allow for only 8 chip (0-7) for contiguous addressing across multiple devices.
That is really weird at me.
Maybe the fault is due the syntax I use that is wrong, maybe it does not work anyway.
I think that I2CEEPROMWIN can be useful for debug simplifying it through the automation of test procedures, for instance by writing and reading of massive amount of data.
In order to deeper the matter I wrote at the beginning of this message I needed to make the chip content unequivocally so that it was easy to detect anomalies.
I2CEEPROMWIN would be a great choice but instead I had to fold on BusPirate_Control and scripting in order to reach the goal.

dme8a8, 2019-11-26:
@USBEprom 
Can you please share the solution with BusPirate_Control.

Right now, there is no way to write or reliably read, I2C EEPROMs with the BusPirate.

USBEprom, 2020-10-23:
**@dme8a8**

I mean something like the one that follows using the I2C protocol directly from the terminal with the Bus Pirate:

I2C>[160 0][161 r:256][162 0][163 r:256][164 0][165 r:256][166 0][167 r:256][168 0][169 r:256][170 0][171 r:256][172 0][173 r:256][174 0][175 r:256]

(Reading whole chip 24LC16)

USBEprom, 2020-11-05:
I took the trouble to analyze the behavior of the program with the help of the logic analyzer loaned to me by the usual friend of mine and I found the following.
Once started, the program immediately initializes the Bus Pirate preparing it for reading / writing according to what is set in the command line, **but it waits 8 minutes and 20 seconds to start reading / writing the chip!**
So actually the program is not as slow as it seems, it just waits for a long time before starting to do its job.
It is not clear to me why such a long delay has been introduced (**8 minutes and 20 seconds!**), probably a typo or I do not know what else (could it be the compiler I use ? [MinGW & MinGW-W64], I do not think so, but I can not rule it out either and however, the timings are exact to the logic analyzer, so it remains a mystery to me).
Decoding the I2C traffic with the logic analyzer it is obvious that, actually as already written by the author James Stephenson in his notes inside the C source, **a 2 bytes addressing is used.
This fact explains the behavior I detected when reading/writing a 24LC16 EEPROM, as this chip uses 1 byte addressing.**
I do not have a suitable chip available to verify the code but I must believe that it works correctly with chips that support 2 bytes addressing, as in the dangerousprototypes forum more than one user wrote that they used it without problems, even if there is the issue of 8 minutes and 20 seconds of waiting, which however may have been attributed to low clock rather than to a fixed delay:

http://dangerousprototypes.com/forum/index.php?topic=4791.msg48475#msg48475

http://dangerousprototypes.com/forum/index.php?topic=2480.msg24148#msg24148

https://www.eevblog.com/forum/testgear/yet-another-tonghui-th2822a-lcr-meter-review-in-pictures/msg384261/#msg384261

So I took a look at the C source of the program, even if I totally have no skills as a programmer, and I noticed that on line 92 the command sleep (500) is entered.
**500 seconds = 8 minutes and 20 seconds, bingo!**
That line is responsible for the 8 minutes and 20 seconds delay I mentioned in the opening, so I first modified it for 5 seconds of waiting and finally considering that time value too much, for a total of 3 seconds, that is:

`92     sleep (3)`

After that, I took care of line 24.
In fact I had written in the first message that it is not clear at me what is meant with the phrase "the address must be 2 bytes", but now that it is I chose to change the related message in the command line.
Then, since as I already wrote at the beginning the wording **[chip addr 0-8]** does not seem to make sense to me because in chips like 24FC64 the available blocks should be 8 numbered from 0 to 7 and not 9 numbered from 0 to 8, I also modified this message in line 24 and the code concerning it in line 28:

`24     printf("pc_serial [com?] [chip addr 0-7] [R/W] [file] (read len) (2 bytes mem addr)\n");`

`28     if(argv[2][0] > '7' || argv[2][0] < '0'){`

Once finished editing the C source, I compiled it to obtain the executables in 32bit and 64bit format that I enclose with the original and modified sources.
As far as I can understand, the executables I have compiled work, but since I do not have adequate chips I can not fully test the result, if someone who can do it wants to try them and then report his impressions he is welcome, thanks!

[I2CEEPROMWIN_NEW_3s.zip](https://github.com/BusPirate/Bus_Pirate/files/5497758/I2CEEPROMWIN_NEW_3s.zip)

USBEprom, 2025-07-13:
**@mikebdp2**

Hi mikebdp2.
Many, many thanks for the news, Sir!
About your improved version of **I2CEEPROMWIN.c** (**https://github.com/mikebdp2/Bus_Pirate/blob/v7.1-09072025/scripts/I2CEEPROMWIN.c**) with what compiler and with what options do You build the executable for Microsoft operating systems? (I have never tried with the Linux version)
I ask for it because even with Your updated version, by building the executable program I am facing the same exact issue I had described here

**https://github.com/BusPirate/Bus_Pirate/issues/59#issuecomment-722677905**
**http://dangerousprototypes.com/forum/index.php?topic=8763.msg71440#msg71440**

and for which I had proposed my solution.
I wrote that I suspected of the compiler I used (**MinGW** and **MinGW-W64**), but now I tried with **Embarcadero** and the defect is still there.
Surely the fault is mine, I am not a firmware engineer, but if I do not change the **line 103** as I described, then the executable that I build from the new source code **I2CEEPROMWIN.c** waits 8 minutes and 20 seconds to start reading / writing the chip.
My solution is still to change **line 103** from

103	sleep(**500**);

to

103	sleep(**3**);

in order to start by waiting only 3 seconds (which seems to me a reasonable time).
Any ideas about it?
Thanks in advance.

Be seeing you.

U.Sb