ElderlyPirate/Bus_Pirate #61: Terminal returning garbage characters

rehelmin, 2017-09-02:
Hello,

I just loaded the latest firmware today (commit id: feb479a056f1cf2559929a64e36c710a162c64f8) and when I open up the terminal I am getting mostly garbage characters returned from a BPv4.

[image]

I am running the screen terminal emulator on Mac OSX El Capitan.

Any idea what might be causing this? Rolling back to the 7.0 release seems to make everything happy again.

Thanks,
Ryan

agatti, 2017-09-03:
Hrmm, does the commit before that one exhibit the same issue?  Things seem to work fine here, but I'll look into it once more.

azolotko, 2017-09-07:
Having the same issue. I noticed that my hex files produced in `Firmware/busPirate.X/dist/BusPirate_v4/production` are much smaller (346K) than those in `package/BPv4-firmware` (962K). Could it be the case that we've missed some build step?

azolotko, 2017-09-08:
c7776d5 is okay, and 6cea461 is not.

agatti, 2017-09-08:
Thanks a lot for the feedback, I'll look at it right away!

agatti, 2017-09-09:
@andersm is there anything that we are missing in your PR #60 ?  Seems like it might require a bit more work and/or figuring out whether certain settings are a prerequisite for the code to work.

andersm, 2017-09-10:
I hadn't realized how XC16 handles function pointers on parts with more than 64KiB flash. I'll try to see if this can be worked around, otherwise the change will have to be reverted.

A dirty workaround might be to modify the linker script to place all strings within the first 64KiB.

agatti, 2017-09-10:
Thanks, @andersm!  @azolotko, @rehelmin: can you please give this change a try and see if it makes things work for you?

azolotko, 2017-09-11:
I've just tried 102260a and it looks good to me. Thanks.

PR #62 (2017-09-11): Use full addresses for printing constant messages
On parts with more than 64KiB flash, function pointers are implemented
using 16-bit "handles". Use __builtin_tbladdress() to pass the full
address to the message printing functions.

Remove unnecessary index arithmetic from bp_message_write_buffer().

Fixes #61.
commit c6097f3e (2017-09-11): Use full addresses for printing constant messages (#62)

On parts with more than 64KiB flash, function pointers are implemented
using 16-bit "handles". Use __builtin_tbladdress() to pass the full
address to the message printing functions.

Fixes #61.
closed 2017-09-11 by agatti