ElderlyPirate/Bus_Pirate #64: Deprecate usage of ds30 GUI for bootloader updates?

agatti, 2017-09-21:
Seems like that ds30 gui is the only option for updating bootloaders but there are three issues that need to be addressed:

1. Do we have the licence to redistribute the binary version of ds30 gui?
2. Reflashing the bootloader on Linux or on macOS might not be possible with a windows-only binary.
3. Downloading an updated version of the free ds30 gui version requires registration to a forum.

The plan here is to find a cross-platform alternative for flashing bootloaders that's not MPLAB-X IPE with a PicKit adapter.

psolyca, 2017-09-21:
Hey

Am I wrong but there is also http://dangerousprototypes.com/docs/Pirate-Loader_console_upgrade_application_(GNU/Linux,_Mac,_Windows) to update bootloader with a CLI application.
It is only for v4+ bootloader.

agatti, 2017-09-21:
Can you replace the bootloader with that tool?  I seem to recall you cannot do that from pirate-loader.

psolyca, 2017-09-21:
I have fast read, it is only for firmware updates.

agatti, 2017-09-21:
No worries.  I wonder whether we've got to replicate the updater ourselves by figuring out the ds30 protocol or not.

satscan, 2017-09-21:
Hi Alessandro,

you can download the latest free edition of ds30 gui (ds30_Loader_140817
zip) from MicroChip.

http://www.microchip.com.tw/modules/newbb/dl_attachment
php?attachid=1477460000&post_id=73379 
 
Best regards,
Marc 
 
 
-------Original Message-------
 
From: Alessandro Gatti
Date: 9/21/2017 10:20:02 AM
To: BusPirate/Bus_Pirate
Cc: Subscribed
Subject: [BusPirate/Bus_Pirate] Deprecate usage of ds30 GUI for bootloader
updates? (#64)
 
Seems like that ds30 gui is the only option for updating bootloaders but
there are three issues that need to be addressed:
Do we have the licence to redistribute the binary version of ds30 gui? 
Reflashing the bootloader on Linux or on macOS might not be possible with a
windows-only binary. 
Downloading an updated version of the free ds30 gui version requires
registration to a forum. 
The plan here is to find a cross-platform alternative for flashing
bootloaders that's not MPLAB-X IPE with a PicKit adapter.

agatti, 2017-09-21:
Interesting.  Searching for "ds30 gui" on google, the microchip link doesn't show up but picbootloader.com has the software for download behind a forum registration.

agatti, 2017-09-21:
Ok, the binaries still don't come with a licence, so I'm not sure we can actually distribute these as well :(

satscan, 2017-09-21:
ds30 loader is open Source under GNU Public License v2.The  COPYING.txt is just missing from the \ directory. It is present in the three firmware directories.

agatti, 2017-09-21:
Yes, although the problem here is for the two .exe files in the archive.  Same issue for Bootloaders/BPv3-bootloader/ds30_Loader_GUI in the source tree.

psolyca, 2017-09-21:
Ok, I found some things to try but really do not what we need, I am really an end user.
My source http://www.microchip.com/forums/m515696.aspx
Compiling under *nux : https://embeddedfreak.wordpress.com/2008/10/10/compiling-mplab-c30-v311b-under-linux/
Or use Eclipse as an IDE : https://github.com/ecdpalma/piccbuilder (too old)
To upload the bootloader maybe this one : http://piklab.sourceforge.net/
I do not know for this one : https://gputils.sourceforge.io/

psolyca, 2017-09-24:
It seems, USBEprom manage to upload a new bootloader with pirate_loader.exe! See #38

USBEprom, 2017-09-28:
@psolyca 

That is correct:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67056#p67056

agatti, 2018-09-05:
Moving to 7.2...