ElderlyPirate/Bus_Pirate #65: Binary Mode Python I2C Sniffer

pacmac, 2017-09-21:
Really hoping someone can help, I am able to sniff in console mode and send and receive commands no problem with binary mode and the python scripts, but how do I go about reading the sniffed output in python ??

I know that 0x0f (15) calls the sniffer mode, but how can I read (and subsequently decode & format) the returned data packets ?

```
def sniff():
  i2c.port.write(b"\x0F")
  while True:
    print(i2c.response())
```

I tried the above but it only outputs loads of lines of a single zero, followed by carriage return ?

agatti, 2017-09-22:
Hi there,

  I'm a bit strapped for time at the moment, but in the meantime you may want to take a look at the code that handles the sniffing: https://github.com/BusPirate/Bus_Pirate/blob/c6fa09018cba7b776766b7211c8c4a82e8e8a426/Firmware/i2c.c#L793 - keep in mind that it is invoked in non-interactive mode.

pacmac, 2017-09-22:
Thanks so much for responding, but unfortunately I know nothing about C, my capabilities are limited to scripting (javascript, python) .

but I have made some progress although there appears to be some lost bytes which messes up my ability to extract the address and data bits from the response reliably.

agatti, 2017-09-25:
Do those bytes get lost even if you sniff data without using your scripts and by manually triggering the I2C sniffer via a terminal on a serial port?

pacmac, 2017-09-25:
No using the console sniff I can copy and paste the output into a text editor and and it seems to be complete.

My work around is always to read 18 bytes, since I know that the device I am sniffing uses 18 bytes for MOST of it's responses.

But what I am really looking for is some code that will convert the binary data into a format that I can extract the bytes I need for further processing.

agatti, 2017-09-25:
Well, the lost bytes may be caused by something on your end since the I2C sniffer code is exactly the same for both human-readable and raw binary output.

Can you please provide some sample data you read that you cannot extract the bytes from?

pacmac, 2017-09-25:
Thanks, I don't think there is any issue with the bus pirate, but what I am looking for is how to decode the data that is returned in binary format to 'packets' of data starting with the start bit [ and ending with the end bit ]

If I can capture this into a list or array using python then I can extract the data I need !

agatti, 2017-09-25:
You mean something like this (completely untested):

```python
def sniff():
	i2c.port.write(b'\x0F')
	in_packet = False
	current_packet = None
	while True:
		byte = i2c.response()
		if byte == b'[':
			if in_packet:
				# missed data
				pass

			current_packet = []
			in_packet = True
			continue
		elif in_packet:
			if byte == b'\\':
				escaped = True
			elif byte == b'-':
				# NACK received
				pass
			elif byte == b'+':
				# ACK received
				pass
			elif escaped:
				current_packet.append(byte)
				escaped = False
			continue
		elif byte == b']':
			if escaped:
				# missed data
				pass
			in_packet = False

			# packet read

			packet_buffer = b''.join(current_packet)
		elif byte == b'+':
			# ACK received
			pass
		elif byte == b'-':
			# NACK received
			pass
		else:
			# missed data
			pass
```

?

pacmac, 2017-09-25:
This is a sample of the data copied from the console sniff mode that I would also like to capture using binary mode in python:

`[0x16+0x08+[0x17+0xB5+0x0B-][0x16+0x09+[0x17+0x5D+0x20-][0x16+0x0A+[0x17+0x00+0x00-][0x16+0x10+[0x17+0x14+0x06-][0x16+0x0F+[0x17+0x13+0x00-][0x16+0x0D+[0x17+0x01+0x00-][0x10+0x00+0x01+0xA0+][0x16+0x16+[0x17+0x80+0x02-][0x16+0x08+[0x17+0xB5+0x0B-][0x16+0x09+[0x17+0x5D+0x20-][0x16+0x0A+[0x17+0x00+0x00-][0x16+0x10+[0x17+0x14+0x06-][0x16+0x0F+[0x17+0x13+0x00-][0x16+0x0D+[0x17+0x01+0x00-][0x10+0x00+0x01+0xA0+][0x16+0x16+[0x17+0x80+0x02-][0x16+0x08+[0x17+0xB5+0x0B-][0x16+0x09+[0x17+0x5D+0x20-][0x16+0x0A+[0x17+0x00+0x00-][0x16+0x10+[0x17+0x14+0x06-][0x16+0x0F+[0x17+0x13+0x00-][0x16+0x0D+[0x17+0x01+0x00-][0x10+0x00+0x01+0xA0+][0x16+0x16+[0x17+0x80+0x02-][0x16+0x08+[0x17+0xB5+0x0B-] `

I would like to be able to see the same data as I see in the console in python.

agatti, 2017-09-25:
Ok, can you please capture the raw bytes as you read them via

```python
def sniff():
  i2c.port.write(b"\x0F")
  while True:
    print(hex(ord(i2c.response())))
```

or something like that?  Whilst the bytes being sent over the serial port and the raw bytes are the same, things are framed differently.

pacmac, 2017-09-25:
Really appreciate your help with this, but it's 10:30pm now, here and I can only really hook up the sniffer in the morning, I will use your code sample and let you know what is printed.

pacmac, 2017-09-26:
When I try and use your sniff raw byte routine, I get an error:

```
print(hex(ord(i2c.response())))
TypeError: ord() expected string of length 1, but int found
```

I tried to catch the error, but then nothing is printed:

```
def sniff():
  i2c.port.write(b"\x0F")
  while True:
    res = i2c.response()
    try:
      print(hex(ord(res)))
    except:
      pass
```

pacmac, 2017-09-26:
I modified the function and get this:

```
def sniff():
  i2c.port.write(b"\x0F")
  while True:
    res = i2c.response(1,True)
    try:
      print(hex(ord(res)),end=", ")
    except:
      pass
```

```
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0x16, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0xc0, 0x2b, 0x5c, 0x0, 0x2d, 0x5d, 
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0x8, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0x30, 0x2b, 0x5c, 0xc, 0x2d, 0x5d, 
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0x9, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0xea, 0x2b, 0x5c, 0x1f, 0x2d, 0x5d, 
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0xa, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0x2, 0x2b, 0x5c, 0x0, 0x2d, 0x5d, 
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0x10, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0xb8, 0x2b, 0x5c, 0xb, 0x2d, 0x5d, 
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0xf, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0xb8, 0x2b, 0x5c, 0xb, 0x2d, 0x5d, 
0x5b, 0x5c, 0x16, 0x2b, 0x5c, 0xd, 0x2b, 0x5b, 0x5c, 0x17, 0x2b, 0x5c, 0x64, 0x2b, 0x5c, 0x0, 0x2d, 0x5d, 
0x5b, 0x5c, 0x10, 0x2b, 0x5c, 0x0, 0x2b, 0x5c, 0x1, 0x2b, 0x5c, 0xa0, 0x2b, 0x5d,
```

I have added carriage returns to the end of each packet to make it easy for you to see, but how do I know which are values and which are format chars ?

pacmac, 2017-09-27:
Thanks so much for your help, I could not see initially how to detect a value from a formatter character, but now realise that a value byte is escaped (//) and it is all now working as expected.

Thanks again.

closed 2017-09-27 by pacmac