ElderlyPirate/Bus_Pirate #68: Unify pirate-loader code to handle both v3 and v4

agatti, 2017-09-25:
Right now there are two separate pirate-loader codebases, one for each model.  Ideally there should be only one tool to be used for uploading both firmware types.

kallisti5, 2017-09-25:
I assume this is about:

  * Bootloaders/BPv3-bootloader/pirate-loader
  * Bootloaders/BPv4-bootloader/pirate-loader

A quick diff shows that the BPv4 pirate-loader only seems to be a slightly newer version of the BPv3 pirate-loader.

```
diff -Naur BPv4-bootloader/pirate-loader/pirate-loader.c BPv3-bootloader/pirate-loader/pirate-loader.c
```

Are we sure the pirate-loader in _Bootloaders/BPv4-bootloader/pirate-loader_ doesn't work with the BPv3?  The only requirement seems to be "v4+ bootloader".

It appears there are special steps + code to upgrade the v2 bootloader to v4.

From what i've gathered...

* BPv3
  * bootloader v4.1  (Bootloaders/BPv3-bootloader/firmware-v4.1)
  * bootloader v4.2  (Bootloaders/BPv3-bootloader/firmware-v4.2)
  * bootloader v4.4  (Bootloaders/BPv3-bootloader/firmware-v4.4)  (SparkFun release version)
* BPv4
  * bootloader v1.0 (Bootloaders/BPv4-bootloader/firmware-v1) (Seeed studio release version?)

agatti, 2017-09-25:
Technically there's v4.5 as well, which includes the jump to bootloader fix.  However, I have no idea on whether BPv4's pirate-loader works on v3 and whether that can flash bootloaders too.  If that's the case, then we should move pirate-loader from Bootloaders/BPv4-bootloader/pirate-loader to tools/ and remove other copies.

kallisti5, 2017-09-25:
Correct. Tomorrow I will:

* Downgrade my BPv3 + BPv4 to as close to the original release as I can using the PicKit 2
* Attempt upgrades on both purely using the latest pirate-loader.

Any additional steps?  Since both came with the latest bootloader (from what I can tell) it seems like all I need to do is drop the firmware version to pre-v6.2

It sounds like we'll need a documented workaround of people needing a PicKit if anyone updated beyond the Dangerous Prototypes v6.1 release on the BPv3 or BPv4?

agatti, 2017-09-25:
For v4 there should be no real workaround needed I think, but for v3 things can get hairy pretty quickly and bricking the device is a real possibility, so yes, some documentation is needed I think.

As far as additional steps go, short of trying to upgrade v3's bootloader to v4.5 from all released bootloader versions maybe there isn't anything left to do - but this is something that can be done later.

kallisti5, 2017-09-25:
> Technically there's v4.5 as well for v3 and v1.1 for v4, which include the jump to bootloader fix.

I updated the list of bootloaders above.  Do we want to bump the bootloader versions? I didn't see it done in e2bd0a093f3030c7818a9e38a804596fe50343bf or 2bfccf2c027c1bf3ffc28082cb0c6b809b43901c

I assume the BL version is the VER_H and VER_L in bootloader.c for BPv4?  (4 + a,  4.10  , 4.1  )

agatti, 2017-09-25:
Ouch, sorry for not bumping the v4 bootloader version - please bump it up then.

On the other hand, the v3 bootloader directory needs a good scrub for sure!

kallisti5, 2017-09-25:
ok. working on a cleanup commit now.

kallisti5, 2017-09-25:
https://github.com/BusPirate/Bus_Pirate/pull/69

tldr:

  * BPv4-bootloader has the latest BPv4 bootloader
     * BPv4 pirate-loader moved to Bootloaders/pirate-loader
  * BPv3-bootloader has the latest BPv3 bootloader
    * old has older bootloaders + the unknown upgrade stuff

agatti, 2017-09-25:
Awesome, let's merge the whole lot once the firmware/bootloaders update tests prove successful then.

kallisti5, 2017-09-25:
Sounds good.  I'll append the version bumps in that PR as well.

USBEprom, 2017-10-27:
@kallisti5

For what I know and experimented pirate-loader does work with Bus Pirate v3 by allowing me to upgrade bootloader from v4.4 to v4.5:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=75&sid=c72e26d3dde472fbd8ab7226df6d3db8
 
http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=75&sid=c72e26d3dde472fbd8ab7226df6d3db8#p67127

Just out of curiosity.

kallisti5 wrote:
> bootloader v4.5 (Bootloaders/BPv3-bootloader/firmware-v4.5) (SparkFun release version)

About bootloader 4.5, what is SparkFun release version?
Truthfully speaking I have never heard of it and I would love to know more.
Is there somewhere its code?
Thanks.

agatti, 2018-09-05:
Moving to 7.2...