ElderlyPirate/Bus_Pirate #71: Starting Bus Pirate with customized baud rate, how?

USBEprom, 2017-10-06:
Hi guys, sorry for disturbing.
As mere curiosity, wanting to set a different customized default baud rate for the terminal how is it supposed to do it?
Normally the default baud rate for the Bus Pirate is set at 115200bps to communicate via terminal.
Now, if for instance wanting to set it to 2000000bps, what should be done to achieve the goal?
I have tried to build a custom firmware that has 2000000bps as default speed for the terminal, sadly without success though.
I have seen that the necessary parameters are inside source code in line 171 of **main.c** and line 247 of **base_io.c**, so I changed the latter as follow:

247  1 /* 2000000 bps */

I do not care that labels into the menu of the Bus Pirate are consistent (it would be better if they were but right now I can wait for that), only that the Bus Pirate starting itself at 2000000bps as default setting instead of the usual 115200bps, but my edit does not work.
The firmware is compiled correctly but then it is not possible to communicate with the Pirate Bus through the terminal even though the COM port is recognized right.
Luckily enough it was possible to put the right firmware by tying PGD and PGC on the Bus Pirate!
Any hint?
Thanks in advance!

jevinskie, 2017-10-06:
I wrote a Python script to automatically generate the UART config register values for baud rates > 115200. I was able to reliably use 1.5 Mbps but I had to hook up (using a short fly wire) and configure the PIC to use the 16 MHz clock output from the FTDI chip. Using the internal RC oscillator on the PIC will cause too much slop in the UART timing between the FTDI and PIC otherwise.

https://github.com/jevinskie/bus-pirate-old/commits/master

agatti, 2017-10-07:
Indeed, 2Mbps might be too fast for the PIC to handle.  On the other hand, I wonder about the usefulness of having the UART going that fast, since the Bus Pirate was never meant to handle high speed data to begin with.  If you really have to sample data sources that spew out data that fast, I'd say you're better off with a hardware logic analyser rather than with a dedicated circuit hooked up to the data source directly.

USBEprom, 2017-10-07:
@jevinskie

Thanks for the reply, me too I thought about scripting.
The whole matter is this:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=1776#p67086

In that kind of scenario custom firmware would be better than use scripting that are depending on other programs and languages as well as on the operating system working on the computer.

Talking about something else.
I looked at the link you wrote and I saw something about #59 (http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=8763).
Maybe you can suggest something about that.

@agatti

Thanks for your reply Sir.
It is not me, see above.
Me too I am of the opinion that since baud rate entry from the terminal is possible, then it is not really best to change the source. 
However it is not a matter of bearable/unbearable or useful/unuseful speed.
2000000bps is just one example, I could write 19200bps and it would have been the same thing.
The point is understand how can be set the default baud rate with which the Bus Pirate starts its terminal.
That is the goal of my question.

agatti, 2017-12-24:
@USBEprom you can try to change `bus_pirate_configuration.terminal_speed` in main.c:171 to a value between 0 and 9.  That value is the index inside `UART_BRG_SPEED` in uart.c:55, so if you want to add your own baud rate just add a new entry in that table and update `bus_pirate_configuration.terminal_speed` accordingly.  Keep in mind that the Bus Pirate v3 uses an external USB to UART adapter so certain custom baud rates may or may not be working at all.

USBEprom, 2017-12-24:
@agatti

Thanks for your kind reply Sir.
You are teaching a lot to me and many others.
I will try your hints and i will let you know.
Thanks again and merry Christmas!

USBEprom, 2018-01-04:
HNY Sir.
Sorry for the late.
I did a test and it seems to work just like you wrote, but only with standard baud rates (300bps, 57600bps, 38400bps and so on), using the not standard value 2000000bps it does not work.
However maybe it is my fault because I should try with lower non-standard values, something like 250000bps (or leven less) for example.
2000000bps could be really too many.
I used the latest repository and with the premise I wrote, it works.
Sadly still #78, but it works.
As always thank you very much Sir, you are the greatest!

USBEprom, 2018-01-06:
OK Sir, as always I am babo while you are 100% right, so you won again!
Sadly there is still #78, but that is a different problem, I confirm that what you wrote is totally correct and working!
I tried many not standard values and always it works, of course even 2000000bps (this latest one is really very fast, much faster compared to standard 115200bps, response of the terminal has significantly improved a lot!!!).
The trick is to do not use the last setting in uart.c (9) but only the previous ones (0-8) to insert the new value, so that line 171 in main.c point it accordingly.
In the end simply it is need to change in uart.c the value wrote in one of the lines between 56 and 64 with the new wished one and accordingly in main.c fill the setting with the corresponding one item number referred to the position of the line containing the just filled new value.
I do not know why but using item number 9 in line 171 of main.c and therefore by  changing line 65 in uart.c, then it does not work, this is not a problem though.
Simply follow the rule of not use item 9 in main.c or aim to change line 65 in uart.c, that is all.
At this point for me the issue can also be closed.
Thank you Sir, you are the best!

closed 2018-01-06 by agatti