ElderlyPirate/Bus_Pirate #74: Errors while compiling for Bus Pirate v3 and mismatched messages.

USBEprom, 2017-11-10:
I am sorry to keep annoy but while compiling for Bus Pirate v3 using the current November 07, 2017 repository still I have same problems as I wrote here:

https://github.com/BusPirate/Bus_Pirate/issues/54#issuecomment-331978946

Plus a dangerousprototypes' forum user noticed that some message into SPI protocol are wrong:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67167#p67164

In detail the thing is that by performing some commands while in SPI mode then the Bus Pirate v3 use wrong messages on the terminal:

SPI>[0x9f r r r]
**/CS ENABLED**
WRITE: 0x9F
READ: 0xC2
READ: 0x20
READ: 0x14
**\CS ENABLED**
SPI>

It is only a cosmetic issue because actually, despite the wrong message, everything works as expected, but it is really annoying
For what I know the problem started with repositories dated after the April 10, 2017 because some custom firmware I build starting from that repository and I released on the dangerousprototypes' forum do not have it:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67167#p67167

Thanks.

agatti, 2017-11-12:
@USBEprom I fixed the compilation error, but I wonder why it never triggered on my installation.  Maybe from now on you may want to use a more modern version of MPLAB-X, v4.05 has been available for a week or two now.

agatti, 2017-11-12:
@USBEprom the backslash issue was fixed in 47f5c59d9c92ee2ace8bea40b985722f8d71aee5 a couple of months ago, by the way.

I'm closing this since both issues seem to have been addressed, please reopen if those issues persist, thank you.

closed 2017-11-12 by agatti

USBEprom, 2017-11-12:
@agatti

Thank you Sir, as usually you are the best!
Now it is possible to compile without errors but sadly still does not works because then there are not setting into menus' items as explained here:

https://github.com/BusPirate/Bus_Pirate/issues/54#issuecomment-331978946

However as you suggested me I am upgrading to MPLABX-v4.05 and xc16-v1.33 in order to exlude their fault.
I will send the result, thanks!

USBEprom, 2017-11-19:
As promised today I was able to set a Linux environment (Ubuntu LTS 14.04.3) where I installed MPLABX_v4.05 and xc16_v1.33 both fully working.
Using it I have built test firmwares starting from the current repository then sadly falling into the same problem described here:

https://github.com/BusPirate/Bus_Pirate/issues/54#issuecomment-331978946

The firmwares are compiled without error but the problem still arises and there are not setting into menus' items, almost them all are without options.
In the end it does not depend on obsolete MPLABX or XC16, the culprit has to be something else wrong that I have to believe is in the repository itself.

USBEprom, 2017-11-19:
A tiny news, if it can be useful.
Actually option items are not totally gone, in fact by using the command "W" [PSU (off/ON)] them all are there.
Seems to me that the warning 

**Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'**

is triggered bad at the wrong time so then by sending the command "W" as suggested by the spurious message not only power supplies are switching ON but the missing options are shown on the terminal.
Here it is a simple example which explain the issue:

i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)

(1)>5
**Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'**
Ready
SPI>**W**
POWER SUPPLIES ON
Set speed:
 1.  30KHz
 2. 125KHz
 3. 250KHz
 4.   1MHz
 5.  50KHz
 6. 1.3MHz
 7.   2MHz
 8. 2.6MHz
 9. 3.2MHz
10.   4MHz
11. 5.3MHz
12.   8MHz

(1)>
Clock polarity:
 1. Idle low *default
 2. Idle high

(1)>
Output clock edge:
 1. Idle to active
 2. Active to idle *default

(2)>
Input sample phase:
 1. Middle *default
 2. End

(1)>
CS:
 1. CS
 2. /CS *default

(2)>
Select output type:
 1. Open drain (H=Hi-Z, L=GND)
 2. Normal (H=3.3V, L=GND)

(1)>
Clutch engaged!!!
SPI>w
Clutch disengaged!!!
POWER SUPPLIES OFF
SPI>**W**
POWER SUPPLIES ON
Set speed:
 1.  30KHz
 2. 125KHz
 3. 250KHz
 4.   1MHz
 5.  50KHz
 6. 1.3MHz
 7.   2MHz
 8. 2.6MHz
 9. 3.2MHz
10.   4MHz
11. 5.3MHz
12.   8MHz

(1)>

I am going to open a new issue that match better the real behaviour of the thing, just in case.
Thanks.

agatti, 2017-11-20:
Ok, I'll close this one and move onto the other report, thank you.