ElderlyPirate/Bus_Pirate #75: help with bulk read

diginfo, 2017-11-16:
I am struggling to understand bulk_trans, I thought that I could use it to read multiple bytes by repeatdly calling read_byte() in a loop but it only returns the first 3 values and then repeats the 3rd value and does not appear to read any more.

Would appreciate if anyone can help with this ?

```
def get_bytes(address, numbytes, ret=False):
  i2c.send_start_bit()
  i2c.bulk_trans(2, [0x16, 0x19]) ## WR ADDRESS
  i2c.send_stop_bit()
  
  out = []
  i2c.send_start_bit()
  i2c.bulk_trans(1, [0x17]) ## RD ADDRESS
  
  data_len = 20
  while(data_len):
    out.append(ord(i2c.read_byte()))
    if data_len > 1:
      i2c.send_ack()
    data_len-=1
  i2c.send_nack()
  i2c.send_stop_bit()

  print(out)
```

agatti, 2017-11-16:
Does the same thing happen when manually issuing the commands from the Bus Pirate when connected via a terminal program?

agatti, 2017-12-18:
Closing due to lack of response.

closed 2017-12-18 by agatti