ElderlyPirate/Bus_Pirate #78: Bus Pirate v3 and mismatched and hidden messages.

USBEprom, 2017-11-19:
With reference to issues 54 and 74:

https://github.com/BusPirate/Bus_Pirate/issues/74
https://github.com/BusPirate/Bus_Pirate/issues/54#issuecomment-331978946

Actually option items are not totally gone, in fact by using the command "**W**" [PSU (off/ON)] them all are there.
Seems to me that the warning 

**Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'**

is triggered bad at the wrong time so then by sending the command "**W**" as suggested by the spurious message not only power supplies are switching ON but the missing options are shown on the terminal.
Here it is a simple example which explain the issue:

i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. LCD
10. PIC
11. DIO
x. exit(without change)

(1)>5
**Clutch disengaged!!!
To finish setup, start up the power supplies with command 'W'**
Ready
SPI>**W**
POWER SUPPLIES ON
Set speed:
 1.  30KHz
 2. 125KHz
 3. 250KHz
 4.   1MHz
 5.  50KHz
 6. 1.3MHz
 7.   2MHz
 8. 2.6MHz
 9. 3.2MHz
10.   4MHz
11. 5.3MHz
12.   8MHz

(1)>
Clock polarity:
 1. Idle low *default
 2. Idle high

(1)>
Output clock edge:
 1. Idle to active
 2. Active to idle *default

(2)>
Input sample phase:
 1. Middle *default
 2. End

(1)>
CS:
 1. CS
 2. /CS *default

(2)>
Select output type:
 1. Open drain (H=Hi-Z, L=GND)
 2. Normal (H=3.3V, L=GND)

(1)>
Clutch engaged!!!
SPI>w
Clutch disengaged!!!
POWER SUPPLIES OFF
SPI>**W**
POWER SUPPLIES ON
Set speed:
 1.  30KHz
 2. 125KHz
 3. 250KHz
 4.   1MHz
 5.  50KHz
 6. 1.3MHz
 7.   2MHz
 8. 2.6MHz
 9. 3.2MHz
10.   4MHz
11. 5.3MHz
12.   8MHz

(1)>

Thanks.

agatti, 2018-01-07:
This should have been fixed with 1070a0ec0efcab3d9d87a637838a8223785877af.  Can you please check if it works for you?

USBEprom, 2018-01-08:
Thank you very, very much Sir, you are great!
I confirm that it is gone!!
Terrific job, well done Sir!!!

closed 2018-01-08 by agatti