ElderlyPirate/Bus_Pirate #79: Document how to build and flash

sajattack, 2018-01-07:
I see no documentation for building this firmware and flashing it to a buspirate.

Marsfan, 2018-01-07:
I cannot speak regarding building firmware (I have no experience with PICs, and I have no clue if [this ](http://dangerousprototypes.com/docs/Compile_the_Bus_Pirate_firmware)doc on the DP website still works for building). But I know that [this](http://dangerousprototypes.com/docs/Pirate-Loader_console_upgrade_application_(Linux,_Mac,_Windows)) document on the DP website covers the proper method for flashing the the bus pirate.

sajattack, 2018-01-07:
Yeah I found the document on flashing online. If that document for compiling is accurate, that's terrible UX. Why can't there just be a makefile?

Marsfan, 2018-01-07:
No clue. I don't do development, I am just an end-user with a github account.

JarrettR, 2018-01-07:
Because Microchip.

You will notice that MPLAB X generates makefiles that are presumably called when hitting "Build" from inside the IDE.
You're welcome to play around with those makefiles and see if they work, and document if it does or what changes must be made to fix it. At the very least, you will still need to install the XC16 compiler.

I'd be curious about this too. I have MPLAB X set up already, so the UX as documented there is fine, but being able to run from Bash does allow some more interesting development setups.

agatti, 2018-01-08:
Theoretically speaking, you can install XC16, have it set into your PATH variable (the installer does ask whether you want that or not) and with some CMake magic get a HEX file ready to be flashed.

However, right now there's no real need for automated build or the like since you cannot easily interact with a BP board from Travis or other CI services.  If avoiding MPLAB-X is a personal preference then by all means, please set it up and I'll gladly merge the pull requests that follow :)

By the way, maybe not many of you noticed but it seems like DP started working on a new Bus Pirate board, ARM based this time (https://github.com/DangerousPrototypes/bus_pirate_ng), so I don't know how much attention this project will have over time or whether to play catch up with whatever changes/improvements the other team will come up with.

sajattack, 2018-01-08:
Ah darn, guess I picked a bad time to buy a bus pirate.

agatti, 2018-01-08:
Not really.  Development of the new version just started (and wasn't really announced anywhere as far as I can tell) - and this doesn't mean your unit will stop working once the new one is done :)  I guess it'll take them some time to get something out, and who knows if they'll drop the v4 like a hot potato - which was the main reason behind this project in the first place.

sajattack, 2018-01-08:
I got it building with the Makefile generated by MPLAB-X. However, when attempting to flash the resulting busPirate.X.production.hex with pirate_loader, it tries to erase page 42, which doesn't exist, and so it fails to flash.

agatti, 2018-01-08:
Believe it or not, it actually flashed successfully.  The last few pages are barred from being written by the bootloader, but the flasher program blindly goes through the whole chip data area.  You can check by unplugging the BP unit, remove the PGC/PGD short, reconnect and check the version :)

sajattack, 2018-01-08:
Ah, right you are. Cool

agatti, 2018-01-08:
Good to know.  I'll keep this issue open so if somebody is willing to write a proper guide/introduction in markdown format for me to merge into the repository and/or write a simplified blurb for README.md, things can be tracked here.

sajattack, 2018-01-08:
Yeah, I'll probably get to work on doing just that.

sajattack, 2018-01-08:
I'll submit a separate PR for the Makefiles. How do I select to build for BPv4?

sajattack, 2018-01-09:
#81

agatti, 2018-01-10:
Thank you!  I'll take a look at it right now and see if I can add something more for macOS and Windows then.

commit c27ec942 (2018-01-10): Document building and flashing (fixes #79) (#81)

* Create building-and-flashing-firmware.md

* Building section

* flashing section

* Add link to building-and-flashing.md in the README

* link fix

* link fix #2
closed 2018-01-10 by agatti