ElderlyPirate/Bus_Pirate #82: Firmware upgrade needs connected PGC/PGD PINs

mike01, 2018-01-13:
After upgrading my Buspirate v3 to the newest Firmware (master branch) I noticed triggering Bootloader/Upgrade-mode via "$" isn't enough anymore: the PGC/PGD PINs need to be connected additionally to trigger the Bootloader/Upgrade-mode, which wasn't needed in the original software.

USBEprom, 2018-01-13:
Hi mike01.
Firmware revisions built from the last released repositories work only with the v4.5 bootloader.
Without upgrading to v4.5 it will not be possible to jump into bootloader from terminal using command $, so that instead it will be necessary to use jumper on PGD and PGC:

https://github.com/BusPirate/Bus_Pirate/issues/64

https://github.com/BusPirate/Bus_Pirate/issues/38

http://dangerousprototypes.com/forum/viewtopic.php?f=28http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=90#p67288

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67056#p67056

Maybe your issue is related to that.

mike01, 2018-01-13:
Ok thank you. So the new Firmware got more dependent to the bootloader by introducing the "skip_pgc_pgd_check" variable which has to be set prior rebooting instead of directly jumping to "#BLJUMPADDRESS". Hm I think I'll stick to the old bootloader for my firmware mods just to be save.

closed 2018-02-06 by agatti