ElderlyPirate/Bus_Pirate #83: Noob: BP 3.6 clone - defect ???

ds2k5, 2018-01-31:
Hello,
got a BP 3.6 clone (printed ob ciruit board)

picocom -b 115200 /dev/ttyUSB0

```
HiZ>i
Bus Pirate v3.5
Firmware v6.1 r1676  Bootloader v4.4
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com

```
But the menu say Version 3.5
```

HiZ>~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.90) OK
VPU(0.00) FAIL
3.3V(3.31) OK
ADC(0.00) FAIL
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI FAIL
CLK FAIL
MISO FAIL
CS FAIL
MODE and VREG LEDs should be on!
Any key to exit
Found 6 errors.

```

**Here some errors, (no devices connected to the BP)
is this OK or is the board damaged ?**

Thanks

agatti, 2018-01-31:
`VPU(0.00) FAIL` and `ADC(0.00) FAIL` might be there because you did not wire up Vpu and ADC pins. `MOSI`, `CLK`, `MISO`, and `CS` failing when the board is in high impedance mode looks strange enough.  I'll see if this is triggered by a software fault, as if it was a wiring issue I'd expect that to fail with the bus being high or low, not while being disconnected.

ds2k5, 2018-01-31:
Thanks, is it possible that need JUMPERs ?

http://dangerousprototypes.com/blog/2009/07/28/bus-pirate-self-test-guide/

USBEprom, 2018-01-31:
@ds2k5

You totally need JUMPERs.
Here you go the output for my Bus Pirate v3.6 while I am purposely performing selftest without JUMPERs:

~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.96) OK
VPU(0.00) FAIL
3.3V(3.31) OK
ADC(0.00) FAIL
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI FAIL
CLK FAIL
MISO FAIL
CS FAIL
MODE and VREG LEDs should be on!
Any key to exit
Found 6 errors.
HiZ>

It totally match with your.
Please, try using the JUMPERs and let we know.

Last but not least.
My Bus Pirate is a version 3.6 from Seeed Studio and like your it is recognized as 3.5 into the terminal:

HiZ>#
RESET

Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>

That is the normal behaviour because :
Bus Pirate v3.6 is exactly the same a Bus Pirate v3.5, the PCB was changed to fit a 'DP6037' standard PCB footprint to make cases easier to design.
v3.6 uses a DP6037 standard PCB footprint with case-friendly dimensions and clearances.
Please take a look at this:

http://dangerousprototypes.com/docs/Bus_Pirate_v3.6

Furthermore you wrote your is a clone, although actually it uses the same PIC as mine which is original.

agatti, 2018-02-01:
```Bus Hi-Z 1
MOSI FAIL
CLK FAIL
MISO FAIL
CS FAIL
```

This shouldn't happen to begin with.

ds2k5, 2018-02-03:
Now with JUMPERs!  It work!

Thanks to all

```

HiZ>~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(5.02) OK
VPU(5.02) OK
3.3V(3.31) OK
ADC(3.31) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on!
Any key to exit
Any key to exit
Found 0 errors.
HiZ>

```

closed 2018-02-04 by ds2k5