ElderlyPirate/Bus_Pirate #84: Compiled firmware not working on board v3.5

larks, 2018-02-03:
Hi,

I have a BP v3.5 and wanted to modify it a bit so that I have two AUX pins by reconfiguring the ADC pin and changing out the resistor networks so the ADC pin no longer has a voltage divider on it's output.

However, I am not able to get the standard firmware working on my unmodified board.
I download the compiler and IDE as per the instructions, cloned the repository and built the project.
When I load the firmware using the pirate-loader, it writes it, then at the end gives me the famous 

  Erasing page 42, a800...ERROR [50]

  Error updating firmware :(

Then when I unplug/re-plug the board and connect with UART, there is no life.
For reference I tried one of the pre-built firmwares that comes in the "package" folder of the repo. These worked, as well as "bpv3_fw7.0_opt0_18092016.hex" that I found in the forum I believe.
The latter also gave the ERROR[50] above, but still worked fine:
  HiZ>i
  Bus Pirate v3.5
  Community Firmware v7.0 - goo.gl/gCzQnW [HiZ UART I2C SPI] Bootloader v4.4
  DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
  http://dangerousprototypes.com

I have been trying to disable all features except UART and I2C as well as combing through all the settings in MPLABX and looking through the linker script but can't find what the problem is.

Is there some known bug in the current version of the firmware? I also tried downloading the different releases, but none of those would compile for me.

USBEprom, 2018-02-03:
@larks

Actually that error message does not mean that something is wrong with the firmware.
More than a matter of bootloader it is a issue with pirate-loader or related tools.
Please read here:

https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-248103673

https://github.com/BusPirate/Bus_Pirate/issues/7#issuecomment-273879409

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=15

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=15#p65317

Furthermore you also need to know that the latest firmware v7.1 releases require the v4.5 version of the bootloader to work completely.
Without upgrading to v4.5 it will not be possible to jump into bootloader from terminal using command $, so that instead it will be necessary to use jumper on PGD and PGC:

https://github.com/BusPirate/Bus_Pirate/issues/82#issuecomment-357425234

larks, 2018-02-03:
@USBEprom 

Thank you for your reply.

I have been reading those, and got that it was an issue with pirate-loader.
Yes, I am using jumper on PGD and PGC to boot to the bootloader. 
Is there any other issue that would make the UART completely silent if I am running bootloader 4.4? 
I thought the only difference was how the bootloader gets loaded?

USBEprom, 2018-02-03:
@larks

For what I know there is no other problem that the non-operation of the $ command to put the Bus Pirate into bootloader mode, problem that can be bypasses by jumpering PGD and PGC.

larks, 2018-02-03:
I did a bit of debugging, and the problem seems to be related to this function:
https://github.com/BusPirate/Bus_Pirate/blob/master/Firmware/base.c#L545-L588

The current implementation is not setting up UART1 in the correct way.
As per the manual, the U1MODE register has to be set before setting the UTXEN bit in the U1STA register.

Switching so that U1MODE is setup before U1STA fixed the problem:
https://github.com/larks/Bus_Pirate/commit/04cb33f02de603fd8a3197df3160edff6e27d2d9

USBEprom, 2018-02-03:
@larks

Sorry I missed what you wrote about the UART side.
How would be possible replicate it exactly?
I built new firmware starting from repository dated 7 January 2018 and I do like to check it for the UART issue you just wrote.
Thanks

larks, 2018-02-03:
@USBEprom 
My steps where:
1. Build project in MPLABX
2. Flash Bus Pirate with built firmware using pirate-loader
3. Connect to BP UART using favorite method
4. No sign of life on UART, i.e. no input or output

Additional information:
Board is Bus Pirate v3.5 from Sparkfun with Bootloader v4.4
Currently tested firmware only has I2C and UART enabled, but same problem occured with the default configuration.

USBEprom, 2018-02-03:
@larks

I know I am a lot babo,sorry.
What do you mean with connect to BP UART using favorite method?
Do you mean start UART protocol by choosing it into the terminal (for instance Putty)?
If so with mine does not happens nothing weird, my Bus Pirate respond correctly without hang or something like that.
The firmware I built has all the available features enabled, I2C hardware mode and BASIC too.
It is firmware v7.1 repository dated 7 January 2018, bootloader is v4.5.
My BusPirate is a v3.6 from Seeed Studio.
Thanks.

larks, 2018-02-03:
@USBEprom 
Yes, connect USB and start e.g. Putty.

Hm, that's strange. Is it possible to ask what micro-controller version is mounted on your Bus Pirate?
I am referring to what this line says when you issue command "**i**" in Putty:

  DEVID:0x0447 REVID:0x3046 (**24FJ64GA002 B8**)

I think I have a Bus Pirate from Seeed Studio in my lab at work, so I can check there to see if the same happens.

In section 17.2 of the PIC24FJ64GA004 FAMILY manual, the setup procedure is mentioned to be as in my changes mentioned above. This is also how it was done in previous versions of the firmware (7.0 and below).

It is also done this way in the example project for UART under Firmware-Alternates:
https://github.com/BusPirate/Bus_Pirate/blob/master/Firmware-Alternates/BusPirateUARTDEMOTEST/main.c#L31-L41

And this project worked for my board.

I am btw using MPLABX v4.05 with XC16 v1.33 toolchain.

USBEprom, 2018-02-03:
@larks

Here is the output of the terminal while issuing the command 'i' on my Bus Pirate v3.6 from Seeed Studio:

**i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>**

Me too I use MPLABX v4.05 with xc16 v1.33 toolchain (on Ubuntu LTS 16.04.3 in my case).
Honestly, for what I remember and know, I have never had any issue on UART side.
The behaviour you wrote is weird at me.

closed 2018-02-06 by agatti