ElderlyPirate/Bus_Pirate #89: Bus Pirate hangs if W not enabled

aten10, 2018-02-09:
Hi,

Tested on firmware version 6.3-beta1 r2151 (spi_fix) and also the SAFE_13042017_1. 
Bootloader v4.5
W is turned off as i am using Vpu with external voltage of 1.7v. 
Bus Pirate just hangs on SPI read command {0x9F rrr]

it works on firmware v 5.10 (r559) bootloader v4.4

agatti, 2018-02-09:
Hi there!  Firmware versions below 7.0 were entirely developed by Dangerous Prototypes, so we didn't really had much to do with that one.  Can you please try with the current code to see if that still happens?

aten10, 2018-02-09:
Hi!
Yes same issue, just hangs. see below. (i reverted back to 6.3 as had found mention of an SPI fix)

SPI>v
Pinstates:
1.(BR)  2.(RD)  3.(OR)  4.(YW)  5.(GN)  6.(BL)  7.(PU)  8.(GR)  9.(WT)  0.(Blk)
GND     3.3V    5.0V    ADC     VPU     AUX     CLK     MOSI    CS      MISO
P       P       P       I       I       I       O       O       O       I
GND     0.00V   0.00V   0.00V   1.77V   L       L       L       H       H
SPI>
i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEY
B LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
CFG1:0xF9DF CFG2:0x3F7F
*----------*
Pinstates:
1.(BR)  2.(RD)  3.(OR)  4.(YW)  5.(GN)  6.(BL)  7.(PU)  8.(GR)  9.(WT)  0.(Blk)
GND     3.3V    5.0V    ADC     VPU     AUX     CLK     MOSI    CS      MISO
P       P       P       I       I       I       O       O       O       I
GND     0.00V   0.00V   0.00V   1.77V   L       L       L       H       H
POWER SUPPLIES OFF, Pull-up resistors ON, Open drain outputs (H=Hi-Z, L=GND)
MSB set: MOST sig bit first, Number of bits read/write: 8
a/A/@ controls AUX pin
SPI (spd ckp ske smp csl hiz)=( 1 0 1 0 1 1 )
*----------*
SPI>P
Pull-up resistors ON
SPI>[0x9f rrr]
/CS ENABLED
WRITE: 0x9F

agatti, 2018-02-13:
Ok, great.  I'll see what I can do for this then!

agatti, 2018-02-14:
Ok, I can reproduce this on v4 as well.  The SPI1 interrupt flag never gets triggered when writing data in this case:

```c
uint8_t spi_write_byte(const uint8_t value) {
  uint8_t result;

  SPI1BUF = value;
  while (!IFS0bits.SPI1IF) { // <-- Gets stuck here.
  }
  result = SPI1BUF;
  IFS0bits.SPI1IF = OFF;

  return result;
}
```

agatti, 2018-02-14:
So, this issue is not really easy to solve per se, as it exposes a fundamental issue on how protocols are handled internally.

Right now, each protocol/module (a.k.a. what shows up when you use the `m` command) setup phase is split in two parts, the first one being performed whenever the protocol is chosen, and the second one being performed when the `W` command is issued.  In this very case, the SPI interface behaviour like speed and interrupt trigger policy is set in the second setup phase, hence the behaviour described above (SPI1 interrupt never firing).  Merging setup phases will mess up a few protocols that depend on having multiple setup steps to properly configure the board first.

That said, doesn't the `W` command only enable the +5v and the +3v3 pins, leaving alone Vpu?  In that case it shouldn't matter whether your power source is external or not.  If you leave the +5v and +3v3 pins unconnected then you should be able to have things running anyway...  Or am I missing something here?

aten10, 2018-02-14:
Ok, for some reason I thought 5v was from the bus pirate! (usb power) I will test tomorrow!
In the meantime I used flashrom to achieve what I wanted to do

aten10, 2018-02-15:
Can confirm works fine, thanks!

closed 2018-02-15 by agatti