ElderlyPirate/Bus_Pirate #90: Add I2C clock stretching in software mode

agatti, 2018-02-14:
As mentioned in #88, I2C clock stretching in software mode is currently missing.

ChristopherSamSoon, 2018-02-14:
Hi Agatti,

I can work on this no problem. The strategy here is to allow the bit-bang mode be modified to allow clock stretching for I2C while still maintaining the same functionality for other modules (because the bitbang functions are shared with the other protocols)

There are two solutions
1) Add an extra parameter to the bitbang functions that indicate whether there should be clock stretching or not. I2C calls will set this parameter to true, the rest (SPI, 1 WIRE, etc...) will set it as false

2) Create a seperate bitbang function that only the i2c mode will call (lets say bitbang_i2c_write_value and bitbang_i2c_read_value)

I will go for the second solution as less things have to change

To add further, I have  a slow I2C device here that I can use to test this clock stretching stuff!!

agatti, 2018-02-14:
That may work - although I'd try to make sure that as much code is shared as possible between bitbang functions.  Keep in mind that the RAW2WIRE protocol does use bitbang functions in an I2C-like fashion too.

ChristopherSamSoon, 2018-11-03:
I2C clock stretching solution is available. Basic idea is to wait for clock line to go high (released by the slave) on the ACK bit. Code snippet below. Code also bails out after a maximum predefined amount of time to wait

```
bool bitbang_read_bit(bool CLKStretch) {
  bool bit_value, clk_value;
  uint32_t max_wait;

  /* Set the MISO pin as input. */
  bitbang_read_pin(miso_pin);

  /* Set CLK high. */
  bitbang_set_pins_high(CLK, delay_profile->clock);

  /* Wait for CLK to go high if desired*/
  if(CLKStretch == true){
      max_wait =0;
      do{          
          clk_value = IOPOR & CLK; // check if clock is released by slave
          if(clk_value != 0){
              break; // clk signal released by slave, exit
          }
          bp_delay_us(1); // wait 1us
          max_wait = max_wait + 1;
          
      }while(max_wait < I2C_SW_CLK_STRETCH_MAXWAIT_US);
  }

  /* Read value. */
  bit_value = bitbang_read_pin(miso_pin);

  /* Set CLK low. */
  bitbang_set_pins_low(CLK, delay_profile->clock);

  return bit_value;
}
```

Cheers

Christopher Sam Soon

jermonca, 2024-02-09:
Do you have an update on clock stretching support during sniffer mode? I find that it is not decoding properly. Thank you.