ElderlyPirate/Bus_Pirate #94: 3 new commits at the "official" DP Bus_Pirate repo

mikebdp2, 2018-06-07:
Hi there @agatti ! Sorry for not writing for so long, have been occupied with many open-source projects... I hope you are doing well. Have noticed 3 new commits at the "official" DP Bus_Pirate repository - https://github.com/DangerousPrototypes/Bus_Pirate/commits/master - and, looks like all of them are to the "hardware" directory, which does not exists at your repo for some reason. These commits also mention Bus Pirate v5.0, but I suspect its just a new version of BPv3 with a slightly larger board. I've reached to @vimark for a comment here - https://github.com/DangerousPrototypes/Bus_Pirate/issues/7

JarrettR, 2018-06-07:
Hi @mikebdp2 

Those are for a mythical "BusPirate 5", which, judging from the commits is a complete redesign using the STM32 ARM Cortex-M3 architecture. It won't be compatible in any way with this repo, or the legacy official one because of that. There's also no mention of it in any "official" channels, because I suspect they're trying to fly under the radar until it's at a sufficiently done state ;)

Presumably a v3.9 is 100% compatible with the rest of the line, just with a different set of hardware. That folder is [here](https://github.com/BusPirate/Bus_Pirate/tree/master/package/hardware) on this repo, but I doubt anyone is using the hardware files from here, that's not really the focus.

informer2016, 2018-06-11:
@JarrettR Thank you very very much for your reply, Jarret! Didn't get any real reply from @vimark regarding the hardware side, perhaps they want to keep this project as secret :stuck_out_tongue_winking_eye: Yes, I could understand them, although I'm confident BPv5 would not impact the sales of BPv4 because BPv4 has the fully 100% open source bootloader/firmware - which is epic win. Most likely that wouldn't be the case with their new BPv5 based on Cortex-M3, because any ARM hardware = binary blobs required, every time I looked to it

agatti, 2018-07-02:
@informer2016 last time I checked they're using a STM32 MCU for the "NG" version, which doesn't really require any closed source code at all.  Maybe you're thinking of extended features like bluetooth and wifi, but as far as I can remember they aren't planning anything like that for the time being.

agatti, 2018-07-02:
@mikebdp2 as @JarrettR mentioned, those commits are just hardware modifications.  However, the real new stuff is [here](https://github.com/DangerousPrototypes/Bus_Pirate/commits/firmware_v8_official) - which has been created without getting in touch with us or anything.  Now, the main issue here would be to backport stuff but then it'd become a mess when it comes to version numbers... Should we just backport the new stuff along with our bugfixes and call it BusPirate Firmware v8-Community Edition then?

mikebdp2, 2018-08-16:
@agatti It seems they stopped committing new code or stopped doing it publicly, last commit is April 10. Yes its possible to backport these new changes, but should be very careful - because we don't know if this code has been well tested on any real hardware (and tested at all?), and does it contain the new bugs

kallisti5, 2018-08-16:
https://github.com/DangerousPrototypes/Bus_Pirate/tree/firmware_v8_official
Looks like DP is completely ignoring the community firmware and still doing their own thing... kind of a bummer. The codebase here is a *lot* cleaner.