ElderlyPirate/Bus_Pirate #95: support runtime interrupt vector assignment

Elemecca, 2018-06-23:
I'm working on a fairly complex new mode that requires interrupts to function properly. Unfortunately with the default way XC16 handles interrupt vectors they can only be assigned at compile time. It's fairly clear that existing modes have avoided using interrupts even where it would have made them more efficient, and I suspect that's why.

I would like to propose that support for assigning interrupt vectors at runtime be added to the firmware. That would allow each mode to have its own interrupt vectors, which would allow modes to use interrupts as they see fit without worrying about conflicts.

There are several different methods for assigning ISRs at runtime. I have one that I prefer, and I'll be submitting a pull request for that soon as I've already written most of the code. I'm creating this issue as a referendum on the concept of runtime ISR configuration in general so we have a place to discuss it even if that PR is rejected for implementation reasons.

agatti, 2018-07-02:
Can you please elaborate a bit further on the interrupts usage?  Right now interrupts are not really used except for UART on v3 boards and for ADC on both v3 and v4 boards, mostly because there was no real need to use them for something else (even USB is running in polling mode on v4).

I am not really against the idea to begin with, but my only concern is that we strive to get feature parity between v3 and v4 board (yes, OpenOCD is still one sore point that I hope to fix soon).  Which interrupts would you use for your mode?

Elemecca, 2018-07-10:
The mode emulates an HP 5004 Signature Analyzer, which is essentially a logic probe that feeds a linear feedback shift register to generate a checksum of the activity on a line. Signatures are used in the troubleshooting procedures in the service manuals for a lot of test gear from the early 80s, so for people like me who do restoration it would be convenient to be able to generate them with a relatively common and inexpensive tool like the Bus Pirate.

It uses the SPI module to capture clocked input data, and `SPIEN` needs to be set very quickly after the gating signal goes active. I used the input capture interrupts to do so with predictable latency without blocking user input. As it's currently written you start it with `[` and then it captures signatures continuously and only prints to the terminal when the signature changes. It would probably be possible to do it without interrupts, either by capturing a single signature synchronously in the read command and polling the gating signals in a tight loop, or by polling in the periodic function and just skipping start events that were too delayed.

I was already writing the mode and the interrupt support to work on v3(.6) and v4, since I have both boards. The interrupt support code works now on v4. What's been preventing me from opening a pull request is that currently v3 won't boot when it's enabled. Something about the table pushing a required variable out of near memory, I think. I'm still looking into that.