ElderlyPirate/Bus_Pirate #98: register to forum dangereousprototypes

colo-des, 2018-06-30:
I was unable to subscribe to the dangerousprototypes forum, it seems that the google re-captcha does not want me, I sent an email to admin and I was rejected by the server, how can I subscribe to the forum?
I would like to report an error using the firmwares that I download of the forum dangerousprototypes the user mikeb  http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498#p65290
I own a buspirate 3 and 4, plus an OLS, and I would like to participate in providing debugging since I work in electronic repairs...I'm a developer using pics, mainly assembler, but then I can create my own versions of buspirate to help find bugs.
The idea is to use the forum for exchange between users and not this medium that is purely for development, but since there is no way to communicate I had to create this new item.

0xff7, 2018-07-02:
Yea It's been like this for months too and and I've been waiting to register. There seems to be no way to get ahold of them as their emails get auto returned, and social medias seems like just ads. It's extremely frustrating being a new bus pirate owner and not being able to post on the forum. :/

agatti, 2018-07-02:
Sorry folks, the dangerous prototypes forum is something we have absolutely no control over.  This project was created because the dangerous prototypes folks left the bus pirate v4 user community high and dry when it came around to support and new firmware developments, effectively forking the firmware project..

kallisti5, 2018-07-02:
Indeed.  I created the github org. due to the complete and utter lack of collaboration + continued support from Dangerous Prototypes around the BusPirate.  @agatti has been doing a great job doing all the hard work, but we have received zero collaboration from Dangerous Prototypes.

Speaking for myself at least, we're always open if Dangerous Prototypes wants to collaborate!

agatti, 2018-07-02:
@kallisti5 I'd love to hear from the Dangerous Prototypes people as well - especially if we can get in early and have new v5-only code being easier to backport to v4 and v3.  Even something as small as sharing common code and use board-specific hardware hooks and whatnot would be a success in my book.

agatti, 2018-07-02:
@kallisti5 Did you take a look at [this](https://github.com/DangerousPrototypes/Bus_Pirate/commits/firmware_v8_official)?  Not sure what's happening on their side, to be honest.

kallisti5, 2018-07-02:
Yeah... more of the same silo stuff DP seems to do. I completely love the BusPirate, but I get the feeling at this point their model is "churning out 'open' designs", but they're not really interested in interfacing the community around those designs.

I mean, BusPirate v4 was a huge improvement with some customer demand (and a well-known brand)... but they just didn't seem interested in working on it after designing the hardware.

colo-des, 2018-07-03:
@agatti thank you for answering and clarifying things, now I understand what is happening
I see that Ian resumed the development but he does not take the annoyance to answer anyone, with that it is enough to know their intentions, but I stay here collaborating here with you.

My problem comes from the hand of mplabx, I do not intend to execute any java of oracle and less give it
root permissions to an installer that I do not know, that has me relegated to using mplab 8.92 over
wine, I only use linux,  I'm looking at how to make an ebuild for my gentoo in the style of
they have made those of archlinux and I have found something but it is half old.

To summarize the story, I passed a link that interests you to read
https://fromorbit.com/2012/01/06/why-microchip-with-mplabx-has-put-the-cart-before-the-horse.html

ebuild of archlinux https://aur.tuna.tsinghua.edu.cn/packages/microchip-mplabx-bin/?comments=all
ebuild for gentoo   https://github.com/fernando-rodriguez/portage-overlay/blob/master/dev-embedded/mplabx/mplabx-3.15-r1.ebuild

here a recent video of an archlinux user using the ebuild
https://www.youtube.com/watch?v=j3T2PszR7IQ

If I do not succeed, I'll have to install everything on wine, so I hope to have mplabx to be able to
create my own versions from this git for my buspiratas 3 and 4.

Thank you very much for having taken the decision to create this community project where we all win.

JarrettR, 2018-07-03:
I don't believe using MPLab 8 is still reasonable. You can run Java in a VM, without internet access, and it will be perfectly safe,

colo-des, 2018-07-04:
@JarrettR Hi, I do not have anything against openjdk, in fact I have it installed in my gentoo testing system
It is the icedtea package

$ eix -Ic icedtea
[I] dev-java/icedtea (3.8.0(8)@14/06/18): A harness to build OpenJDK using Free Software build tools and dependencies

$ eselect java-vm list 
Available Java Virtual Machines:
  [1]   icedtea-8  system-vm

The bad thing is that the mplabx installer has a java version of oracle that mplab installs next to it.
I do not know why, since openjdk works perfectly and is not a potential danger as if it is the one of oracle.
It also you like it or not have to run it as root, which makes the ebuil of archlinux and the one of gentoo, is to create a false root environment and jail the installer making him believe that it was given root powers, so that the unzip and install where it pleases, then what follows the ebuild package installation removing the java from oracle and other unpleasant things and then allow it to be installed where you want it out of the system in the folder of the user for example.

Regarding VMs, I really do not like them very much, there were problems with the detection of USB devices and I would have problems with my programmer pickit2 ... I was reading that mplabx supported until its version 4.0.1 to pickit2 and already only supports pickit3, is that true?
https://www.microchip.com/forums/m1047279.aspx
And I'm still with Mplab 8.92 just because most of my old projects are in assembler... but yes, it's time to move to mplabx, but microchip does not make it easy for me.

mikebdp2, 2018-08-16:
@colo-des and @0xff7 (and @ipatch) : using my existing account at DangerousPrototypes forums, I have just sent lots of private messages *(and e-mails when its possible)* to all the members with a green rank *(mods/admins)* as well as to some of the "hero members" with a large post count : telling about the broken captcha. They have already been read by at least someone *(I see some of them in my "Sent" instead of "Outbox")*, also created this issue - https://github.com/DangerousPrototypes/Bus_Pirate/issues/9 . Hopefully the registration will be fixed soon, please check it from time to time to see when its' available

@colo-des Primary place of new developments is our repository here. Feel free to contribute both with your code and new issues containing your debug info

Have you heard about pk2cmd open-source flashing software for Pickit2 tool? You could use MPlab or MPlabX inside VM to build the new stuff, then use pk2cmd at your host OS *(without VM, so no USB problems)* to flash the compiled images to your chip

Look through these articles I wrote - http://dangerousprototypes.com/docs/Compile_the_Bus_Pirate_firmware and http://dangerousprototypes.com/docs/Bus_Pirate_v4_and_Pickit2_(bootloader_and_firmware_updates) , they will give you an idea how to set up a perfect dev environment

MPlabX is nice IDE , it supports both Pickit2 and Pickit3, but Pickit3 is buggy crap without a proper Linux support (e.g. there is no such a thing as pk3cmd open-source flashing tool) . So its great that you have a Pickit2

If any other questions, please write below

ipatch, 2018-08-16:
@mikebdp2 please keep us updated if you get any new info please.

mikebdp2, 2018-08-17:
@ipatch News: at this thread - https://github.com/DangerousPrototypes/Bus_Pirate/issues/9 - Ian told us they're going to fix captcha ASAP

colo-des, 2018-08-19:
@mikebdp2 Great contribution to be able to communicate with Ian and let him know about the subject of the captcha, every so often I will continue to try to subscribe ... sincerely many thanks.

My case it all started when I tried to find new firmware versions for my buspirata3 and buspirata4 and started reading the threads of the dangerousprototypes forum, where I read what was posted by the user mikeb http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498#p65290
I presume you are yourself since between mikeb and mikebdp2 ....

Since 1995 I am with pics, in assembler mainly, my first programmer was a picstartplus by microchip serial port that I still have and use with the picp program:
Then I bought a pickit2 and since then it is my pics programmer, using of course pk2cmd.

$ eix -I picp
[I] dev-embedded/picp
      Available versions: (~) 0.6.8
      Installed versions: 0.6.8 (00:10:39 07/20/18)
      Homepage: http://home.pacbell.net/theposts/picmicro/
      Description: A commandline interface to Microchip's PICSTART + programmer

$ eix -I pk2cmd
[I] dev-embedded/pk2cmd
     Available versions:  (~)1.20
     Installed versions:  1.20(22:17:59 11/02/18)
     Homepage:            http://www.microchip.com/pickit2
     Description:         An application for working with the Microchip PicKit2 PIC programmer

Since I'm not in agreement with the installation policy of mplabx with java version of oracle and by requesting root permission, I have installed everything in wine, I have mplab-8.92 and mplabx-4.20, both run perfectly well and without problems. pickit3 it was wrong, I have not had the opportunity to try one, but as you say is that I could not either since I do not use windows, I only use linux...gentoo testing since 2008.

Right now I'm using the versions posted in the link above by mikeb in the dangerousprototypes forum.
I have updated using pickit2 with pk2cmd using the bootloader+firmware versions

$ pk2cmd -P PIC24FJ64GA002 -M -F bpv3_fw7.11_opt1_12062018.hex
Using PE
PICkit 2 Program Report
30-6-2018, 20:10:04
Device Type: PIC24FJ64GA002
Program Succeeded.
Operation Succeeded

$ minicom /dev/ttyUSB0
Welcome to minicom 2.7
OPCIONES: I18n 
Port /dev/ttyUSB0, 14:25:17

HiZ>i
Bus Pirate v3.5
Community Firmware v7.11- goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE PIC DIO] Bootloader v255.255
DEVID:0x0447 REVID:0x3043 (24FJ64GA00 2 B5)
http://dangerousprototypes.com

$ pk2cmd -P PIC24FJ256GB106 -M -F bpv4_boot4.11_opt1_12062018.hex
Using PE
PICkit 2 Program Report
30-6-2018, 20:40:55
Device Type: PIC24FJ256GB106
Program Succeeded.
Operation Succeeded

$ minicom /dev/ttyACM0
Welcome to minicom 2.7
OPCIONES: I18n 
Port /dev/ttyACM0, 14:33:53

HiZ>i
Bus Pirate v4
Community Firmware v7.0 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO]
DEVID:0x1019 REVID:0x0003 (24FJ256GB106 A5)
http://dangerousprototypes.com

At the moment I'm a little short of time, but I've been looking at the .hy .c with mplabx, I hope to get full soon, I have some doubts about whether or not to use the bootloader since in my case I charge the firmwares with pickit2.

Thanks again

agatti, 2018-08-30:
So, can this be closed then?

closed 2018-09-02 by agatti