ElderlyPirate/Bus_Pirate #101: LIN protocol support

whitelynx, 2018-08-22:
Related to [issue #14 in the old Google Code repo](https://code.google.com/archive/p/the-bus-pirate/issues/14).

I could use this for troubleshooting a broken [IKEA Bekant desk](https://github.com/robin7331/IKEA-Hackant), which uses LIN.

whitelynx, 2018-08-22:
Quote from the original issue:
> I originally thought LIN was just UART over a single wire with a transceiver, but it also has some special characters and auto baud rates (I believe). I have yet to find public domain LIN code in C to include in the project. I'm going to remove myself as the owner, but leave the project as 'started'.

I found a couple of LIN libraries, but they're both Arduino-related:
- [macchina/LIN](https://github.com/macchina/LIN) - for Arduino; MIT licensed
- [gandrewstone/LIN](https://github.com/gandrewstone/LIN) for Arduino; 2-clause BSD licensed

Would it make sense to grab one of these and port it?

agatti, 2018-08-30:
Wouldn't LIN support be tied to a particular transceiver model, then?  I mean, initialisation sequences, modes set-up, and so on aren't standardised across transceivers as far as I know.  I suspect your best bet for now would be to use the binary I/O scripting support and try to implement transceiver communication on top of this - if there is anything that is not available in the interface that is mandatory to get it working, then let us know and we'll see what can be done for it!