ElderlyPirate/Bus_Pirate #102: jtag on bp 3a

pljakobs, 2018-08-25:
So I have successfully built and flashed v7.1 from this repo, but could not find jtag among the protocols.
From what I gather, it should be available for v3 hardware - is it no longer configured through the modes menu?

Also, I know the flash memory of the pic on the v3 boards isn't sufficient to hold a single binary with all features, it would be excellent to have a choice, which features to build to create a personalized set.

agatti, 2018-08-30:
As far as I know, JTAG is meant to be used only via the binary I/O protocol.  I checked the source code for version 6.1 as it came out from dangerousprototypes' git repository and there is no trace of a command line-driven JTAG mode.

pljakobs, 2018-08-30:
hmm.. maybe I was mistaken then, I thought I had seen jtag in the protocol selection menu in some tutorials.

Marsfan, 2018-08-31:
I believe at one point there was an alternative firmware that had a JTAG mode. I seem to recall trying it and not having much luck, but I believe that was because I have no clue how to use JTAG.

Marsfan, 2018-09-01:
@USBEprom Can you clarify this statement for me

> alternative firmware just like the AVR programmer

Are you referring to the STK500 emulator, because I though avrdude could directly interact with the bus pirate standard firmware.

USBEprom, 2018-09-01:
**@Marsfan**

Yes, of course I meant STK500 emulator because for the same reason of the JTAG feature, lack of available  memory into 24FJ64GA002 PIC used on Bus Pirate v3, **firmware-STK500v2** too has been pushed as alternative:

https://github.com/BusPirate/Bus_Pirate/tree/master/Firmware-Alternates

However you are fully right, **avrdude** could directly interact with the Bus Pirate standard firmware, v7.1 too.
I used **avrdude** as shell and as GUI without any problem.

agatti, 2018-09-02:
STK500 will probably be removed in the next repository cleanup, as I am not even sure it still compiles.  Is there any reason for us to keep this around?

Marsfan, 2018-09-03:
@agatti All I can say regarding the STK500 is when I tried to use it about a year ago with Atmel Studio, Atmel Studio did not recognize it as a STK500 device.

agatti, 2018-09-05:
@Marsfan Fair enough, those will be gone in 7.2 then.

closed 2018-09-05 by agatti