ElderlyPirate/Bus_Pirate #106: Feature request: SiLabs C2 support

pjz, 2018-09-06:
I was considering building a https://github.com/conorpp/u2f-zero which requires a SiLabs JTAG/C2 to program it.  It's not a lot of $$, but if I have a Bus Pirate, do I really need it?  

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=5113 indicates that the protocol is fairly simple and even points to C2 drivers in the kernel that may have helpful code. 

So it would be great if this could be added to the list of programmers that the Bus Pirate can replace. Thanks!

agatti, 2018-09-06:
That's definitely doable although maybe starting to get a proof of concept using the `RAW2WIRE` mode and its associated binary I/O interface could provide the most benefit before rewriting this in C.  Are there any open source programmers for C2?  I mean, something like flashrom/avrdude/openocd/etc..

pjz, 2018-09-06:
https://github.com/Guntermann-Drunck/c2tool might be relevant.

pjz, 2018-09-06:
Also, someone one did a dev board and custom programmer/debugger [wayback in 2007](https://web.archive.org/web/20070222114409/http://www.ise.pw.edu.pl:80/~wzab/cygnal/).