ElderlyPirate/Bus_Pirate #109: Sump not working

Sodium-Hydrogen, 2018-10-17:
I have both a bp v3.6 and v4 running the newest firmware and I cannot get either of them to work as a logic analyzer with [Jawi's OLS client v9.8](https://github.com/jawi/ols). I can get the OLS working with the default factory firmware v5.10 so I know the client works. Any help is appreciated.

Sodium-Hydrogen, 2018-10-17:
I've verified on the BP v3.6 that the following firmwares work with SUMP
* v5.10
* v6.1
* v6.2
* v6.3

USBEprom, 2018-11-01:
I confirm this issue, just tested it.
Old firmwares v5.10, v6.1, v6.2 and v6.3 are working fine while sadly instead since the very first release of the v7.0 it never worked, not even the upgrade provided here http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498#p65290 and of course none of the firmwares I built and shared on the dangerousprototypes forum too.
It is not a matter of client due the fact that the problem exists both with ols-0.9.8 and ols-0308 from Jawi which work like a charm with the firmwares mentioned above.
Although the client fails to successfully start SUMP on Bus Pirate with firmware v7.x., metadata of the Bus Pirate on the client ols-0.9.8 are correctly acquired without any problem.
The problem is that acquisition does not start by honoring triggers settings and instead it activates itself spontaneously in an arbitrary way in its end  displaying spurious signals on video also referred to disconnected inputs, namely inputs left free and therefore not physically connected to anything, floating!
By starting acquisition it does not start if not spontaneously after some time and ending with meaningless spurious patterns, something like as in attachment.
[image]