ElderlyPirate/Bus_Pirate #113: v3 bootloader does not compile

Rondom, 2018-11-04:
I tried compiling the v3-bootloader v4.5, but did not succeed (see below).

* XC16 1.35
* MPLABX v5.00
* Project Bootloaders/BPv3-bootloader/firmware-v4.5/ds30loader.X

Output:

    make -f nbproject/Makefile-default.mk SUBPROJECTS= .build-conf
    make[1]: Entering directory '/home/andreas/cvs/bus_pirate/Bootloaders/BPv3-bootloader/firmware-v4.5/ds30loader.X'
    make  -f nbproject/Makefile-default.mk dist/default/production/ds30loader.X.production.hex
    make[2]: Entering directory '/home/andreas/cvs/bus_pirate/Bootloaders/BPv3-bootloader/firmware-v4.5/ds30loader.X'
    "/opt/microchip/xc16/v1.35/bin/xc16-gcc"   ../ds30loader.s  -o build/default/production/_ext/1472/ds30loader.o  -c -mcpu=24FJ64GA002  -omf=elf -DXPRJ_default=default  -legacy-libc  -I".." -Wa,-MD,"build/default/production/_ext/1472/ds30loader.o.d",--defsym=__MPLAB_BUILD=1,-g,--no-relax
    ../settings.inc: Assembler messages:
    ../settings.inc:46: Error: can't open p24fxxxx.inc for reading
    ../settings.inc:67: Error: Invalid mnemonic: 'config'
    ../settings.inc:68: Error: Invalid mnemonic: 'config'
    ../ds30loader.s:227: Error: Invalid operands specified ('bclr OSCCON,#SOSCEN').
    ../ds30loader.s:227:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:228: Error: Invalid operands specified ('bclr CLKDIV,#RCDIV0').
    ../ds30loader.s:228:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:229: Error: Invalid operands specified ('btss OSCCON,#LOCK').
    nbproject/Makefile-default.mk:112: recipe for target 'build/default/production/_ext/1472/ds30loader.o' failed
    ../ds30loader.s:229:        Check operand #2. Operand must be between 0 and 15, inclusive.
    make[2]: Leaving directory '/home/andreas/cvs/bus_pirate/Bootloaders/BPv3-bootloader/firmware-v4.5/ds30loader.X'
    nbproject/Makefile-default.mk:90: recipe for target '.build-conf' failed
    make[1]: Leaving directory '/home/andreas/cvs/bus_pirate/Bootloaders/BPv3-bootloader/firmware-v4.5/ds30loader.X'
    nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed
    ../ds30loader.s:264: Error: Invalid operands specified ('bclr LATB,#LATB1').
    ../ds30loader.s:264:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:265: Error: Invalid operands specified ('bset TRISB,#TRISB1').
    ../ds30loader.s:265:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:266: Error: Invalid operands specified ('bset CNPU1,#CN5PUE').
    ../ds30loader.s:266:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:268: Error: Invalid operands specified ('bclr LATB,#RB0').
    ../ds30loader.s:268:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:269: Error: Invalid operands specified ('bclr TRISB,#TRISB0').
    ../ds30loader.s:269:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:274: Error: Invalid operands specified ('btsc PORTB,#RB1').
    ../ds30loader.s:274:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:290: Error: Invalid operands specified ('bset RPINR18,#U1RXR0').
    ../ds30loader.s:290:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:291: Error: Invalid operands specified ('bclr RPINR18,#U1RXR1').
    ../ds30loader.s:291:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:292: Error: Invalid operands specified ('bset RPINR18,#U1RXR2').
    ../ds30loader.s:292:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:293: Error: Invalid operands specified ('bclr RPINR18,#U1RXR3').
    ../ds30loader.s:293:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:294: Error: Invalid operands specified ('bclr RPINR18,#U1RXR4').
    ../ds30loader.s:294:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:297: Error: Invalid operands specified ('bset RPOR2,#RP4R0').
    ../ds30loader.s:297:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:298: Error: Invalid operands specified ('bset RPOR2,#RP4R1').
    ../ds30loader.s:298:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:299: Error: Invalid operands specified ('bclr RPOR2,#RP4R2').
    ../ds30loader.s:299:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:300: Error: Invalid operands specified ('bclr RPOR2,#RP4R3').
    ../ds30loader.s:300:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:301: Error: Invalid operands specified ('bclr RPOR2,#RP4R4').
    ../ds30loader.s:301:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:304: Error: Invalid operands specified ('bset LATA,#LATA1').
    ../ds30loader.s:304:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:305: Error: Invalid operands specified ('bclr TRISA,#TRISA1').
    ../ds30loader.s:305:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:343: Error: Invalid operands specified ('bset UMODE,#BRGH').
    ../ds30loader.s:343:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:344: Error: Invalid operands specified ('bset UMODE,#UARTEN').
    ../ds30loader.s:344:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:345: Error: Invalid operands specified ('bset USTA,#UTXEN').
    ../ds30loader.s:345:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:562: Error: Invalid operands specified ('bset NVMCON,#WR').
    ../ds30loader.s:562:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:566: Error: Invalid operands specified ('btsc NVMCON,#WR').
    ../ds30loader.s:566:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:579: Error: Invalid operands specified ('btss USTA,#URXDA').
    ../ds30loader.s:579:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:594: Error: Invalid operands specified ('btss PORTB,#RB1').
    ../ds30loader.s:594:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:600: Error: Invalid operands specified ('bclr UIFS,#URXIF').
    ../ds30loader.s:600:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:601: Error: Invalid operands specified ('bclr UIFS,#UTXIF').
    ../ds30loader.s:601:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:602: Error: Invalid operands specified ('bclr USTA,#UTXEN').
    ../ds30loader.s:602:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:603: Error: Invalid operands specified ('bclr UMODE,#UARTEN').
    ../ds30loader.s:603:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:611: Error: Invalid operands specified ('bclr LATA,#LATA1').
    ../ds30loader.s:611:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:612: Error: Invalid operands specified ('bset TRISA,#TRISA1').
    ../ds30loader.s:612:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:628: Error: Invalid operands specified ('bclr CNPU1,#CN5PUE').
    ../ds30loader.s:628:        Check operand #2. Operand must be between 0 and 15, inclusive.
    ../ds30loader.s:629: Error: Invalid operands specified ('bset TRISB,#TRISB0').
    ../ds30loader.s:629:        Check operand #2. Operand must be between 0 and 15, inclusive.
    make[2]: *** [build/default/production/_ext/1472/ds30loader.o] Error 255
    make[1]: *** [.build-conf] Error 2
    make: *** [.build-impl] Error 2

    BUILD FAILED (exit value 2, total time: 189ms)

agatti, 2018-11-04:
Fair enough - I;ll just fix it right away.

agatti, 2018-11-04:
@Rondom can you please check if this makes things work for you?

andersm, 2018-11-04:
It looks like you need to delete COE_OFF from __CONFIG1 too, that setting doesn't exist on this MCU.

agatti, 2018-11-04:
@andersm that should have already been taken care of - did I miss an instance of that flag?

andersm, 2018-11-04:
No, I was just looking at an outdated working tree.