ElderlyPirate/Bus_Pirate #117: Possible improvement for the 1-WIRE protocol.

USBEprom, 2018-11-10:
If I have not misunderstood to enable Overdrive (~160kps) on the 1-WIRE devices that allow for it, not all 1-WIRE slaves support Overdrive (~160kps), it is need to send 0x69 (Overdrive Match ROM command) in normal speed followed by the 8-byte ROM identifier in overdrive speed or by sending 0x3c (Overdrive Skip ROM command) in normal speed.
This one immediately puts all overdrive capable slaves on the bus into overdrive mode.
Once a slave is in overdrive mode, all subsequent communication must take place at overdrive speed.
(http://forum.arduino.cc/index.php?topic=201776.msg1486950#msg1486950)
The Bus Pirate provide Overdrive (~160kps) support but actually it can not use it because in order to do so would need to go from Standard (~16.3kbps) to Overdrive (~160kps) and the other way around, keeping active pull-ups and power supply, which is not currently allowed.
In fact, once one of the two items has been set, Standard (~16.3kbps) or Overdrive (~160kps), to change it it is need to return to the menu with the "m" command, thus immediately making the power supply and the pull-up resistors are deactivated, therefore it is in no way possible to initialize the slave devices that support it in the Overdrive mode (~160kps).
Even using externally powered external pull-up resistors it does not improve the thing, especially when working with emulated 1-WIRE slave devices, as this could result in the risk of damaging the Arduino board and or the Bus Pirate itself.
In my opinion a solution would be to be able to switch from Standard (~16.3kbps) to Overdrive (~160kps) and the other way around without contextually deactivating power supply and pull-up resistors on the Bus Pirate side.
I do not know if this is possible or not, even if it can be recommended, it is only a thought of mine that I want to share.

tudstudent, 2026-10-23:
Implemented. PR #220 adds a command for this: in 1-Wire mode, `O` toggles the bus speed in place (standard <-> overdrive) and prints the new speed. The power supplies and pull-ups keep running - no mode change - so a device that was switched to overdrive (reset + `0x3C`) can be addressed at overdrive speed in the same session, and switched back the same way.

Verified on hardware with a device on the bus: standard READ ROM -> reset + `0x3C` -> `O` -> overdrive READ ROM -> `O` -> standard READ ROM. The device-side log confirms the reset widths flip ~500 us -> ~70 us -> ~500 us and the commands decode correctly at both speeds.

USBEprom, 2026-10-24:
**@tudstudent**

Thank you very much for this Sir, respect!
Much appreciated!
Hoping all this will be merged in the code. 

Be seeing you. 

U.Sb