ElderlyPirate/Bus_Pirate #124: UART Transparent bridge and Live monitor bruken in 6.3_r2151

Twilight-Logic, 2019-01-31:
I tried to use flashrom with my buspirate and found it required version 6.2 or higher of the firmware. After discovering the community releases, I downloaded and flashed BPv3-firmware-v6.3-r2151.hex on my BPv3. I couldn't read the EEPROM chip I was trying to read (which might be another issue) but today I needed to debug a UART port and discovered that Transparent bridge and Live monitor did not work. Characters were not being received in either direction. I spent a while trying to figure it out and eventually downloaded BPv3-firmware-v6.2-r1981.hex. When I tried to flash it with ds30 and found I couldn't because ds30 was reporting that it couldn't get a response from the bootloader.....

I had flashed the 6.3 firmware by simply going to a terminal and typing $ to start the bootloader and then using ds30 to confirm all was well and upload the firmware. Well it turned out that the behaviour had changed and I now needed to take both actions, i.e. link PGGND/PGC together AND go to terminal and type $ to start the bootloader. After getting the 6.2 firmware onto the BP successfully, the Live monitor and Transparent bridge now worked. I was still getting comms in only one direction with Transparent bridge, but that might be something to do with the UART I'm trying to debug.

Whatever the case, both Transparent bridge and Live monitor appear to be broken in firmware v6.3.

USBEprom, 2019-02-02:
The current community firmware is the v7.x.
If you want, could you try this firmware for your specific application to see what happens and report it here?

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=120#p67834

That I know it is the latest release ready to use that can be used with the Bus Pirate v3.
Please, read its description carefully.
Of course if you wish you can also compile by yourself the current repository:

https://github.com/BusPirate/Bus_Pirate

Your help would be greatly appreciated, many thanks in advance!

Twilight-Logic, 2019-02-03:
I'm not finding v7.x to download anywhere?
BPv3-firmware-v6.3-r2151.hex seem to be the latetest available.

Where can I find v7.x please?

Twilight-Logic, 2019-02-04:
Thanks. I downloaded that archive. I see it does contain a lot of other stuff as well including the S_1-05112018.hex file as well. I will give this a try. The filename confused me at first as it gives the impression that this is a bootloader of some kind?

I did read the thread in that link you posted. The blue "links" in the second post are not hot links so you have to copy and paste into the URL box in the browser. The first one took me to the home of the BusPirate community Github site. I looked for a version 7 in packages as well as in the /archive directory shown in the second link, but there is no v7 in Packages and no ./Archive directory visible. I then copied the second link and opened that in the browser. This downloaded the master.zip file, but there appears to be no version 7 in this file, unless its in the sources in the Firmware directory?

The file structure of the GitHub is at best unclear. For example, I looked in the ./tools directory hoping to find ds30 and/or pirate loader, which appeared to be nowhere to be found, although I did stumble on a rahter expensive commercial version of ds30. Eventually I found they are bundled in with the package download. For Linux, I eventually found the ds30 command line separately in a forum post link. Was it unreasonable of me to expect to find these tools in "tools"? 

The Dangerous Prototyes site that comes up when you search for BusPirate firmware seems to contain older stuff and old links. I now understand that is has been abandoned but it doesn't seem to link to any of the newer community stuff?

I haven't tried the new firmware yet, but I would like to say thanks to the community developers for continuing the work on BusPirate. 

I now understand that the storage space is so limited the v3 board that nothing new can be added hence the v3 BusPirate is finished in terms of new development. I researched the matter before purchased and found on the Dangerous prototypes and Sparkfun sites clear warnings that if I wanted a stable board I should buy a v3 board, so this is what I did. The reason given was that the v4 code is still being developed and buggy. No mention that the v3 board has run out of storage for future development.... From what I can tell, v6.2 and v6.3 firmware was released quite some time ago so I suspect that these statements are out of date and therefore mis-leading. If I had known that, I might have purchased a v4 board. Perhaps the mentioned sites should be encouraged to remove it to avoid drawing prospective purchase of legacy hardware?

USBEprom, 2019-02-04:
> 
> 
> Thanks. I downloaded that archive. I see it does contain a lot of other stuff as well including the S_1-05112018.hex file as well. I will give this a try. The filename confused me at first as it gives the impression that this is a bootloader of some kind?

As described in the link to the download, S_1-05112018.hex is the so called SAFE version (S=SAFE) of the firmware I built and shared on the dangerousprototypes forum.
Into the archive bootloader is named BPv3-bootloader-upgrade-v4xtov4.5.hex, that is the latest release available for the Bus Pirate v3.
S_1-05112018.hex (and U_1-05112018.hex too) require the new bootloader v4.5 in order to work completely (viewtopic.php?f=28&t=8498&start=75#p67056 and viewtopic.php?f=28&t=8498&p=67056#p67127).
Otherwise if do not, enter BOOTLOADER using command "$" will not work (https://github.com/BusPirate/Bus_Pirate/issues/58).
Actually it is not really a problem on the firmware side but rather of the bootloader.
In order to fix the matter it is need to upgrade the bootloader to the new revision v4.5:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=619#p67037

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=60#p67043

https://github.com/BusPirate/Bus_Pirate/issues/38

https://github.com/BusPirate/Bus_Pirate/issues/58

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67056#p67056

So, if entering BOOTLOADER by closing the jumper between the PGC and PGD pins on the ICSP header of the Pirate Bus v3 instead of using the "$" command from the terminal is not a problem, it is not strictly necessary to update the bootloader too, but it would be good to do it anyway.
In the archive it is also provided a pre configured environment that anyone can use with minimal changes to simplify the upgrade of the only firmware or together the bootloader and the firmware in a single step.
There are also instructions on how to use the whole thing under any operative system (Window, Linux and Apple) :

"How to use UPGRADE_TO_BL_v4.5.bat.rtf" (specific instructions for Windows users)
"How to update with pirate-loader.rtf" (generic cross-platform instructions for Windows, Linux and Apple users).

> I did read the thread in that link you posted. The blue "links" in the second post are not hot links so you have to copy and paste into the URL box in the browser. The first one took me to the home [...]

I am use Firefox on Windows 10 computer and I have not the problem you wrote about >The blue "links" in the second post<.
However master.zip in https://github.com/BusPirate/Bus_Pirate is only source of the project, it does contain compiled .hex in the path:

Bus_Pirate-master\package\BPv3-firmware\

There it is BPv3-firmware-v7.1-1.hex together a bunch of other previous releases, please take a look there if you do not trust to use S_1-05112018.hex.
Otherwise it is need the user compile the firmware from hisself, hence or you build it from yourself, or you use the BPv3-firmware-v7.1-1.hex into master.zip, or you use firmwares S_1-05112018.hex / U_1-05112018.hex I shared in the forum, unless someone provides it to you a different one firmware v7.x.
Both S_1-05112018.hex and U_1-05112018.hex are firmware v7.1 for the Bus Pirate v3.
Please read carefully here: https://github.com/BusPirate/Bus_Pirate/issues/115.

> Searching on the Internet, I found a second GitHub repository containing just the v7 release, but only for BP4, but also confusingly called "BusPirate/Bus_Pirate". Whether you get to one or the other depends on the exact search terms. "buspirate community" gets you links to the repository with the older stuff. "buspirate community v7" gets you links to the newer v7 for BP4 only repository. I got the impression that v7 was for BP4 only?

No, firmware v7 is for both Bus Pirate revision 4 and Bus Pirate revision 3, please read the root of the link I wrote where you found the package BPv3-bootloaderv4xtov45-update-vb_ENG-h.7z.
There in the beginning of the thread (http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498#p65073) it is clearly explained what exactly is Community Firmware 7.0.
Please read it carefully.
Community members have made titanic efforts to make the new firmware v7.x compatible with the Bus Pirate v4, v3 and even previous hardware!

> The file structure of the GitHub is at best unclear. For example, I looked in the ./tools directory hoping to find ds30 and/or pirate loader, which appeared to be nowhere to be found, although I did stumble on a rather expensive commercial version of ds30. I eventually discovered that these tools are bundled in with the package download. For Linux, I found the ds30 command line program separately in a forum post link. Was it unreasonable of me to expect to find these tools in "./tools"?

I can not write nothing about the file structure of the GitHub but write that the tools you are looking for are into the same path inside of which are the .hex files:

Bus_Pirate-master\package\BPv3-firmware\

About ds30 loader please read carefully this:

https://github.com/BusPirate/Bus_Pirate/issues/77

> The Dangerous Prototypes site that comes up when you search for "BusPirate firmware" seems to contain older stuff and old links. I also came across a Google site with even older stuff. I now understand that Dangerous Prototypes have abandoned the project but neither they nor Sparkfun appear to link to any of the newer community stuff?

[... cut ...]
Remember that even it is now, on the Bus Pirate v3 can be put specific firmware that can allow it to do specific things, just like the ones into path Bus_Pirate-master\Firmware-Alternates\ and Bus_Pirate-master\package\BPv3-firmware\ into master.zip.
However if for you what the Bus Pirate v3 can do is not enough you can always wait for the "v5" or a different device, that only depends on you.

> I haven't tried the new firmware on the BP3 board yet, but I would like to say thanks to the community developers for continuing the work on BusPirate.

Me too I join the due thanks!

Twilight-Logic, 2019-02-06:
I am using FF 64.0 on Linux, but I have also tried in Chromium.
This is the section I was referring to:

> Hi guys.
> Thanks to Christopher Sam Soon who has fixed and improved I2C, 2-WIRE and 3-WIRE protocol, agatti has built a new repository of the firmware v7.1 for the Bus Pirate (**https://github.com/BusPirate/Bus_Pirate**) so that I was able to build new firmwares v7.1 for the Bus Pirate revision 3 (**https://github.com/BusPirate/Bus_Pirate/archive/master.zip**).
> I made two versions that i named S_1-05112018.hex and U_1-05112018.hex.
> Both of them have all the features enabled, BASIC too, the command line allows up 256 characters.
> The first one (S_1-05112018.hex), the so called SAFE version (S=SAFE), is built on the configuration.h changed as follow.
> Into configuration.h I changed lines from 316 to 331 in order to enable all the features available, BASIC too.
> 

This links (in the original post) that I have highlighted are definitely not clickable.

I have now upgraded the bootloader to v4.5 as per the instructions and the upgraded to firmware v7. Unfortunately, the Transparent Bridge and the Live Monitor features are still broken in v7 of the firmware.

I can also confirm that bootloader 4.5/firmware v7 does not respond to the $ command to jump to the bootloader for for programming. The word BOOTLOADER is displayed, but the device does not respond with BL4+ when return is pressed.

Thankfully, the programming link still does work!

I have reverted back to v6.2 for now. Something changed in v6.3 that broke the UART features and this seems to have been carried over into v7. The bootloader $ function changed somewhere after v5.10. It worked OK with the original 5.10 firmware that came supplied on the board.

USBEprom, 2019-02-06:
> 
> 
> I am using FF 64.0 on Linux, but I have also tried in Chromium.
> This is the section I was referring to:
> 
> > Hi guys.
> > Thanks to Christopher Sam Soon who has fixed and improved I2C, 2-WIRE and 3-WIRE protocol, agatti has built a new repository of the firmware v7.1 for the Bus Pirate (**https://github.com/BusPirate/Bus_Pirate**) so that I was able to build new firmwares v7.1 for the Bus Pirate revision 3 (**https://github.com/BusPirate/Bus_Pirate/archive/master.zip**).
> > I made two versions that i named S_1-05112018.hex and U_1-05112018.hex.
> > Both of them have all the features enabled, BASIC too, the command line allows up 256 characters.
> > The first one (S_1-05112018.hex), the so called SAFE version (S=SAFE), is built on the configuration.h changed as follow.
> > Into configuration.h I changed lines from 316 to 331 in order to enable all the features available, BASIC too.
> 
> Those links that I have highlighted are definitely not clickable in the original post on dangerousprototypes.com.

Ok, I did not understand that it was referred to the dangerousprototypes forum.
There it is just as you wrote, sorry, the text was non intended as clickable but simply cut and paste.

> I have now upgraded the bootloader to v4.5 as per the instructions and the upgraded to firmware v7. Unfortunately, the Transparent Bridge and the Live Monitor features are still broken in v7 of the firmware.

Thanks for reporting it!
Could you please describe deeper the whole scenario in which you find the problem?
This would help to solve it.
Thank you!

 
> I can also confirm that bootloader 4.5/firmware v7 does not respond to the $ command to jump to the bootloader for for programming. The word BOOTLOADER is displayed, but the device does not respond with BL4+ when return is pressed.

Are you sure about this?
Bootloader v4.5 fix the bug you just wrote, could you please post the result of the 'i' command?
Even if you are back to firmware v6.2 the bootloader still needs to be v4.5 unless you have replaced it on purpose.
What you wrote is rather weird because until now nobody reported it.
You wrote bootloader 4.5/firmware v7 does not respond to the $ command to jump to the bootloader for programming, ok, but exacly what firmware?
Please, do you mean S_1-05112018.hex / U_1-05112018.hex, BPv3-firmware-v7.1-1.hex or what?
Thanks

> Thankfully, the programming link still does work!

As per the instructions it is still possible to revert to previous bootloader simply following them by using the wanted bootloader, otherwise it is need a programmer like PICKit2 or stuff like that.
All this is clearly explained in the thread where the package is.

> I have reverted back to v6.2 for now. Something changed in v6.3 that broke the UART features and this seems to have been carried over into v7. The bootloader $ function changed somewhere after v5.10. It worked OK with the original 5.10 firmware that came supplied on the board.

The bootloader $ function is fine with firmware v7.x and the latest bootloader v4.5, really I do not understand why you have the problem you wrote.
What version exactly is the Bus Pirate you own?
Thanks.
The bootloader $ problem you wrote exists while using firmware v7.x with bootloader v4.4 and previous releases, it should not exist with firmware v7.x and bootloader v4.5 because the latter fix that issue.

About the issue you are dealing with, do you know this?:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=9026

Twilight-Logic, 2019-02-07:
> Could you please describe deeper the whole scenario in which you find the problem?

DETAILED DESCRIPTION OF THE PROBLEM:

Here are the steps:

- connect BusPirate to a serial device, e.g. another UART
- m for mode, select 3. UART
- select baud rate, databits/parity, stop bits, idle 1 *default, open drain
- P enable pullup resistors
- select either macro (1) or macro (2)

In macro (1) Characters should be displayed on the serial UART terminal when typed on the BusPirate and vice versa, but nothing appears when using firmware 6.3 or 7. I only tested with the safe version S_1-05112018.hex.

Regarding the bootloader issue steps to reproduce the issue are:

- without connecting the link, open terminal to BusPirate
- type $ <return>
- in response to 'Are you sure?' type y

The word BOOTLOADER appears.

- hit <return> - nothing happens
- try to connect with ds30 or pirate-loader - it will complain that the bootloader is not responding correctly.

- Power down the BusPirate. Connect the link. Power it back up again.
- open a terminal. Hit <return>. BL4+ is displayed

Buspirate now responds to programming by ds30 and pirate-loader.

HISTORY:
- Original board received with firmware version 5.10 and bootloader 4.4.
- firmware upgraded using only the $ command to invoke the bootloader to version BPv3-firmware-v6.3-r2151.hex
- noticed issue with UART functions
- attempted to downgrade to version BPv3-firmware-v6.2-r1981.hex using the $ command to invoke the bootloader. Found the problem described above.
- found that I now had to insert the link. Firmware was then downgraded successfully.
- upgraded bootloader to v4.5 following instructions in the supplied package BPv3-bootloaderv4xtov45-update-vb_ENG-h.7z
- rebooted BP and installed firmware as per instructions on screen S_1-05112018.hex
- UART function still has the issue described above
- attempted to downgrade back to firmware BPv3-firmware-v6.2-r1981.hex using the $ command
- noted same issue with bootloader
- installed link and performed the downgrade
- UART function now works. Bootloader still has the same issue.

VERSION INFORMATION:

> What version exactly is the Bus Pirate you own?

There is no version information on the board itself.

This is what BusPirate currently reports:

Bus Pirate v3.5
Firmware v6.2-beta1 r1981 Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com

Board was purchased just a couple of weeks ago.

USBEprom, 2019-02-07:
@Twilight-Logic
Thank you very much for the extremely accurate report!

> > Could you please describe deeper the whole scenario in which you find the problem?
> 
> DETAILED DESCRIPTION OF THE PROBLEM:
> 
> Here are the steps:
> 
>     * connect BusPirate to a serial device, e.g. another UART (test hardware included an FTDI USB to serial adapter and an Arduino with CH340 based UART, both with 0/5v logic levels)
> 
>     * m for mode, select 3. UART
> 
>     * select baud rate, databits/parity, stop bits, idle 1 *default, open drain (the exact serial parameters do not have a bearing on the problem, but must obviously match the DUT)
> 
>     * P enable pullup resistors
> 
>     * select either macro (1) or macro (2)
> 
> 
> In macro (1) Characters should be displayed on the serial UART terminal when typed on the BusPirate and vice versa, but nothing appears when using firmware 6.3 or 7. I only tested with BPv3-firmware-v6.3-r2151.hex and the safe version of 7, file S_1-05112018.hex.

This must be checked

> 
> Regarding the bootloader issue steps to reproduce the issue are:
> 
>     * without connecting the link, open terminal to BusPirate
> 
>     * type $
> 
>     * in response to 'Are you sure?' type y
> 
> 
> The word BOOTLOADER appears.
> 
>     * hit [return] - nothing happens
> 
>     * close terminal
> 
>     * try to connect with ds30 or pirate-loader
> 
> 
> The programmer will complain that the bootloader is not responding correctly.
> 
>     * power down the BusPirate
> 
>     * connect the link
> 
>     * power BusPirate back up again.
> 
>     * open a terminal. Hit [return]. BL4+ is displayed
> 
> 
> Buspirate now responds to programming by ds30 and pirate-loader.

For what I can understand your Bus Pirate is the same of mine, but I have no the issue you wrote.
Silicon revision of the PIC welded on the board too is the same, so that the UNSAFE version of the firmware can be used without fear.
Here is the output of the terminal while performing what you have described:

i
Bus Pirate v3.5
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO] Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
BL4+

What is the terminal you are using and what is the operating system you are running?
What is the answer you have while performing self test (command "~") on your Bus Pirate?
Can you post it?

> HISTORY:
> 
>     * Original board received with firmware version 5.10 and bootloader 4.4.
> 
>     * firmware upgraded using only the $ command to invoke the bootloader to version BPv3-firmware-v6.3-r2151.hex (flashrom requires v6.2 or above)
> 
>     * noticed issue with UART functions
> 
>     * attempted to downgrade to version BPv3-firmware-v6.2-r1981.hex using the $ command to invoke the bootloader. Found that this would no longer work.
> 
>     * discovered that I now had to insert the link. Firmware could then be downgraded successfully.
> 
>     * downgraded successfully to BPv3-firmware-v6.2-r1981.hex
> 
> 
> After raising these issues here:
> 
>     * installed link
> 
>     * upgraded bootloader to v4.5 following instructions in the supplied package BPv3-bootloaderv4xtov45-update-vb_ENG-h.7z using UPGRADE_TO_BL_v4.5.bat
> 
>     * rebooted BP, opened terminal, got new bootloader message
> 
>     * closed terminal, successfully installed firmware S_1-05112018.hex using UPGRADE_FW_S_1-05112018.hex.bat
> 
>     * removed link and restarted BusPirate
> 
>     * tested and found that UART function still has the issue described above
> 
>     * attempted to downgrade back to firmware BPv3-firmware-v6.2-r1981.hex using the $ command
> 
>     * downgrade fails with the same problem -  the bootloader is not responding
> 
>     * installed link and performed the downgrade
> 
>     * UART function now works again. The $ command still has the same issue.
> 
> 
> VERSION INFORMATION:
> 
> > What version exactly is the Bus Pirate you own?
> 
> I missed it initially but after closer examination eventually found 'v3.6a' etched on the edge of the board in the copper. It happens to be in just the right position for it to be covered over by the right thumb when you naturally hold the board with the logo the right way up!
> 
> This is what the BusPirate currently reports in response to the 'i' command:
> 
> Bus Pirate v3.5
> Firmware v6.2-beta1 r1981 Bootloader v4.5
> DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
> http://dangerousprototypes.com
> 
> Board was purchased just a couple of weeks ago.

As already suggested try to see what result gives you the self test (command "~") just to exclude hardware problems although using different firmwares everything works, so the problem should not be that, better check though.
While you are there, do you know this?:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=7047#p60489

Due the fact right now with previous releases of the firmware together the new bootloader v4.5 everything works as it should, then this last can not be the culprit.

Twilight-Logic, 2019-02-08:
I am using the PuTTY terminal on both Linux Mint 19 and Windows 7.

SELF TEST RESULT:

HiZ>~
Disconnect any devices
Connect (Vpu to +5V) and (ADC to +3.3V)
Space to continue
Ctrl
AUX OK
MODE LED OK
PULLUP H OK
PULLUP L OK
VREG OK
ADC and supply
5V(4.97) OK
VPU(4.99) OK
3.3V(3.28) OK
ADC(3.26) OK
Bus high
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 0
MOSI OK
CLK OK
MISO OK
CS OK
Bus Hi-Z 1
MOSI OK
CLK OK
MISO OK
CS OK
MODE and VREG LEDs should be on!
Any key to exit
Found 0 errors.
HiZ>

MODE and VREG leds were illuminated.

- And since you had the problem even before updating, the culprit can not even be that.

I didn't say that! The original board with v5.10 firmware and bootloader v4.4 worked fine, it was just that the firmware version was too early for the flashrom utility to work with it. Flashrom requires firmware v6.2 or above. Problems started only after the first (v6.3) firmware update.

USBEprom, 2019-02-09:
@Twilight-Logic
 
> I am using the PuTTY terminal on both Linux Mint 19 and Windows 7. I have also tried C-Kermit.

Thanks for the clarification.

> SELF TEST RESULT:
>
> HiZ>~
> Disconnect any devices
> Connect (Vpu to +5V) and (ADC to +3.3V)
> Space to continue
> Ctrl
> AUX OK
> MODE LED OK
> PULLUP H OK
> PULLUP L OK
> VREG OK
> ADC and supply
> 5V(4.97) OK
> VPU(4.99) [...]

So it seems like everything is right there.

> I didn't say that! The original board with v5.10 firmware and bootloader v4.4 worked fine, it was just that the firmware version was too early for the flashrom utility to work with it. Flashrom [...]

Please pay attention that the issue you have opened is described as "UART Transparent bridge and Live monitor bruken in 6.3_r2151" and I am responding to that, here does not matter the issue about command "$" you wrote.
If you wish you can open a special issue for the "$" command, that would make the report less confusing, thanks.
Therefore it is your opinion that transparent bridge and live monitor are functioning with firmware 5.10, which nowhere in your posts seems to me to be explicitly stated, can you confirm it please?
In any case as I wrote this must be checked.
By going ahead, surely jumping to bootloader via software in firmware v6.2 does not works, please read this:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=6690

The hardware of my Bus Pirate is the same as yours, checking out some firmware I found the following:

i
Bus Pirate v3b
**Firmware v6.0 r1625**  Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
**BL4+BL4+BL4+BL4+BL4+ (WORKING)**

i
Bus Pirate v3.5
**Firmware v6.1 r1676**  Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
**BL4+BL4+BL4+ (WORKING)**

i
Bus Pirate v3.5
**Firmware v6.2-test r1862**  Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
**BL4+BL4+BL4+BL4+BL4+ (WORKING)**

i
Bus Pirate v3.5
**Firmware v6.2-beta1 r1981**  Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
**(NOT WORKING)**

i
Bus Pirate v3.5
**Firmware v6.3-beta1 r2151**  Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
**(NOT WORKING)**

i
Bus Pirate v3.5
**Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB LCD PIC DIO]** Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA00 2 B8)
http://dangerousprototypes.com
HiZ>$
Are you sure? y
BOOTLOADER
**BL4+BL4+BL4+BL4+ (WORKING)**

Based on the above the firmware v7.x, as I have already written, has not the problem you describe while enter bootloader with command "$".
Do not take it badly, I do not want to offend you, but I suspect that for your inexperience with the device you unconsciously are making some mistake, otherwise your Bus Pirate has a problem and you should return it where you took it since you bought it a short time ago.
Please, understand what I mean.

Surely problems with transparent bridge and live monitor have started with firmware v6.2:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=4007 

Anyway here you go firmware v6.3-beta1 r2151 fixed by Tayken for SPI problems:

http://dangerousprototypes.com/forum/download/file.php?id=11261

It has the issue you wrote about enter bootloader with command "$" but flashrom is fully functioning with it:

i
Bus Pirate v3.5
**Firmware v6.3-beta1 r2151**  Bootloader v4.5
DEVID:0x0447 REVID:0x3046 (24FJ64GA002 B8)
http://dangerousprototypes.com
HiZ>

I:\Bus Pirate\FlashROM\flashrom v0.9.9-1-stable-f5dd7ce Win64>flashrom.exe -p buspirate_spi:dev=COM3
flashrom v0.9.9-1-stable-f5dd7ce on Windows 6.2 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

**Calibrating delay loop... OK.
Found Winbond flash chip "W25Q16.V" (2048 kB, SPI) on buspirate_spi.
No operations were specified.**

I:\Bus Pirate\FlashROM\flashrom v0.9.9-1-stable-f5dd7ce Win64>

Same thing with the firmware S_1-05112018.hex and U_1-05112018.hex.

> Here is what I get without the link:
> 
> $
> Are you sure? y
> BOOTLOADER
> 
> After this point the BP no longer responds to keyboard input.
> 
> With the link ([return] hit 3 times):
> 
> BL4+BL4+BL4+
> 
> If I keep hitting [return] then more instances of 'BL4+' will be displayed.

What you have just written is exactly what makes me suspect a your mistake for inexperience, as what you describe is exactly the normal behavior that you must expect!
That being the case, there's no problem there!
However, you could please provide the exact firmwares with which you have the problem, as well as the version of flashrom that gives you problems (better if for Windows since you wrote that you also use Windows 7) and maybe specify the chip you would like to work?
Thank you.

Twilight-Logic, 2019-02-10:
As requested I have opened a separate issue for the bootloader issue.

USBEprom, 2019-02-10:
@Twilight-Logic

>
> As requested I have opened a separate issue for the bootloader issue. Please, by all means, move any related comments into that issue. Sorry about the confusion.

Thanks for your understanding, no need to apologize at all.

> Regarding the transparent bridge and live monitor issue, let me try and make it clear one final time:
> 
> firmware v5.10 - OK
> firmware BPv3-firmware-v6.2-r1981.hex - OK
> firmware BPv3-firmware-v6.3-r2151.hex - broken
> firmware S_1-05112018.hex (v7.0?) - broken
> firmware U_1-05112018.hex (v7.0?) - untested

Thanks a lot for your list!

> If you have a firmware 7.1 you want me to test then by all means tell me where I can download it from and I will give it a try.

The firmware v7.x I have and I am using is firmware U_1-05112018.hex, but I do not know if also it has the issue you wrote, I have to check that.
Since firmware U_1-05112018.hex is the same as firmware S_1-05112018.hex except that this last one does not have the I2C hardware protocol unlocked, and you wrote it does not work while in UART transparent bridge and live monitor, so I think it would be very unlikely that it would work for you.
In any case it is up to you whether to try it or not.
The only other firmware v7.x ready to use I know is the BPv3-firmware-v7.1-1.hex in the path Bus_Pirate-master\package\BPv3-firmware\ inside master.zip.
You could even try that, if you did not already.

One last thing please, about the behavior of transparent bridge and live monitor you noticed, could you please tell if you know this?:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=9026

Many thanks

Twilight-Logic, 2019-02-10:
Yes, I read that thread and I am aware of this issue. I have recently acquired and have now also tested with an FTDI USB UART board with the same result as above. However, with the FTDI board I do get two way comms for firmware BPv3-firmware-v6.2-r1981.hex.

I have now also tried firmware U_1-05112018.hex. As you suspected, it does not solve the UART problem.

USBEprom, 2019-02-11:
@Twilight-Logic

Thanks for having shared your experience and for your detailed reports.
Today I have been able to do some tests with the help of the Arduino MEGA 2560 of the usual friend of mine and I confirm the problem you wrote also with firmware U_1-05112018.hex and firmware S_1-05112018.hex.
So in the end UART transparent bridge and live monitor is also broken in firmware U_1-05112018.hex and firmware S_1-05112018.hex.
Unluckily I can not do anything to fix the problem, I can only do tests, so for its resolution will be require the help of third parties, sorry.

Twilight-Logic, 2019-02-12:
Thanks for performing the tests and confirming the issue. Will you pass this onto the third parties or do I need to contact someone?

agatti, 2019-02-13:
Would it be possible to try out different code versions to see exactly which code change broke the feature?  The two post-DangerousPrototypes versions you mention are two years apart, and a lot changed in the meantime.

If you are able to build the firmware yourself, can you try to check out older versions in the git repository to narrow down what broke the code?

Twilight-Logic, 2019-02-15:
agatti, thanks for your comment but how many versions of firmware have there been between firmware BPv3-firmware-v6.2-r1981.hex and firmware BPv3-firmware-v6.3-r2151.hex? I actually see only one on the GitHub, namely BPv3-firmware-v6.3-r2088.hex. I downloaded that and gave it a try. I have to report that the UART mode transparent bridge feature is broken in that one as well.

Unless there is a later v6.2 that is not on the GitHub, BPv3-firmware-v6.2-r1981.hex seems to be the last firmware version where this worked.

USBEprom, 2019-02-16:
@Twilight-Logic

> Will you pass this onto the third parties or do I need to contact someone?

For third parties I mean someone who knows how to program that can correct the problem.
I definitely could not do it, the most I can do is check firmware and functionality.
There is no need to pass this to them, they read and in case they provide the solution, there are no specific people to invoke, anyone able to help will do it if he want to do so.

OneRedPetal, 2019-07-14:
I had this problem too,  but reading the notes on the 6.3 firmware release came across this

>  Bus Pirate firmware version change to v6.3 + Issues 57 and 24 ... 
>
>issue 57
>http://dangerousprototypes.com/track/view.php?id=57
>a clutch is added between the setup process and the start of using a protocol, Once setup is done all the pins are still at HiZ. only when you type in the 'W' commend does the clutch engage - pins are actualy connected to peripherals...

I found that  after configuring the UART through m, 3, baudrate select , Data bits (1), Stop bits (1),
Polarity  (1),  output type (2)  following the instructions above and entering  "W" followed by selecting the  macro, n my case I wanted live monitor so (2),  then it worked fine.   It is a bit confusing since W turns the PSU on.

I have a Buspirate 3.6 with community firmware v7.1   SAFE_16072017_1.hex and bootloader 4.5

USBEprom, 2019-07-14:
@OneRedPetal

Thanks a lot for having shared what you wrote and for your hint!
Today I have been able to repeat some tests with the help of the Arduino MEGA 2560 of the usual friend of mine and I confirm that what you wrote works like a charm also with firmware 09032019_OPT1_SAFE.hex and 09032019_OPT1_UNSAFE.hex.
So actually UART transparent bridge and live monitor is not broken in firmwares 09032019_OPT1_SAFE.hex and 09032019_OPT1_UNSAFE.hex, as well in other previous Community Edition releases and of dangerousprototypes too, just it is need to operate like you wrote.
I agree with you that it is a bit confusing because actually "W" turns the PSU on but I think that doing what is written in http://dangerousprototypes.com/track/view.php?id=57 must not be considered like a workaround, but simply the right way that has to be followed, paying attention do not connect the power supplies if you do not need it: trite now that you wrote it.
At this point I do not know if it would be useful to close the issue or not, because by honoring the statements in your link all works as it is supposed to do.

psolyca, 2019-07-14:
It is also the way I use it and it works.

Twilight-Logic, 2019-07-15:
I flashed my BusPirate this morning with S_1-05112018.hex (one of the firmware files that I tested previously) and tried the W command as suggested. It does indeed work! I confess that I had wondered what the "Clutch disengaged!!!" statement was all about at the time. I haven't tried the 2019 firmware yet.

Since in my scenario, both devices (BusPirate and Arduino) were powered independently from USB with only two data wires linking them, I did not see any relevance in turning on the BusPirate on-board power supply - so I just ignored  the prompt:

"To finish setup, start up the power supplies with command 'W'"

Whereas I see the point behind this clutch arrangement, I didn't make the connection with the 'W' command. Perhaps the thinking was to save a bit of program space as constructing a separate function would no doubt have required more bytes?

Whatever the reasoning behind this arrangement, I now agree with USBEprom that the functionality is not broken, just somewhat obscured (however unintentionally) by the introduction of the "clutch" into the 'W' command. Thank you OneRedPetal for finding this information and posting it here for the benefit of all who might come across this issue.